lubeda/EspHoMaTriXv2 #143: [FEATURE REQUEST] Fullscreen progress bar

chertvl, 2023-11-10:
# Feature Request

### Is your feature request related to a problem?

It would be nice if we could create a full screen progress bar screen.

### Describe the solution / feature you'd like

I see it like this:
On the left there will be 16px text for a short title,
on the right there will be a value of 16px to display the real value,
and at the bottom there is a 1x32px progress bar that spans the entire width of the screen.

I know that we already have an icon_screen_progress, but the icon sometimes gets in the way, it will be something similar, where, in addition to the value, a short description of 4 letters will also fit.

Most often, it would be more convenient to template this screen in HA and not think about the fact that you need to substitute some kind of icon to understand what kind of progress bar it is.

An approximate diagram from my imagination:

### Additional context

[image]

andrewjswan, 2023-11-13:
Added to 2023.9.1

chertvl, 2023-11-18:
Added by https://github.com/lubeda/EspHoMaTriXv2/pull/144
Many thanks to @andrewjswan and @lubeda 
Feel free to reopen issue if some improvements is needed.

closed 2023-11-18 by chertvl