lubeda/EspHoMaTriXv2 #145: [QUESTION] Adding to Default Queue Loop

anthonylavado, 2023-11-11:
# Question

As I understand the documentation, there are two default queues - one on startup, and one on empty queue with no other screens. I know I can add a screen to the queue by calling it from a Home Assistant automation. Is there a way to add a screen with information from HA to the default queue loop?

As an example, the screen I'm adding is a weather screen, and the automation in Home Assistant does all the work to pick the correct icon, information, etc. But instead of having this called on the trigger, I'd just want it to be in the cycle of screens on the device. Date, Time, Weather. Let's say 10 seconds for date, 10 seconds for time, 20 seconds to show the current weather, 20 seconds to show the high/low temperatures for the day.

I know I could set the trigger to be more frequent on the HA side, but while the screen is in the queue, it's blocking (as is the current behaviour). If adding it to that queue isn't possible, is my option then to "build" a new queue with long lifetimes in Home Assistant? If I go that route, would it still prevent other screens (such as notification ones) from displaying once they are triggered?

Thanks! 

## Additional information

- used Hardware:
  - Ulanzi TC001

  - EspHoMaTriXv2 version: 2023.9.1

## Logs

N/A

## Additional context

None, but happy to test any possible solutions :)

lubeda, 2023-11-11:
Hi, 

based on branch 20233.9.1. There is one queue, that is empty on startup. To show always the clock and the date use the on_empty_queue in the yaml. E.G.

```
on_empty_queue:
    then:
      lambda: |-
        id(rgb8x32)->clock_screen(10,10);
        id(rgb8x32)->date_screen(10,5);
```

For the weather use an automation in home assistant that triggers on a state change of the weather entity. and call

`icon_screen` with a very long lifetime e.g. more than 24 hours = 24*60 minutes.

So date and time are always in the queue and the weather is also always in the queue and refreshed on every state change.

You can set the screen_time as you like. For the temperature you can do the same as for the weather.

Hope that helps.

anthonylavado, 2023-11-11:
Ah okay. So I already have `on_empty_queue` set up like that, which makes things easier.

Based on the name, I suppose I thought that having a screen in queue (such as the weather from automation) would "block" the defaults, since the queue isn't empty. Now I understand that once the `screen_time` expires, it goes back to the default screens before cycling back through the "added" screens.

Thank you for explaining! I may add my automations to the wiki (or my site) as it would work for anyone in Canada against the free national weather service (Environment Canada).

closed 2023-11-11 by anthonylavado

riker09, 2023-11-13:
I have to admit when reading the available documentation and judging from the action name `on_empty_queue` I made the same mistake as @anthonylavado and assumed my screens would overwrite the default date/clock screens. A short test cleared that up, but I do believe the documentation could be improved on that matter. By the way, is there a "Improve documentation" ticket (or perhaps, a branch) where one could collaborate?

lubeda, 2023-11-13:
Hi, I'm happy about every contribution. I think the easiest way is by PRs per chapter. All should be based on 2023.9.1. Another possibility are wiki articles.