lubeda/EspHoMaTriXv2 #146: [FEATURE REQUEST] no special font instead 8pix and 6 pix fonts

lubeda, 2023-11-14:
It would be nice to get rid of the default font flag in the functions calls. I use 8 pix fonts when there is nothing in the lower row. With Progressbar or day of week I use 6pix fonts.

This should be automatic but would be a breaking change.

andrewjswan, 2023-11-14:
I have it 90% of the time it's 6 font, 8 isn't always convenient. Still I think it is necessary either to make (I don't know how) it optional or let the user decide.

chertvl, 2023-11-14:
Always using only 6px font, it looks better as i think ;)

trip5, 2023-11-14:
ESPHome compresses the fonts as far as I can tell.  If you specify the same fonts, it should compress extremely well so the 2nd font doesn't use any space.

And please no, 2 fonts at 8 size is absolutely needed for me.  I don't use any 6 size fonts (yes, lol even though I make them).

closed 2023-12-27 by lubeda