lubeda/EspHoMaTriXv2 #147: [FEATURE REQUEST] Optimize a version to use with ESP32 and microphone for assist

lubeda, 2023-11-14:
In my lab I made (hacked together with duck tape and hot glue) a version that works with microphone as an assist satelite. 

The functionality was there but the display had some hiccups. I think because of timing issues. Voice is very depdendent on timing so the `esp32_rmt_led_strip`plattform.

My dream is to have some ehmtx devices in some rooms with a microphone for assist (Goodbye Alexa)

trip5, 2023-11-14:
Woah.  I thought I saw some things in the repo that looked like that...

Did you add a microphone or mic array?  There's a speaker in there but I assumed that could only be used as a speaker?

Would you be able to share pictures, diagrams, anything?

lubeda, 2023-11-15:
These are old videos with the press button to speak assist :https://www.youtube.com/shorts/wHDdW07svjc and https://www.youtube.com/watch?v=CsOir14DCbM 

With wake word the source has to be adapted but the impression is more or less the same

trip5, 2023-11-15:
Oh.  I thought you had experimented with getting the Ulanzi to act as a voice assistant.  I have my doubts that a bare ESP32 such as exists in the Ulanzi can handle wakeword...

All the other examples I've seen, voice assistant still struggles with wakewords even though the assistant is the only component of the ESPHome device.

I myself have just ordered a ESP32 Korvo https://www.aliexpress.com/item/1005002803964499.html to play with.  It seems to have a lot of dedicated hardware to it to make a voice assistant...

I'd almost think that it would be better to simply stuff another ESP32 inside the Ulanzi case, replace the speaker and just have 2 different devices in the same box.

lubeda, 2023-11-17:
the esp32 is capable to do voice activation. The case of the ulanzi is hard to open, I will buy a new one and try this: https://github.com/lubeda/EspHoMaTriXv2/wiki/ulanzi-hardware-mode and try also to add a mic.

closed 2025-02-24 by lubeda