lubeda/EspHoMaTriXv2 #150: No Update ESPHome 2023.11.0

clfberlin, 2023-11-15:
No Update possible in Home Assistant with Ulanzi. 
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

## Additional information

ESPHome 2023.11.0
Home Assistant
Core 2023.11.2
Supervisor 2023.11.3
Operating System 11.1.rc1
Frontend 20231030.2

## To Reproduce

Simple update results in error (and failed)

## Logs

`INFO ESPHome 2023.11.0
INFO Reading configuration /config/esphome/ulanzi-og.yaml...
INFO Updating https://github.com/lubeda/EspHoMaTriXv2@main
WARNING 
WARNING This will only work with esphome >= 2023.7.0
WARNING Please check the documentation and wiki https://github.com/lubeda/EspHoMaTriXv2
WARNING 
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1040, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1027, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 457, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 674, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/esphome/esphome/components/font/__init__.py", line 356, in to_code
    font = load_ttf_font(path, config[CONF_SIZE])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/font/__init__.py", line 340, in load_ttf_font
    from PIL import ImageFont
  File "/usr/local/lib/python3.11/dist-packages/PIL/ImageFont.py", line 35, in <module>
    from . import Image
  File "/usr/local/lib/python3.11/dist-packages/PIL/Image.py", line 82, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory`

## Services calls

One thing worth noting: Home Assistant is running on a 32bit system

andrewjswan, 2023-11-15:
Just updated, no problem.
[image]

clfberlin, 2023-11-15:
Thx. Then this is an individual issue (again). Sorry.

closed 2023-11-15 by clfberlin