lubeda/EspHoMaTriXv2 #155: [BUG]

lebherz, 2023-11-16:
# Bug report
switch to esphomematrixV2 brings me an Error:
`INFO ESPHome 2023.11.1
INFO Reading configuration /config/esphome/esphomematrix.yaml...
ERROR Error while reading config: Invalid YAML syntax:

Error reading file /config/esphome/ehmtx_on_time.yaml: [Errno 2] No such file or directory: '/config/esphome/ehmtx_on_time.yaml'`
## Describe the bug

can't compile

## Additional information
my esphomematrix.yaml:
`
substitutions:
  devicename: ehmtx8266V2
  mypin: D4
  board: nodemcuv2
  
external_components:
  - source:
      type: git
      url: components/ehmtxv2
      #ref: stable # optional select a special branch or tag

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: $devicename
    password: "InM2TlqVfJe4"

image:
   - file: 1pixel.gif
     id: breaking20237

animation:
  - file: 1pixel.gif
    id: breaking202371

esphome:
  name: $devicename
  comment: 8x32 RGB Display im Wohnzimmer

esp8266:
  board: $board

font: 
  - file: mateine.ttf
    id: default_font
    size: 16
    glyphs:  |
       !"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@üöäÜÖÄß
  
logger:
  level: WARN

api:
            
light:
  - platform: neopixelbus
    id: ehmtx_light
    type: GRB
    variant: WS2812
    pin: $ledpin
    num_leds: 256
    color_correct: [30%, 30%, 30%]
    name: "$devicename Light"
    restore_mode: ALWAYS_OFF
    effects:
      - e131:
          universe: 1
          channels: RGB
    on_turn_on:
      lambda: |-
         id(ehmtx_display)->set_enabled(false);
    on_turn_off:
       lambda: |-
         id(ehmtx_display)->set_enabled(true);

e131:
  method: multicast

time:
  - platform: homeassistant
    id: ehmtx_time
    on_time: !include ehmtx_on_time.yaml
      
display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (x % 2 == 0) {
        return (x * 8) + y;
      }
      return (x * 8) + (7 - y);
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();

ehmtxv2:
  id: rgb8x32
  show_seconds: true
  rainbow_interval: 32
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  icons2html: true
  default_font_id: default_font
  special_font_id: default_font
  special_font_yoffset: 8
  default_font_yoffset: 8

  icons: 
    - file: _icons/blitz.gif
      id: blitz
    - file: _icons/temp.gif
      id: temp
    - file: _icons/solar.gif
      id: solar
    - file: _icons/warnung.gif
      id: warnung
    - file: _icons/rolladenzu.gif
      id: rolladenzu
    - file: _icons/rolladenauf.gif
      id: rolladenauf
    - file: _icons/wetterwarnung.gif
      id: wetterwarnung
    - file: _icons/check.gif
      id: check
    - id: xani
      lameid: 6075
    - id: xsta
      lameid: 11236
    - url: https://developer.lametric.com/content/apps/icon_thumbs/48720.gif
      pingpong: true
      id: pipo
      frame_duration: 300
    - lameid: 5965
      frame_duration: 180
      id: d180
    - lameid: 5965
      frame_duration: 80
      id: d080
    - id: fullscreen
      file: sample8x32.gif
    - id: samplegif
      file: sample8x8.gif

`

- used Hardware:
  - ESP8266
  - EspHoMaTriXv2 version: 2023.7.0

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Configuration

```text
(optional) The YAML you used in epshome without any password
```

## Screenshots

[image]

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Services calls

```text
(optional) The YAML of your service calls
```

andrewjswan, 2023-11-17:
Remove this line:
```Yaml
on_time: !include ehmtx_on_time.yaml
```
and try again ...

lebherz, 2023-11-17:
Oh my god... So. Simple... Thx!

closed 2023-11-17 by lebherz

lubeda, 2023-11-17:
sorry for my sloppy coding and documentation style. fixed by @andrewjswan