lubeda/EspHoMaTriXv2 #160: [BUG] Breaking changes in esphome 2023.11 due to text component cause YAML syntax errors

N3rdix, 2023-11-20:
# Bug report
Since 2023.11 it is no longer possible to pass in variables into services with the name "text" as this seems now to be a reserved word from https://github.com/esphome/esphome/pull/5336. This breaks some yaml examples and won't compile correctly any longer due to YAML syntax errors like this:
`[text] is an invalid option for [variables].`

## Describe the bug

Functions like icon_screen use "text" as an argument and cannot be called any longer from [homeassistant api](https://esphome.io/components/api.html) .

## Additional information

- used Hardware:
  - ESPHome 2023.11.2
  - ESP32
  - EspHoMaTriXv2 version: newest

## To Reproduce

Take a service call as example and try to pass in a variable named "test"
```
api:
  services:
    - service: alarm
      variables:
        icon_name: string
        text: string
      then:
        lambda: |-
          id(rgb8x32)->icon_screen(icon_name,text,7,20,true);
          id(rgb8x32)->force_screen(icon_name);
```

## Expected behavior

Update the yaml examples and warn for a breaking change as service calls (as in my example above) might not work any longer once the variable names were changed in the yaml. 
I could create at least a PR to adjust the sample YAMLs and use an adjusted variables name and `c_str()` for all calls of icon_screen and other affected methods.

```
- service: alarm
    variables:
      icon_name: string
      display_text: string
    then:
      lambda: |-
        id(rgb8x32)->icon_screen(icon_name,display_text.c_str(),7,20,true);
        id(rgb8x32)->force_screen(icon_name);
```

lubeda, 2023-11-20:
I have to think about it, `text` is now a reserved word in esphome :-(

lubeda, 2023-11-20:
The solution:

```yaml
api:
  services:
    - service: alarm
      variables:
        icon_name: string
        message: string
      then:
        lambda: |-
          id(rgb8x32)->icon_screen(icon_name,message,7,20,true);
          id(rgb8x32)->force_screen(icon_name);
```

closed 2023-11-20 by lubeda

N3rdix, 2023-11-20:
Just compiled and tested, works fine 👍 Thanks for the quick response. Shall we adjust the samples anyways?

Just out of curiousity, how did you find out that "message" can be used? I could not find any hint in esphome PRs.

lubeda, 2023-11-20:
Its just a name, you can use more or less anything e.g. hotzenplotz but you can't use reserved words like `text, api, ....