lubeda/EspHoMaTriXv2 #164: [WIP] Advanced clock

andrewjswan, 2023-11-24:
# Improvement 

### Describe the solution / feature you'd like

This improvement will allow you to set the color of Hours and Minutes separately. 
If the seconds display mode is enabled `show_seconds: true`, the seconds will be displayed with a flashing separator rather than a dot in the corner.
The mode for time formats works: `%H:%M` `%H:%M:%S` `%I:%M%p` `%I:%M %p`
If `advanced_clock` is not enabled (`false`), or color values are not specified and the seconds display is not enabled `show_seconds: false`, then there will be no visual changes on the Clocks screen.

### Additional context

https://github.com/andrewjswan/EspHoMaTriXv2/tree/2023.9.1-Advanced_Clock

lubeda, 2023-11-24:
Hi, again very nice.
Is the flag `advanced_clock` necessary? The blinking dot was a workaround because I was too lazy to find a way to make the colon blink.

So `show_seconds` could make the colon blink instead of the dot, and the `set_adv_clock_color` could set the colors. The breaking change is acceptable in my opinion!

andrewjswan, 2023-11-24:
> Is the flag `advanced_clock` necessary?

Not everyone needs this functionality, so it's better to let it be optional.

> So `show_seconds` could make the colon blink instead of the dot

It is if this mode is enabled.

> and the `set_adv_clock_color` could set the colors. The breaking change is acceptable in my opinion!

It retains full compatibility, just added functionality, I'm still turning a simple watch into a spacecraft. :)

You can test it :)

```
  - source:
      type: git
      url: https://github.com/andrewjswan/EspHoMaTriXv2
      ref: 2023.9.1-Advanced_Clock
    components: [ ehmtxv2 ]
    refresh: 60s
```

andrewjswan, 2023-11-29:
[image]
[image]

andrewjswan, 2023-11-29:
```
ehmtxv2:
  id: rgb8x32
  icons2html: true
  display_graph: ehmtx_graph
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  time_format: "%H:%M"
  time_format_big: "%H:%M:%S"
  date_format: "%d•%m"
  weekdays: "일월화수목금토"

  default_font_id: default_font
  default_font_yoffset: 7
  special_font_id: special_font
  special_font_yoffset: 8
  blend_steps: 32
  scroll_count: 2 # scroll long text at least two times
  scroll_interval: 80 # milliseconds
  rainbow_interval: 32 # milliseconds

  show_seconds: false
  vertical_scroll: false
  advanced_clock: true
```