lubeda/EspHoMaTriXv2 #166: [FEATURE REQUEST] Ability to force text, rainbow_text screens

idodov, 2023-11-27:
Adding the capability to enforce not just icon screens would be highly appreciated.

riker09, 2023-11-29:
I don't fully understand what this feature request is about. Can you please explain in greater detail?

[image]

These two screens are not what you are referring to?

idodov, 2023-11-29:
When using the Ulanzi_force_screen service to immediately display on the LED matrix, i need to include "icon_name," which is not available on the full-text screen.

andrewjswan, 2023-11-29:
> which is not available on the text_screen & rainbow_text_screen

In 2023.9.1 - Just specify any one. Like: `rainbow` :)

idodov, 2023-11-29:
Where I need to specify the name?

I use the latest firmware. When I try to add a name to the screen I get **Error running action:** extra keys not allowed @ data['icon_name']. Got None.
```yaml
service: esphome.ulanzi_rainbow_text_screen
enabled: true
data:
  default_font: true
  text: >-
    {{ state_attr('sensor.bidi', 'media_artist_bidi') }} - {{
    state_attr('sensor.bidi', 'media_title_bidi') }}
  lifetime: >-
    {{ (state_attr('media_player.era300', 'media_duration') | float(default=0) /
    1) | int(default=1) if state_attr('media_player.era300', 'media_duration')
    is not none else 4 }}
  screen_time: 40
  icon_name: rainbow
```
I also tried this method without luck. Sometimes, I have more than one screen in the loop.
```yaml
service: esphome.ulanzi_force_screen
data:
  icon_name: rainbow
  mode: 8
```

lubeda, 2023-11-29:
Hi, I never used it but a * for `icon_name` should work

andrewjswan, 2023-11-29:
> Where I need to specify the name?

Only in force like
```
service: esphome.ulanzi_force_screen
data:
  icon_name: rainbow
  mode: 8
```

idodov, 2023-11-29:
I tried, but the screen appears a few seconds after I send the force screen command. The clock screen appears, not the text rainbow screen, which shows up after a few seconds.

This is a snip from my automation:
```yaml
      - conditions:
          - condition: trigger
            id:
              - title
              - music
          - condition: state
            entity_id: media_player.era300
            state: playing
        sequence:
          - service: esphome.ulanzi_del_screen
            data:
              icon_name: "*"
              mode: 8
            enabled: true
          - service: esphome.ulanzi_del_screen
            data:
              icon_name: "*"
              mode: 5
          - service: esphome.ulanzi_rainbow_text_screen
            enabled: true
            data:
              default_font: true
              text: >-
                {{ state_attr('sensor.bidi', 'media_artist_bidi') }} - {{
                state_attr('sensor.bidi', 'media_title_bidi') }}
              lifetime: >-
                {{ (state_attr('media_player.era300', 'media_duration') |
                float(default=0) / 1) | int(default=1) if
                state_attr('media_player.era300', 'media_duration') is not none
                else 4 }}
              screen_time: 40
          - service: esphome.ulanzi_force_screen
            data:
              mode: 8
              icon_name: rainbow
```

andrewjswan, 2023-11-29:
2023.9.1?

idodov, 2023-11-29:
This is the log. Version is above 2023.9.1 
```log
Logs pixel.yaml
INFO ESPHome 2023.11.6
INFO Reading configuration /config/esphome/pixel.yaml...
INFO Updating https://github.com/lubeda/EspHoMaTriXv2@main
WARNING 
WARNING This will only work with esphome >= 2023.7.0
WARNING Please check the documentation and wiki https://github.com/lubeda/EspHoMaTriXv2
WARNING 
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Asia/Jerusalem'
INFO Detected timezone 'Asia/Jerusalem'
INFO Starting log output from ulanzi.local using esphome API
INFO Successfully connected to ulanzi in 0.529s
INFO Successful handshake with ulanzi in 0.096s```

idodov, 2023-11-29:
I use the **main** firmware from your git

closed 2023-11-30 by idodov