lubeda/EspHoMaTriXv2 #174: [FEATURE REQUEST] Add the ability to change the icon color

andrewjswan, 2023-12-05:
# Feature Request

### Describe the solution / feature you'd like

Let's say we have an icon with some image, but we need to display it in different colors (green, yellow, red) depending on the value.
We call the service `icon_set_color(icon_name, r ,g ,b)` which changes all significant points of the icon to the specified color. Those. instead of three (or more) icons, we only need one, and we can display it in different colors.

lubeda, 2023-12-05:
Hi, why not use the small bitmap command? I don't see the use case. Can you explain i little bit more or give a sample

andrewjswan, 2023-12-05:
A bitmap is a way out, I agree, but only for static icons, and if we have an animated icon, for example `lameid: 14429`, then we won’t be able to use a bitmap.

lubeda, 2023-12-05:
ok so you like to manipulate a dual color gif eg. black and red like in 14429 and you will change every red pixel to the specified color?

andrewjswan, 2023-12-05:
Yes, but not everyone is red, and everyone is not black, i.e. If the color is displayed, change it to the one that is specified.
And so on every call to this service.

andrewjswan, 2026-09-07:
I'm still looking for a nice solution, so let it hang there for now...

closed 2026-10-21 by github-actions[bot]