lubeda/EspHoMaTriXv2 #184: [BUG] services are not working with EspHome 2023.12.3

nicetiger, 2023-12-23:
# Bug report

## Describe the bug

setting a screen does not work or works only once.

## Additional information

- used Hardware:
  - ESP8266 (D1 mini)
  - EspHome 2023.12.3
  - EspHoMaTriXv2 version: default and also 2023.9.1
  - HA Version 2023.12.3

## To Reproduce

setting a screen with the service 
```
service: esphome.matrixv2_icon_screen
data:
  default_font: true
  icon_name: attention
  text: test
  lifetime: 1
  screen_time: 1
  r: 255
  g: 255
  b: 255
```

causes "Unknown error" in the UI.

I can find the following error in the HA logs:
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 26, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 782, in handle_execute_script
    script_result = await script_obj.async_run(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1578, in async_run
    return await asyncio.shield(run.async_run())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 420, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 470, in _async_step
    self._handle_exception(
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 493, in _handle_exception
    raise exception
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 468, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 704, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 666, in _async_run_long_action
    return long_task.result()
           ^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2067, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2104, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/esphome/manager.py", line 746, in execute_service
    await entry_data.client.execute_service(service, call.data)
  File "/usr/local/lib/python3.11/site-packages/aioesphomeapi/client.py", line 1175, in execute_service
    int_type = "int_" if apiv >= APIVersion(1, 3) else "legacy_int"
                         ^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: '>=' not supported between instances of 'NoneType' and 'APIVersion'
```
and

```
Logger: homeassistant.helpers.script.websocket_api_script
Source: helpers/script.py:468
First occurred: 17:54:04 (81 occurrences)
Last logged: 22:22:47

websocket_api script: Error executing script. Unexpected error for call_service at pos 1: '>=' not supported between instances of 'NoneType' and 'APIVersion'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 468, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 704, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 666, in _async_run_long_action
    return long_task.result()
           ^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2067, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2104, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/esphome/manager.py", line 746, in execute_service
    await entry_data.client.execute_service(service, call.data)
  File "/usr/local/lib/python3.11/site-packages/aioesphomeapi/client.py", line 1175, in execute_service
    int_type = "int_" if apiv >= APIVersion(1, 3) else "legacy_int"
                         ^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: '>=' not supported between instances of 'NoneType' and 'APIVersion'
```

## Expected behavior

Screen is displayed, no error in HA

## Configuration

```
esphome:
  name: matrixv2

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "MaTriXv2 Fallback Hotspot"
    password: ""

captive_portal:

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      #ref: stable # optional select a special branch or tag
      ref: 2023.9.1

font:
  - file: mateine.ttf
    id: default_font
    size: 16
    glyphs:  |
      !"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@
  - file: MatrixChunky6.bdf
    id: special_font
    glyphs:  |
      ! "#$%&'()*+,-./0123456789:APMTapmt

e131:
  method: multicast # default: register E1.31 to Multicast group

time:
  - platform: homeassistant
    id: ehmtx_time

light:
  - platform: neopixelbus
    id: ehmtx_light
    type: GRB
    variant: WS2812
    pin: GPIO4
    num_leds: 256
#    color_correct: [30%,30%,30%]
    name: "MaTriX Light"
    restore_mode: ALWAYS_OFF
    on_turn_on:
      lambda: |-
         id(ehmtx_display)->set_enabled(false);
    on_turn_off:
       lambda: |-
         id(ehmtx_display)->set_enabled(true);

display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (x % 2 == 0) {
        return (x * 8) + y;
      }
      return (x * 8) + (7 - y);
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();

ehmtxv2:
  id: rgb8x32
  show_seconds: true
  rainbow_interval: 32
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  icons2html: true
  default_font_id: default_font
  special_font_id: special_font
  special_font_yoffset: 7
  default_font_yoffset: 8
#  default_clock_font: false
  icons: 
    - id: attention
      lameid: 555
    - id: fullscreen
      file: sample8x32.gif
    - id: samplegif
      file: sample8x8.gif
  
  on_empty_queue:
    then:
      lambda: |-
        id(rgb8x32)->clock_screen(10,10,false);
        id(rgb8x32)->date_screen(10,5,false);
  on_start_running:
     then:
       lambda: |-
          id(rgb8x32)->bitmap_screen("[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63519,63519,63519,63519,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63519,0,0,0,0,2016,0,0,0,0,0,0,0,0,0,0,31,0,0,0,0,0,0,0,0,0,63488,0,63488,0,0,0,63519,0,0,0,0,2016,2016,0,0,0,65514,0,65514,0,0,0,31,0,0,0,64512,0,0,64512,0,63488,63488,0,63488,63488,0,0,63519,63519,63519,0,0,2016,0,2016,0,65514,0,65514,0,65514,0,31,31,31,0,0,0,64512,64512,0,0,63488,63488,63488,63488,63488,0,0,63519,0,0,0,0,2016,0,2016,0,65514,0,65514,0,65514,0,0,31,0,0,0,0,64512,64512,0,0,0,63488,63488,63488,0,0,0,63519,63519,63519,63519,0,2016,0,2016,0,65514,0,65514,0,65514,0,0,0,31,31,0,64512,0,0,64512,0,0,0,63488,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]",1,10);
          id(rgb8x32)->bitmap_small("[2016,0,0,0,2016,0,0,0,2016,0,0,0,2016,0,0,0,2016,0,0,0,2016,0,0,0,0,2016,0,2016,0,31,31,0,0,0,2016,0,31,0,0,31,0,0,0,0,0,0,31,0,0,0,0,0,0,31,0,0,0,0,0,0,31,31,31,31]", "2023.x.y", 1, 10);
          id(rgb8x32)->clock_screen(10,10,false);
          id(rgb8x32)->date_screen(10,5,false);

animation:
  - id: animation_black
    file: "black.gif"

image:
  - id: image_black
    file: "black.gif"
```

andrewjswan, 2023-12-25:
EspHoMaTriXv2 2023.9.1, ESPHome 2023.12.5 - All work without problem ...

lubeda, 2023-12-27:
Hi, 
please try to remove the e131 stuff, the esp8266 has a poor performance and in conjunction with a poor wifi i can imagine the source of the error.

I highly recommend using an ESP32 with the later versions of esphomematrix 

```yaml
e131:
  method: multicast # default: register E1.31 to Multicast group
```

trip5, 2023-12-27:
To the OP, if you're stuck with an Awtrix form-factor, it's not a bad option to swap out the D1 mini for a D1 Mini ESP32 as so: https://www.aliexpress.com/item/32816117396.html

My Awtrix had a socket and the D1 Mini ESP32 has the same pin placement and everything so all I had to do was change this in my ESPHome YAML and I was up and running with fully-functioning EHMTXv2.
````
esp32:
  board: mhetesp32minikit
````

Bonus that it appears to be a full ESP32 dual-core as opposed to the "other ESP32" small boards like S3 which are only single-core.

frittes, 2024-01-28:
> To the OP, if you're stuck with an Awtrix form-factor, it's not a bad option to swap out the D1 mini for a D1 Mini ESP32 as so: https://www.aliexpress.com/item/32816117396.html
>
> My Awtrix had a [...]

Thanks for the hint. Ported my config with an advised ESP32-Mini to idf-framework and now I have a esphomatrix with bluetooth proxy. Had some of them for spare.

closed 2025-01-10 by lubeda