lubeda/EspHoMaTriXv2 #189: Character not being rendered correctly even it's in the font

Alex-joomla, 2023-12-28:
# Question

I call a service node in node red to add a Icpn_screen:

```{
    "icon_name": "dishwasher",
    "text": "{{payload}}",
    "lifetime": "1",
    "screen_time": "1",
    "r": 255,
    "g": 255,
    "b": 255,
    "default_font": true
}```

payload is this: `"Geschirrspüler: finished / noch 0 Minuten"`

But what I get are two big white blocks with a x2F in the middle  instead of the Dash. The Dash is a Glyph, i know, but it's availible in my font:
```
font:
  - file: CalciumThin.ttf
    id: default_font
    size: 16
    glyphs:  |
      !?"%‰()+*=,'-_.:°µ²³0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnÖÄÜöäüopqrstuvwxyz€$@<>/
```

## Additional information

- used Hardware:
  - Ulanzi PixelClock

  - EspHoMaTriXv2, newest

## Logs

## Additional context

trip5, 2023-12-28:
This likely isn't an issue with ehmtx. Like the dash being used in the payload isn't the same as the hyphen-dash you're including in the glyphs section. Node red may be giving you an em dash or en dash...?

Alex-joomla, 2023-12-28:
Thank you for your hints. Interesting thing is: if I feed the glyph into the call service node via a flow it gets rendered incoreectly. But if I type it directly in the call service node, it works.
Wont work:
[image]

Works:
[image]

Should I take it to a NodeRed Forum?

trip5, 2023-12-28:
Probably?  I've somehow missed adding Nodered to my setup.

It's a bit weird... 2F is clearly the code for the solidus slash... but the blocks around it are unprintable characters / characters not available in the font (they could be anything, even a 00 null character)... you should probably find out what those are...

Try viewing the payload through Home Assistant in developer services and copying and pasting into Notepad++ as I said.  You'll have a better idea of what exactly the problem is.

trip5, 2024-08-18:
Did you ever work this out?

lubeda, 2024-12-29:
stale

closed 2024-12-29 by lubeda