lubeda/EspHoMaTriXv2 #190: I`m preparing a release

lubeda, 2023-12-30:
Hi there,
in the last time there were so many new features added (thanx @andrewjswan a) that it is now necessary to restructure the repo, clean it up and make the documentation easier to read and understand. I'm preparing it in the [2024.1.0-prerelease](https://github.com/lubeda/EspHoMaTriXv2/tree/2024.1.0-prerelease) branch.

So stay calm, until the new release (with no new features), I suggest using the 2023.9.1 branch.

See you soon.

trip5, 2023-12-30:
Happy New Year Lubeda!

andrewjswan, 2023-12-31:
Happy New Year!

fthiery, 2024-01-21:
Hi, I'm testing the pre release and the text supposed to be the date over the calendar icon is just a blinking exclamation mark. Is this issue the right place to report bugs ?

lubeda, 2024-01-21:
This is bug in your YAML. Somehow the display gets no time. Check your configuration and the log

fthiery, 2024-01-21:
I double checked and the ulanzi-easy.yaml file is 100% identical to the one from the install/ folder of this repo (i did create a secrets.yaml file containing the ota and wifi passwords -- you may want to update this in the Readme btw, because it still suggests editing the ulanzi-easy.yaml file directly).

Plus, i DO get the time (just not the day):
[image]

Logs:
```
INFO ESPHome 2023.12.8
INFO Reading configuration /config/esphome/ulanzi-easy.yaml...
INFO Updating https://github.com/lubeda/EspHoMaTriXv2@2024.1.0
WARNING 
WARNING !!!!This version (2023.9.1) has breaking changes!!!!
WARNING Please check the documentation and wiki https://github.com/lubeda/EspHoMaTriXv2
WARNING This will only work with esphome >= 2023.7.0
WARNING 
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Paris'
INFO Detected timezone 'Europe/Paris'
INFO Starting log output from 192.168.1.81 using esphome API
INFO Successfully connected to ulanzi @ 192.168.1.81 in 0.200s
INFO Successful handshake with ulanzi @ 192.168.1.81 in 0.048s
```

lubeda, 2024-01-21:
Can you edit the log-level to debug a post a new log?

```YAML
logger:
  level: DEBUG
```

I tried it on two devices, and it works for me. We have to investigate a bit.

fthiery, 2024-01-21:
Here is the DEBUG log:

```
[12:49:40][I][app:102]: ESPHome version 2023.12.8 compiled on Jan 21 2024, 12:42:29
[12:49:40][I][app:104]: Project Ulanzi.EHMTXv2 version 2024.1.0
[12:49:40][C][wifi:573]: WiFi:
[12:49:40][C][wifi:405]:   Local MAC: xxxxxxxxxxxx
[12:49:40][C][wifi:410]:   SSID: [redacted]
[12:49:40][C][wifi:411]:   IP Address: 192.168.1.81
[12:49:40][C][wifi:413]:   BSSID: [redacted]
[12:49:40][C][wifi:414]:   Hostname: 'ulanzi'
[12:49:40][C][wifi:416]:   Signal strength: -13 dB ▂▄▆█
[12:49:40][C][wifi:420]:   Channel: 11
[12:49:40][C][wifi:421]:   Subnet: 255.255.255.0
[12:49:40][C][wifi:422]:   Gateway: 192.168.1.254
[12:49:40][C][wifi:423]:   DNS1: 192.168.1.3
[12:49:40][C][wifi:424]:   DNS2: 192.168.1.254
[12:49:40][C][logger:439]: Logger:
[12:49:40][C][logger:440]:   Level: DEBUG
[12:49:40][C][logger:441]:   Log Baud Rate: 115200
[12:49:40][C][logger:443]:   Hardware UART: UART0
[12:49:40][C][i2c.arduino:053]: I2C Bus:
[12:49:40][C][i2c.arduino:054]:   SDA Pin: GPIO21
[12:49:40][C][i2c.arduino:055]:   SCL Pin: GPIO22
[12:49:40][C][i2c.arduino:056]:   Frequency: 50000 Hz
[12:49:40][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[12:49:40][I][i2c.arduino:069]: Results from i2c bus scan:
[12:49:40][I][i2c.arduino:075]: Found i2c device at address 0x44
[12:49:40][I][i2c.arduino:075]: Found i2c device at address 0x68
[12:49:40][C][ledc.output:166]:   LEDC Channel: 0
[12:49:40][C][ledc.output:167]:   PWM Frequency: 1000.0 Hz
[12:49:40][C][ledc.output:168]:   Bit depth: 16
[12:49:40][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Left button'
[12:49:40][C][gpio.binary_sensor:016]:   Pin: GPIO26
[12:49:40][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Middle button'
[12:49:40][C][gpio.binary_sensor:016]:   Pin: GPIO27
[12:49:40][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Right button'
[12:49:40][C][gpio.binary_sensor:016]:   Pin: GPIO14
[12:49:40][C][light:103]: Light 'ulanzi Light'
[12:49:40][C][light:105]:   Default Transition Length: 1.0s
[12:49:40][C][light:106]:   Gamma Correct: 2.00
[12:49:40][C][template.switch:068]: Template Switch 'Auto-Adjust Brightness'
[12:49:40][C][template.switch:070]:   Icon: 'mdi:brightness-auto'
[12:49:40][C][template.switch:091]:   Restore Mode: restore defaults to ON
[12:49:40][C][template.switch:057]:   Optimistic: NO
[12:49:40][C][rtttl:029]: Rtttl
[12:49:40][C][status:034]: Status Binary Sensor 'ulanzi Status'
[12:49:40][C][status:034]:   Device Class: 'connectivity'
[12:49:40][C][sht3xd:033]: SHT3xD:
[12:49:40][C][sht3xd:034]:   Address: 0x44
[12:49:40][C][sht3xd:038]:   Update Interval: 60.0s
[12:49:40][C][sht3xd:040]:   Temperature 'ulanzi Temperature'
[12:49:40][C][sht3xd:040]:     Device Class: 'temperature'
[12:49:40][C][sht3xd:040]:     State Class: 'measurement'
[12:49:40][C][sht3xd:040]:     Unit of Measurement: '°C'
[12:49:40][C][sht3xd:040]:     Accuracy Decimals: 1
[12:49:40][C][sht3xd:041]:   Humidity 'ulanzi Relative Humidity'
[12:49:40][C][sht3xd:041]:     Device Class: 'humidity'
[12:49:40][C][sht3xd:041]:     State Class: 'measurement'
[12:49:40][C][adc:097]:   State Class: 'measurement'
[12:49:40][C][adc:097]:   State Class: 'measurement'
[12:49:40][C][adc:097]:   Unit of Measurement: 'lx'
[12:49:40][C][adc:097]:   Accuracy Decimals: 0
[12:49:40][C][adc:107]:   Pin: GPIO35
[12:49:40][C][adc:109]:  Attenuation: auto
[12:49:40][C][adc:142]:   Update Interval: 10.0s
[12:49:40][C][ds1307:022]: DS1307:
[12:49:40][C][ds1307:023]:   Address: 0x68
[12:49:40][C][ds1307:027]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[12:49:40][C][homeassistant.time:010]: Home Assistant Time:
[12:49:40][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[12:49:40][C][web_server:168]: Web Server:
[12:49:40][C][web_server:169]:   Address: ulanzi.local:80
[12:49:40][C][mdns:115]: mDNS:
[12:49:40][C][mdns:116]:   Hostname: ulanzi
[12:49:40][C][ota:097]: Over-The-Air Updates:
[12:49:40][C][ota:098]:   Address: ulanzi.local:3232
[12:49:40][C][ota:101]:   Using Password.
[12:49:40][C][api:139]: API Server:
[12:49:40][C][api:140]:   Address: ulanzi.local:6053
[12:49:40][C][api:144]:   Using noise encryption: NO
[12:49:40][C][EHMTXv2:2487]: EspHoMatriXv2 version: 2024.1.1
[12:49:40][C][EHMTXv2:2488]: Icons: 28 of 90
[12:49:40][C][EHMTXv2:2489]: Clock interval: 0 s
[12:49:40][C][EHMTXv2:2490]: Date format: %d.%m.
[12:49:40][C][EHMTXv2:2491]: Time format: %H:%M
[12:49:40][C][EHMTXv2:2492]: Interval (ms) scroll: 80
[12:49:40][C][EHMTXv2:2495]: Show day of week
[12:49:40][C][EHMTXv2:2503]: Weekstart: Monday
[12:49:40][C][EHMTXv2:2504]: Weekdays: SOMODIMIDOFRSA Count: 14
[12:49:40][C][EHMTXv2:2505]: Display: On
[12:49:40][C][EHMTXv2:2506]: Night mode: Off
[12:49:40][C][EHMTXv2:2507]: Weekday accent: Off
[12:49:40][C][EHMTXv2:2508]: Replace Time and Date: Off
[12:49:43][D][EHMTXv2:938]: oldest queue element is: 0/1
[12:49:43][D][EHMTXv2:128]: queue: icon clock: "day#0" for: 10.0 sec
[12:49:45][D][sensor:094]: 'ulanzi Illuminance': Sending state 377.93787 lx with 0 decimals of accuracy
[12:49:51][D][api:102]: Accepted 192.168.1.3
[12:49:51][D][api.connection:1121]: Home Assistant 2024.1.5 (192.168.1.3): Connected successfully
[12:49:51][D][time:045]: Synchronized time: 2024-01-21 12:49:51
[12:49:51][D][ds1307:101]: Write 11:49:51 2024-01-21  CH:OFF RS:3 SQWE:OFF OUT:OFF
[12:49:53][D][EHMTXv2:938]: oldest queue element is: 0/1
[12:49:53][D][EHMTXv2:128]: queue: icon clock: "day#0" for: 10.0 sec
[12:49:55][D][sensor:094]: 'ulanzi Illuminance': Sending state 375.73999 lx with 0 decimals of accuracy
```

As i said, i did not touch the yaml file, so the external_components part is:

```
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: "2024.1.0"
    refresh: 600s 
    components: [ ehmtxv2 ]   
```

lubeda, 2024-01-21:
OK, 
one thing you have to transtlate the weekdays from german to french , just add the weekdays line to your config.

```yaml
ehmtxv2: 
  weekdays: "DiLuMaMeJeVeSa"
```

And please try again.

fthiery, 2024-01-21:
Thanks for the suggestion, which i tried; no change unfortunately:

```
[13:09:59][C][EHMTXv2:2487]: EspHoMatriXv2 version: 2024.1.1
[13:09:59][C][EHMTXv2:2488]: Icons: 28 of 90
[13:09:59][C][EHMTXv2:2489]: Clock interval: 0 s
[13:09:59][C][EHMTXv2:2490]: Date format: %d.%m.
[13:09:59][C][EHMTXv2:2491]: Time format: %H:%M
[13:09:59][C][EHMTXv2:2492]: Interval (ms) scroll: 80
[13:09:59][C][EHMTXv2:2495]: Show day of week
[13:09:59][C][EHMTXv2:2503]: Weekstart: Monday
[13:09:59][C][EHMTXv2:2504]: Weekdays: DiLuMaMeJeVeSa Count: 14
[13:09:59][C][EHMTXv2:2505]: Display: On
[13:09:59][C][EHMTXv2:2506]: Night mode: Off
[13:09:59][C][EHMTXv2:2507]: Weekday accent: Off
[13:09:59][C][EHMTXv2:2508]: Replace Time and Date: Off
[13:09:59][D][sensor:094]: 'ulanzi Illuminance': Sending state 508.61026 lx with 0 decimals of accuracy
[13:09:59][I][EHMTXv2:2146]: set_brightness 137 => 53.73 %
[13:10:01][D][api:102]: Accepted 192.168.1.3
[13:10:01][D][api.connection:1121]: Home Assistant 2024.1.5 (192.168.1.3): Connected successfully
[13:10:01][D][time:045]: Synchronized time: 2024-01-21 13:10:01
[13:10:01][D][ds1307:101]: Write 12:10:01 2024-01-21  CH:OFF RS:3 SQWE:OFF OUT:OFF
[13:10:08][D][EHMTXv2:938]: oldest queue element is: 0/1
[13:10:08][D][EHMTXv2:128]: queue: icon clock: "day#0" for: 10.0 sec
```

[image]

Here is the diff between my device's `ulanzi-easy.yaml` and `install/ulanzi-easy.yaml`:

```
diff --git a/install/ulanzi-easy.yaml b/install/ulanzi-easy.yaml
index e42838d..db2b3e3 100644
--- a/install/ulanzi-easy.yaml
+++ b/install/ulanzi-easy.yaml
@@ -105,7 +105,9 @@ binary_sensor:
     name: "Right button"
 
 logger:
-  level: WARN
+  #level: WARN
+  level: DEBUG
+
 
 api:
   services:
@@ -231,6 +233,7 @@ ehmtxv2:
   time_component: ehmtx_time
   time_format: "%H:%M"
   date_format: "%d.%m."
+  weekdays: "DiLuMaMeJeVeSa"
   show_seconds: false
   default_font_id: px6_font
   default_font_yoffset: 7
```

lubeda, 2024-01-21:
Now i have the same bug! I think you have to use 2023.9.1 until i found the problem.

désolé ;-) I like spending my holiday time in France, mostly in the Bretagne and Normandy.

fthiery, 2024-01-21:
Oh, strange that you did not reproduce 1 hour ago... But i'm glad you do now. No problem i can wait ! Vielen Dank ;)

lubeda, 2024-01-21:
> strange
The icon name in the sample was wrong:

This will work:
```YAML
on_empty_queue:
    then:
      - lambda: |-
          id(rgb8x32).set_infotext_color(20,20,20,20,20,20,false,1);
          id(rgb8x32).icon_clock("calendar|day#0",1440,10,false);
```

fthiery, 2024-01-21:
Great, that did it, it works now, many thanks !!! 

I'm not sure it is intentional but you commited the DEBUG level.

Maybe another, issue, the get_status service call seems to do nothing for me (i open the log in esphome, then run the get_status service within HA -- i expect to see the status in the esphome logs, but i see nothing; what am i missing ?)

lubeda, 2024-01-21:
Hi, DEBUG level helps me if there are issues, for normal use it isn't necessary. The get_status service should write details to the ESP-log with the level INFO, so now you should see something in your log.

fthiery, 2024-01-21:
You are talking about the log that is viewed by clicking here, right ?

[image]

Then i call the service

[image]

But nothing:

[image]

lubeda, 2024-01-21:
Again, strange. I will investigate....

fthiery, 2024-01-21:
Actually, in DEBUG i see the status (i guess):

```
[19:36:24][I][EHMTXv2:1314]: status time: 21.1.2024 19:36
[19:36:24][I][EHMTXv2:1315]: status brightness: 61 (0..255)
[19:36:24][I][EHMTXv2:1316]: status date format: %d.%m.
[19:36:24][I][EHMTXv2:1317]: screen_pointer: 0
[19:36:24][I][EHMTXv2:1319]: current screen mode: 15
[19:36:24][I][EHMTXv2:1323]: status time format: %H:%M
[19:36:24][I][EHMTXv2:1324]: status date format: %d.%m.
[19:36:24][I][EHMTXv2:1325]: status display on
[19:36:24][I][EHMTXv2:1326]: status night mode off
[19:36:24][I][EHMTXv2:1327]: status weekday accent off
[19:36:24][I][EHMTXv2:1328]: status replace time and date off
[19:36:24][D][EHMTXv2:128]: queue: icon clock: "day#0" for: 10.0 sec
[19:36:24][I][EHMTXv2:1344]: queue: 23 empty slots
[19:36:24][W][component:214]: Component api took a long time for an operation (0.06 s).
[19:36:24][W][component:215]: Components should block for at most 20-30ms.
```

Is this expected (if so the readme should maybe mention it)?

lubeda, 2024-01-21:
Yes, that is the expected result.

fthiery, 2024-01-22:
Ok, got confused because the default install yaml was WARN (hiding the get_status log). I can contribute improvements to the readme, should I open a PR for it ?

lubeda, 2024-01-22:
Sure, every PR is welcome.

fthiery, 2024-01-22:
Done, i hope you'll like my changes !
https://github.com/lubeda/EspHoMaTriXv2/pull/202

popy2k14, 2024-01-29:
Hi. Testing now the pre-release and have a question about the night mode.

Looked at the code and saw that when night mode is on all screens are skipped which are not flagged as night mode.

How can I specify just "clock" as night mode screen and apply a color?

popy2k14, 2024-01-29:
Got it, there is night_mode_screens like so:

```
ehmtxv2: 
  night_mode_screens: [2]
```

Experimenting now.

popy2k14, 2024-01-29:
Working as expected, but how to correctly use the trigger?

This doesn't work:

`
ehmtxv2: 
  on_night_mode:
    then:
      lambda: |-
             id(rgb8x32)->set_clock_color(235, 0, 0);

andrewjswan, 2024-01-30:
> Working as expected, but how to correctly use the trigger?

You need to turn on the night mode `night_mode_on`, and this trigger should work.

popy2k14, 2024-01-30:
> You need to turn on the night mode `night_mode_on`, and this trigger should work.

I'll switched it on (tested with a switch and the exported service).

My definition:

```ehmtxv2:
  id: rgb8x32
  icons2html: true
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  time_format: "%H:%M"
  date_format: "%d.%m"
  show_seconds: false
  # Uncomment below if using the mateine font
  default_font_id: default_font
  special_font_id: default_font
  default_font_yoffset: 8
  special_font_yoffset: 8  
  # Comment out below if using mateine font
  #default_font_id: default_font
  #default_font_yoffset: 8
  #special_font_id: special_font
  #special_font_yoffset: 6
  # until here
  #boot_logo: "[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63519,63519,63519,63519,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63519,0,0,0,0,2016,0,0,0,0,0,0,0,0,0,0,31,0,0,0,0,0,0,0,0,0,63488,0,63488,0,0,0,63519,0,0,0,0,2016,2016,0,0,0,65514,0,65514,0,0,0,31,0,0,0,64512,0,0,64512,0,63488,63488,0,63488,63488,0,0,63519,63519,63519,0,0,2016,0,2016,0,65514,0,65514,0,65514,0,31,31,31,0,0,0,64512,64512,0,0,63488,63488,63488,63488,63488,0,0,63519,0,0,0,0,2016,0,2016,0,65514,0,65514,0,65514,0,0,31,0,0,0,0,64512,64512,0,0,0,63488,63488,63488,0,0,0,63519,63519,63519,63519,0,2016,0,2016,0,65514,0,65514,0,65514,0,0,0,31,31,0,64512,0,0,64512,0,0,0,63488,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]"

  clock_interval: 20      # show the clock at least each x seconds
  show_dow: false         # draw the day indicator on the bottom of the screen, defaults to true
  #show_date: true         # show the date for show_screen - show_clock seconds otherwise only shows the clock for show_screen seconds, defaults to true
  week_start_monday: true # default monday is first day of week, false = sunday
  night_mode_screens: [2]
  icons:
    - id: ha
      lameid: 7956
    - id: tempc
      lameid: 2422
    - id: plug
      lameid: 403
    - id: humidity
      lameid: 51764
    - id: co2
      lameid: 30662
    - id: weather_clear_night
      lameid: 53383
    - id: weather_cloudy
      lameid: 53384
    - id: weather_fog
      lameid: 12196  
    - id: weather_hail
      lameid: 53385  
    - id: weather_lightning
      lameid: 50231  
    - id: weather_lightning_rainy
      lameid: 49299  
    - id: weather_partlycloudy
      lameid: 53802  
    - id: weather_pouring
      lameid: 49300  
    - id: weather_rainy
      lameid: 2284  
    - id: weather_snowy
      lameid: 2289  
    - id: weather_snowy_rainy
      lameid: 49301  
    - id: weather_sunny
      lameid: 1246
    - id: weather_windy
      lameid: 17076  
    - id: weather_windy_variant
      lameid: 15618  
    - id: weather_exceptional
      lameid: 5464    
    - id: washing_machine_ready
      lameid: 26673
    - id: dryer_ready
      lameid: 48497
    - id: dishwasher_ready
      lameid: 47488
    - id: door_ringing
      lameid: 24800
    - id: fire
      lameid: 24873
    - id: water
      lameid: 24116
  on_empty_queue:
    then:
      lambda: |-
        id(rgb8x32)->clock_screen(10,10);
        id(rgb8x32)->date_screen(10,5);
  on_start_running:
     then:
       lambda: |-
          id(rgb8x32)->bitmap_screen("[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63519,63519,63519,63519,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63519,0,0,0,0,2016,0,0,0,0,0,0,0,0,0,0,31,0,0,0,0,0,0,0,0,0,63488,0,63488,0,0,0,63519,0,0,0,0,2016,2016,0,0,0,65514,0,65514,0,0,0,31,0,0,0,64512,0,0,64512,0,63488,63488,0,63488,63488,0,0,63519,63519,63519,0,0,2016,0,2016,0,65514,0,65514,0,65514,0,31,31,31,0,0,0,64512,64512,0,0,63488,63488,63488,63488,63488,0,0,63519,0,0,0,0,2016,0,2016,0,65514,0,65514,0,65514,0,0,31,0,0,0,0,64512,64512,0,0,0,63488,63488,63488,0,0,0,63519,63519,63519,63519,0,2016,0,2016,0,65514,0,65514,0,65514,0,0,0,31,31,0,64512,0,0,64512,0,0,0,63488,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]",1,10);
          id(rgb8x32)->bitmap_small("[2016,0,0,0,2016,0,0,0,2016,0,0,0,2016,0,0,0,2016,0,0,0,2016,0,0,0,0,2016,0,2016,0,31,31,0,0,0,2016,0,31,0,0,31,0,0,0,0,0,0,31,0,0,0,0,0,0,31,0,0,0,0,0,0,31,31,31,31]", "2023.x.y", 1, 10);
          id(rgb8x32)->clock_screen(10,10);
          id(rgb8x32)->date_screen(10,5);
  on_next_screen:
    - homeassistant.event:
        event: esphome.new_screen
        data_template:
          iconname: !lambda "return icon.c_str();"
          text: !lambda "return text.c_str();"
  on_night_mode:
    then:
      lambda: |-
        //if (x) {
            id(rgb8x32)->set_clock_color(235, 0, 0);
        //} else {
            //id(rgb8x32)->set_clock_color(255, 255, 255);
        //}

```

The switch:

```
  - platform: template

    name: "Night mode"
    id: switch_night_mode
    icon: mdi:sleep
    restore_mode: RESTORE_DEFAULT_OFF
    lambda: |-
      if (id(night_enable)) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      lambda: |-
        id(night_enable) = true;
        id(aab_enable) = false;
        id(rgb8x32)->set_night_mode_on();
        id(rgb8x32)->set_brightness(30);
        //id(rgb8x32)->set_clock_color(235, 0, 0);         
    turn_off_action:
      lambda: |-
        id(night_enable) = false;  
        id(aab_enable) = true;
        id(rgb8x32)->set_night_mode_off();
        //id(rgb8x32)->set_clock_color(255, 255, 255);
```

I can see that nightmode is enabled in the logs and also just the night_mode_screens are showed.

But the trigger is not called 

When I uncomment the set_clock_color in the switch, it's working but I want my customizations on the callback, for a cleaner for 

Thx

andrewjswan, 2024-01-30:
> But the trigger is not called

I checked everything, it should work. What if we add a little feedback?

```
  on_night_mode:
    then:
      lambda: |-
        ESP_LOGE("night", "night mode badabum");
        id(rgb8x32)->set_clock_color(235, 0, 0);
```

andrewjswan, 2024-01-30:
I think this switch can be simplified a little and one global variable removed.
```
  - platform: template

    name: "Night mode"
    id: switch_night_mode
    icon: mdi:sleep
    restore_mode: RESTORE_DEFAULT_OFF
    lambda: |-
        return id(rgb8x32)->night_mode;
    turn_on_action:
      lambda: |-
        id(rgb8x32)->set_night_mode_on();
        id(aab_enable) = !id(rgb8x32)->night_mode;
        id(rgb8x32)->set_brightness(30);
    turn_off_action:
      lambda: |-
        id(rgb8x32)->set_night_mode_off();
        id(aab_enable) = !id(rgb8x32)->night_mode;
```

popy2k14, 2024-01-30:
> I checked everything, it should work. What if we add a little feedback?
> 
> ```
>   on_night_mode:
>     then:
>       lambda: |-
>         ESP_LOGE("night", "night mode badabum");
>         id(rgb8x32)->set_clock_color(235, 0, 0);
> ```

Thx for checking.
Also checked the cop code and not found an difference to other triggers (like start running) which are working.

I'll added the badabum log 😄 but sadly, it's not triggered:

```
[13:36:05][D][switch:012]: 'Night mode' Turning ON.
[13:36:05][D][EHMTXv2:176]: night mode on
[13:36:05][I][EHMTXv2:2304]: set_brightness 30 => 11.76 %
[13:36:05][D][number:012]: 'Brightness': Sending state 30.000000
[13:36:06][D][number:012]: 'Brightness': Sending state 30.000000
[13:36:07][D][number:012]: 'Brightness': Sending state 30.000000
...
```

Is it possible for you to test this on your side?

Thx

andrewjswan, 2024-01-30:
> Also checked the cpp code and not found an difference to other triggers (like start running) which are working.

Same thing, I checked everything I could and couldn’t find a reason why it might not work, it’s very strange. 
Maybe @lubeda has some ideas.

> Is it possible for you to test this on your side?

I'll try to check today, but I can't promise it, I think we need open a new Issue.

andrewjswan, 2024-01-30:
Look like i found, `on_show_display` and `on_night_mode` not work ... and `on_next_screen` look like to ...

popy2k14, 2024-01-30:
Thx for checking again and your results.
Maybe it has Todo with:

- python registering of the trigger function
- on_night_mode has an argument?

Thx

andrewjswan, 2024-01-30:
> Thx for checking again and your results.

I just don't know python at all, so it's a bit difficult.

`on_night_mode` has a current status argument, name `state`, the same as `on_show_display`.

I don’t even know about Issue, if the fix works, then it’s probably not necessary.

popy2k14, 2024-01-30:
I am in the same boat, don't know much about python and esphome ecosystem.

Tried without boolean Argument,
And it doesn't work.
Any ideas?

andrewjswan, 2024-01-30:
> Tried without boolean Argument,
> And it doesn't work.

This not work?
```
  on_night_mode:
    then:
      lambda: |-
        ESP_LOGE("night", "night mode badabum");
        id(rgb8x32)->set_clock_color(235, 0, 0);
```

popy2k14, 2024-01-31:
> > Tried without boolean Argument,
> > And it doesn't work.
> 
> This not work?
> 
> ```
>   on_night_mode:
>     then:
>       lambda: |-
>         ESP_LOGE("night", "night mode badabum");
>         id(rgb8x32)->set_clock_color(235, 0, 0);
> ```

Yes

andrewjswan, 2024-01-31:
> Yes

Those the changes didn't help? True, they are not yet in the main branch, I think you need to check with this configuration.
```
external_components:
  - source:
      type: git
      url: https://github.com/andrewjswan/EspHoMaTriXv2/
      ref: "2024.1.0-Fix_for_on_show_display_on_night_mode_on_next_screen_triggers"
    refresh: 60s 
    components: [ ehmtxv2 ]   
```

popy2k14, 2024-01-31:
Sorry, I meant I tried my own fotk hack version, which did not work.

Tried now your branch and "badabum" 😊
It's working now, thx.

```
[10:53:51][D][switch:012]: 'Night mode' Turning ON.
[10:53:51][D][EHMTXv2:176]: night mode on
[10:53:51][E][night:186]: night mode badabum
[10:53:51][D][EHMTXv2:878]: default clock color r: 235 g: 0 b: 0
[10:53:51][I][EHMTXv2:2304]: set_brightness 30 => 11.76 %
[10:53:51][D][switch:055]: 'Auto-Adjust Brightness': Sending state OFF
[10:53:51][D][switch:055]: 'Night mode': Sending state ON
[10:53:52][D][number:012]: 'Brightness': Sending state 31.000000
```

andrewjswan, 2024-01-31:
> It's working now, thx.

Waiting for @lubeda to merge this PR to the 2024.1.0 branch.

andrewjswan, 2024-01-31:
> It's working now, thx.

Merged ...

popy2k14, 2024-01-31:
I have now set screenmode 2 (clock) as night_mode_screens, it's working as expected.
On migration my HA automations from 2023.7.1, i stumbled accross my fire, water & ringing alarm icon_screens.

When night mode is on, they dont show up now (also with force_screen), which isn't that great.

Any chance to get a force_screen flag to temorarily override night_mode_screens and show the forced screen?

thx

popy2k14, 2024-01-31:
Another thing.
I was updating my clock constantly OTA the last days without an issue.
Since i switched 2024.1.0, sometimes it doesn't come up and is dead.
I have to reflash it over serial to get it working again.

andrewjswan, 2024-01-31:
> Since i switched 2024.1.0, sometimes it doesn't come up and is dead.

When I debug code, I often modify it using OTA, this has never happened before.

popy2k14, 2024-01-31:
> 
> or to
> 
> ```
> night_mode_screens: [2,16]
> ```
> 
> and use `alert_screen {"icon_name","text", "screen_time", "default_font", "r", "g", "b"}`

Thx using alert_screen now because I can seperaten of important alerts and not.

popy2k14, 2024-01-31:
> > Since i switched 2024.1.0, sometimes it doesn't come up and is dead.
> 
> When I debug code, I often modify it using OTA, this has never happened before.

Yeah, I think an issue with the hardware...

andrewjswan, 2024-01-31:
> Thx using alert_screen now because I can seperaten of important alerts and not.

`night_mode_screens: [2, 5, 16]` ?

popy2k14, 2024-01-31:
All good regarding the night_mode_screens :-)

A question regarding the function:

```
clock_screen(int lifetime, int screen_time, bool default_font, int r, int g, int b)
```

I have this in my config, because otherwise no clock is shown. That changed on newer releases.

```
on_empty_queue:

    then:
      lambda: |-
        id(rgb8x32)->clock_screen(10,10);
        id(rgb8x32)->date_screen(10,5);
  on_start_running:
     then:
       lambda: |-
          id(rgb8x32)->clock_screen(10,10);
          id(rgb8x32)->date_screen(10,5);
```

I noticed after some time the clock and date is gone.

Sure 😁 lifetime is set to 10 minutes.

Can lifetime be set to "always show clock&date"?
Or do I have to use esphome timers to add it again.

Thx

popy2k14, 2024-01-31:
I have now done this as a workaround:

```
on_next_screen:

      then:
        lambda: |-
            //workaround clock disapears after lifetime

            id(rgb8x32)->clock_screen(1440,10);
            id(rgb8x32)->date_screen(1440,5);
```
I think it's an overkill and there should be (or is?) an lifetime infinity value like -1.

andrewjswan, 2024-01-31:
> So queue is not empty and no clock is shown.

Look https://github.com/lubeda/EspHoMaTriXv2/issues/93

popy2k14, 2024-01-31:
> > So queue is not empty and no clock is shown.
> 
> Look #93

That's exactly my issue!
I don't see permanent screens in the documentation, so I am assuming that is not implemented?
Should I'll open an issue for that?

My workaround for now:

```
#workaround that clock doesnt disappear after lifetime

interval:
  - interval: 720min
    then:
      lambda: |-
        ESP_LOGI("clock/date", "set clock and date screen!");
        id(rgb8x32)->clock_screen(1440,10);
        id(rgb8x32)->date_screen(1440,5);
```

That should keep the clock alive.

andrewjswan, 2024-01-31:
The Issue was not opened, apparently everyone decided it similar to your solution, I recreate it once a day.

Whether to open it or not is up to you.

popy2k14, 2024-01-31:
Done see #208

popy2k14, 2024-02-02:
As stated, since I am on 2024.1.0 the clock crashes 50% after OTA (black screen) and needs a serial reflash to bring it back to live. This happens on my two clocks. Two times also just a few minutes after flash. The WAF is currently not so high, so I reverted  to V1, and all runs fine again.

andrewjswan, 2024-02-02:
> This happens on my two clocks.

I have Ulanzy TC001 and no problem? In you clock Esp32? or 8266? Maybe decrease icon count and check?

popy2k14, 2024-02-04:
I have two ulanzi's with ESP32 and both showed the issue.
It's not just an Software crash.
And cold start (unplug and replug) does NOT solve the issue.
As stated the clock's are dead! I have to reflash them.

Sorry the WAF was very low, i am now back at v1 since two days, and they are running flawlessly.
Currently cant help debug any further.

andrewjswan, 2024-02-04:
Yesterday I flashed Ulanzi without any problems.

andrewjswan, 2024-02-15:
Look like we need 2024.2.0 release :)

andreasbrett, 2024-02-29:
Looks like we need a 2024.3.0 release ;-P

lubeda, 2024-02-29:
OK, I released a very but didn't improve the documentation like I had planned. :-( Too much work too few time!

**Thanks for your support!**

riker09, 2024-03-01:
You (and all the other contributors) are doing an amazing job. Thanks so much for all your hard work on this helpful software.

[7 more replies]