lubeda/EspHoMaTriXv2 #203: [QUESTION] Remove/Change boot animation

popy2k14, 2024-01-27:
# Question

Sometimes my ulanzi's are rebooting and showing the fancy boot animation.
One is in a bedroom and i want to get rid of the boot animation.

There is a "boot_logo", but that's just showing a logo after boot, not what i want.

Is there such a feature?

## Additional information

- used Hardware:
  - ESP32
  - EspHoMaTriXv2 version: [e.g. 2023.7.0]

lubeda, 2024-01-27:
Hi,
I don't know in detail, because i'm working on an upcoming release wich will be a bit different in details. But what about setting the boot_logo to all "0" (all pixels black) an calling the service

```yaml
service: esphome.ulanzi_bitmap_screen
data:
  icon: >-
   "[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63519,63519,63519,63519,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63519,0,0,0,0,2016,0,0,0,0,0,0,0,0,0,0,31,0,0,0,0,0,0,0,0,0,63488,0,63488,0,0,0,63519,0,0,0,0,2016,2016,0,0,0,65514,0,65514,0,0,0,31,0,0,0,64512,0,0,64512,0,63488,63488,0,63488,63488,0,0,63519,63519,63519,0,0,2016,0,2016,0,65514,0,65514,0,65514,0,31,31,31,0,0,0,64512,64512,0,0,63488,63488,63488,63488,63488,0,0,63519,0,0,0,0,2016,0,2016,0,65514,0,65514,0,65514,0,0,31,0,0,0,0,64512,64512,0,0,0,63488,63488,63488,0,0,0,63519,63519,63519,63519,0,2016,0,2016,0,65514,0,65514,0,65514,0,0,0,31,31,0,64512,0,0,64512,0,0,0,63488,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]" 
  lifetime: 2
  screen_time: 16
```

With the upcoming 2024.1.0 there will be a in build animation or a logo which you can choose as you wish.

popy2k14, 2024-01-27:
thx, but i cant do that a boot time to overwrite it.
It has to be online, which, at boot, it isn't.

Just to be sure, we are talking about the rainbow animation which goes from left to right after you plug your power in or restart the ESP otherwise.

No problem, i will wait for the new release and hope i can disable the boot animation.

andrewjswan, 2024-01-30:
> Just to be sure, we are talking about the rainbow animation which goes from left to right after you plug your power in or restart the ESP otherwise.

In 2024.1.0, you can change this animation to your own, and specify yours as completely empty. Then it will just be a black screen.

popy2k14, 2024-01-30:
Thx. Working with 2024.1.0-prerelease and defining it to black.

Closing because already fixed 
Sorry für bringing up this and it's already fixed in the newest pre-release.

closed 2024-01-30 by popy2k14