lubeda/EspHoMaTriXv2 #204: [QUESTION] Force clock screen (without date)

popy2k14, 2024-01-27:
# Question

I have the following HA automation when someone enters the room:

- display on
- del screen *
- switch_date off (see yaml code)

So after entering the room, the time was always shown to the user on v1.
Have now migrated from EspHoMaTriXv1 to v2.

An this is not working, sometimes it shows the date instead of the time.

How can i force the clock on v2?
I know there is a "force_screen" service, but what's the icon_name of the clock?

yaml:
`
  - platform: template
    name: "Datum"
    id: switch_date
    icon: mdi:calendar
    restore_mode: RESTORE_DEFAULT_ON
    lambda: |- 
      if (id(date_enable)) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      lambda: |-
        if (!id(night_enable)) {
          id(date_enable) = true;
          id(rgb8x32)->set_show_date(true);
        }
    turn_off_action:
      lambda: |-
        if (!id(night_enable)) {
          id(date_enable) = false;  
          id(rgb8x32)->set_show_date(false);
        }
`

## Additional information

- used Hardware:
  - ESP32
  - EspHoMaTriXv2 version: [e.g. 2023.7.0]

lubeda, 2024-01-27:
You can "force" screens, see [Mode overview](https://github.com/lubeda/EspHoMaTriXv2/tree/main#modes)
```

service: esphome.ulanzi_force_screen
data:
  icon_name: "*"
  mode: 2
```

popy2k14, 2024-01-27:
thx for the hint, tried now with:

`
service: esphome.pixel_clock_wz_force_screen
data:
  icon_name: "*"
  mode: 2
`

as you suggested.
Sadly i does'nt work.
When i send the above force_screen cmd when the clock is on another screen, i just ignores it.
But when i send a force to another icon_name, it works:

`
service: esphome.pixel_clock_wz_force_screen
data:
  icon_name: tempc
  mode: 5
`

Any hint's what i am doing wrong ?
thx

lubeda, 2024-01-27:
This is probably a bug

popy2k14, 2024-01-27:
Ok thx.
I have changed the title to a [BUG].

andrewjswan, 2024-01-29:
In my opinion, this has already been fixed in 2023.9.1 and, accordingly, in 2024.0.1

popy2k14, 2024-01-29:
@andrewjswan thx for the hint. I can confirm that 2023.9.1 fixed this issue.

Because of upcoming release, I am closing this.