lubeda/EspHoMaTriXv2 #208: [FEATURE REQUEST] Permanent screens

popy2k14, 2024-01-31:
# Feature Request

With upcoming 2024.1.0 there is no direct way to set an screen (like clock and date) permanent (without a lifetime).

In my case the primary function of the device is a clock, and so it should always display a clock (like it did before).

### Is your feature request related to a problem?

See discussion here: #93 

### Describe the solution / feature you'd like

An setting with permanent screen modes which ignores lifetime and keeps in the queue forever...

Or...

An new parameter to every function like "permanent=true". Is this set , the same happens as above.

### Describe alternatives you've considered

Not solution but a workaround:

```
#workaround that clock doesnt disappear after lifetime

interval:
  - interval: 720min
    then:
      lambda: |-
        ESP_LOGI("clock/date", "set clock and date screen!");
        id(rgb8x32)->clock_screen(1440,10);
        id(rgb8x32)->date_screen(1440,5);
```

This should keep the clock alive.

popy2k14, 2024-05-01:
I personally find it rather irritating that you, as a user have the possibility to have an empty/black screen and must serve the clock as content! It's a clock and should always show the clock (as it did in previous versions).

I know the feature was added because you as a user have more possibilitys in features,
But i find it irritating for most of us users.

Looked at the code and there is an defen already so clock_screens does'nt get remove.
Maybe as an beginning fpr https://github.com/lubeda/EspHoMaTriXv2/issues/208 just add this define as default for clock and date_screen and add and config option to set other screens also as static later.

lubeda, 2024-05-01:
Well, the code is a mess, an array for the queue elements is a mess, the class design for the queue is a mess, the interface to the esphome service calls are messy ....

I am using only 3.56% of the capabilities of EspHoMaTriXv2, the rest is Schnickschnack (German for "bells and whistles")  to me.
I am also truly interested in using the display as a voice assistant satellite, which implies ESP32 and an optimized timing for resource sharing.

So i like the idea of a permanent screen, but only with a complete rethinking of the way to control the display (homeassistant service calls) and the internal structure. The last pull request from @andrewjswan shows the problem, he added a helpful feature, but how he had to pass the parameter in a "special" way. (a numeric parameter as part of a string) which is horror/abuse.... 

I will not spend my time on optimizing the code since it is working for me. But if you can provide PRs based on the "latest" branch feel free... I will support you

popy2k14, 2024-05-01:
I don't looked at the code so deeply but that doesn't sound good 😞

Sorry, don't have much time to optimize it either and it also works for my needs.

lubeda, 2024-11-01:
a workaround is to use the on_empty_queue event to add the screen at the end of the lifetime

closed 2024-11-01 by lubeda