lubeda/EspHoMaTriXv2 #210: Date color

pepe59, 2024-02-03:
I don't know if it's a bug but if I set the clock and date color, after 24 hours the time color is correct but the date color goes back to white.
I have to call service again and it's fine for another 24 hours.

```
service: esphome.ulanzi_set_clock_color
data:
  r: 255
  g: 140
  b: 0
```

lubeda, 2024-02-03:
Hi, 
I think this is no bug. This is more like a hidden automatism. Try changing your YAML like this:

```yaml
on_empty_queue:
    then:
      - lambda: |-
          id(rgb8x32).set_infotext_color(20,20,20,20,20,20,false,1);
          id(rgb8x32).icon_clock("calendar|day#0",1440,10,false,255,140,0);
```
So the clock color isn't overwritten by the default color.

pepe59, 2024-02-03:
Yes, the clock color is not overridden by the default color.
Date only.
I am using the code from the ulanzi-easy.yaml documentation for the Ulanzi TC001 device.
If I add your recommended yaml to the end of ulanzi-easy.yaml

```
  on_next_screen:
    - homeassistant.event:
        event: esphome.new_screen
        data_template:
          iconname: !lambda "return icon.c_str();"
          text: !lambda "return text.c_str();"

  on_empty_queue:
    then:
      - lambda: |-
          id(rgb8x32).set_infotext_color(20,20,20,20,20,20,false,1);
          id(rgb8x32).icon_clock("calendar|day#0",1440,10,false,255,140,0);
```
an error occurs in the code
`[on_empty_queue] is an invalid option for [ehmtxv2]. Did you mean [allow_empty_screen]?`

pepe59, 2024-02-05:
Installed as per documentation
```
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: main
    refresh: 60s 
    components: [ ehmtxv2 ]   

esphome:
  comment: "EHMTXv2 from LuBeDa"
  name: $devicename 
  project:
    name: "Ulanzi.EHMTXv2"
    version: "2.0.0"
  on_boot:
    then:
      - ds1307.read_time:
```

andrewjswan, 2024-02-05:
`on_empty_queue` added in 2023.8.0 - https://github.com/lubeda/EspHoMaTriXv2/tree/2023.8.0?tab=readme-ov-file#esphomatrix-202380
Try Pre Release version - https://github.com/lubeda/EspHoMaTriXv2/tree/2024.1.0-prerelease

pepe59, 2024-02-05:
Can I find sample code somewhere for the Ulanzi screen where it's just the clock alternating with the date and days of the week like it was in previous versions ?

andrewjswan, 2024-02-06:
What was in the previous version?  Everything that was available in the old version is available here.

andrewjswan, 2024-02-06:
> I don't know how to remove it or change it to another one. I don't like it very much

Only clock with icon:
```
  on_start_running:
     then:
       lambda: |-
          id(rgb8x32)->icon_clock("another_one_icon", 1440, 10, false, 255, 140, 0);

  on_empty_queue:
    then:
      lambda: |-
          id(rgb8x32)->icon_clock("another_one_icon", 1440, 10, false, 255, 140, 0);
```

Clock and Date with icon:
```
  on_start_running:
     then:
       lambda: |-
          id(rgb8x32)->icon_clock("another_one_icon", 1440, 10, false, 255, 140, 0);
          id(rgb8x32)->icon_date("another_one_icon", 1440, 5, true);

  on_empty_queue:
    then:
      lambda: |-
          id(rgb8x32)->icon_clock("another_one_icon", 1440, 10, false, 255, 140, 0);
          id(rgb8x32)->icon_date("another_one_icon", 1440, 5, true);
```

Or without icon:
```
  on_start_running:
     then:
       lambda: |-
          id(rgb8x32)->clock_screen(1440,10);
          id(rgb8x32)->date_screen(1440,5);

  on_empty_queue:
    then:
      lambda: |-
          id(rgb8x32)->clock_screen(1440,10);
          id(rgb8x32)->date_screen(1440,5);
```

> That's why I can't tell if the color of the date will change after 24 hours.

```
id(rgb8x32)->set_clock_color(255, 0, 0); // Red color
id(rgb8x32)->set_clock_color(0, 255, 0); // Green color
id(rgb8x32)->set_clock_color(0, 0, 255); // Blue color
```

pepe59, 2024-02-06:
Thank you very much. I wandered a bit in the documentation. Using sample codes, I now understood how to configure.
Everything works perfectly.
I will close the issue.

closed 2024-02-06 by pepe59