lubeda/EspHoMaTriXv2 #212: Strapping GPIO15 prevents OTA flashing from working

noiob, 2024-02-16:
I have a fairly new Ulanzi smart clock (might be a newer hardware revision? maybe that's why it hasn't come up yet) and I've struggled to configure the easy_ulanzi.yaml until I realized that the warning about GPIO15 being a boot selector pin was actually correct, removing the pin definition made it work.

I'd propose just removing anything buzzer-related from the default config or at least adding a warning. I don't think not having the buzzer connected  is a big loss to anyone since it's pretty quiet and sounds awful.

andrewjswan, 2024-02-16:
When assembling the Ulanzi firmware, I have this warning, but everything works, including the buzzer.

lubeda, 2024-02-21:
With esphome 2024.2.0 I experience the same!

andrewjswan, 2024-02-21:
```
INFO ESPHome 2024.2.0
INFO Reading configuration /config/esphome/hall_pixel_clock.yaml...
WARNING 
WARNING !!!!This version (2024.1.0) has breaking changes!!!!
WARNING Please check the documentation and wiki https://github.com/lubeda/EspHoMaTriXv2
WARNING This will only work with esphome >= 2023.7.0
WARNING 
WARNING 'esp_hall_pixel_clock': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
...
Compiling .pioenvs/esp_hall_pixel_clock/src/main.cpp.o
Linking .pioenvs/esp_hall_pixel_clock/firmware.elf
RAM:   [=         ]  14.6% (used 47920 bytes from 327680 bytes)
Flash: [========= ]  92.4% (used 1696177 bytes from 1835008 bytes)
Building .pioenvs/esp_hall_pixel_clock/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/esp_hall_pixel_clock/firmware.bin"], [".pioenvs/esp_hall_pixel_clock/firmware.elf"])
Wrote 0x1af840 bytes to file /data/.build/esp_hall_pixel_clock/.pioenvs/esp_hall_pixel_clock/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 38.80 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.11
INFO Uploading /data/./.build/esp_hall_pixel_clock/.pioenvs/esp_hall_pixel_clock/firmware.bin (1701952 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 20.80 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.11 using esphome API
INFO Successfully connected to esp_hall_pixel_clock @ 192.168.1.11 in 3.092s
INFO Successful handshake with esp_hall_pixel_clock @ 192.168.1.11 in 0.137s
[19:56:46][I][app:102]: ESPHome version 2024.2.0 compiled on Feb 21 2024, 19:55:46
```

andrewjswan, 2024-03-08:
It’s strange, on 2024.2.0 and 2024.2.1 the firmware via OTA went through without problems for me (I didn’t remove the buzzer), I just checked Rainbow (I flashed it 10 times), but yesterday on 2024.2.2 I already received the same error when trying to OTA , just reboot Esp.

lubeda, 2024-12-29:
use a newer ehmtx version without the buzzer

closed 2024-12-29 by lubeda