lubeda/EspHoMaTriXv2 #219: Unable to compile latest version - missing libmagic???

trizmark, 2024-03-14:
# Question

I've been running this FW on my ulanzi for a while and last time I compiled it was back in Sept. (ESPHome 2023.8.3).
I wanted to add a couple of extra icons to the display, but for some reason I'm unable to compile it at all. I tried both on my Win and my Mac machine, but no luck.

Any idea what am I missing?

I have libmagic and pylibmagic installed and no amount of searching on the web yielded any useful results...

```
% pip install libmagic             
Requirement already satisfied: libmagic in /opt/homebrew/lib/python3.11/site-packages (1.0)
% pip install pylibmagic
Requirement already satisfied: pylibmagic in /opt/homebrew/lib/python3.11/site-packages (0.5.0)
```

## Additional information

- used Hardware:
  Ulanzi TC001

  - EspHoMaTriXv2 version: 2024.3.0

## Logs

```text
ERROR Unable to import component ehmtxv2:
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.11/site-packages/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/Users/zmarkella/Documents/Git_repos/ESPHomeConfigs/ulanzi/.esphome/external_components/d62ebc04/components/ehmtxv2/__init__.py", line 10, in <module>
    import esphome.components.image as espImage
  File "/opt/homebrew/lib/python3.11/site-packages/esphome/components/image/__init__.py", line 10, in <module>
    from magic import Magic
  File "/opt/homebrew/lib/python3.11/site-packages/magic/__init__.py", line 209, in <module>
    libmagic = loader.load_lib()
               ^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/magic/loader.py", line 49, in load_lib
    raise ImportError('failed to find libmagic.  Check your installation')
ImportError: failed to find libmagic.  Check your installation
ERROR Unable to import component animation:
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.11/site-packages/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/opt/homebrew/lib/python3.11/site-packages/esphome/components/animation/__init__.py", line 5, in <module>
    import esphome.components.image as espImage
  File "/opt/homebrew/lib/python3.11/site-packages/esphome/components/image/__init__.py", line 10, in <module>
    from magic import Magic
  File "/opt/homebrew/lib/python3.11/site-packages/magic/__init__.py", line 209, in <module>
    libmagic = loader.load_lib()
               ^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/magic/loader.py", line 49, in load_lib
    raise ImportError('failed to find libmagic.  Check your installation')
ImportError: failed to find libmagic.  Check your installation
ERROR Unable to import component image:
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.11/site-packages/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/opt/homebrew/lib/python3.11/site-packages/esphome/components/image/__init__.py", line 10, in <module>
    from magic import Magic
  File "/opt/homebrew/lib/python3.11/site-packages/magic/__init__.py", line 209, in <module>
    libmagic = loader.load_lib()
               ^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/magic/loader.py", line 49, in load_lib
    raise ImportError('failed to find libmagic.  Check your installation')
ImportError: failed to find libmagic.  Check your installation
```

## Additional context

Add any other context about the problem here.

trizmark, 2024-03-27:
Thanks for the tip, but that doesn't explain why I get the exact same issue on my Windows machine.

andrewjswan, 2024-06-21:
> Thanks for the tip, but that doesn't explain why I get the exact same issue on my Windows machine.

```
pip install "pillow==10.2.0"
pip install libmagic
pip install python-magic-bin
```

trip5, 2024-06-21:
Out of curiosity, why not run ESPHome in Docker?  All the dependencies are part of the image... actually, as you might guess from andrewjswan's comment, there's on ongoing issue regarding ESPHome being stuck with an outdated version of Pillow...

Also, this definitely isn't an issue with EHMTXv2... it's an ESPHome issue.  Try this thread: https://github.com/esphome/issues/issues/5427

andrewjswan, 2024-06-21:
Docker on Windows sounds funny, there is nothing complicated about installing ESPHome in Python.

trip5, 2024-06-21:
If you say so but this thread and that thread say otherwise.  Literally takes seconds to update ESPHome, no futzing about with versions or Python environments.  And it's speed is close enough to running on bare-metal.

And yes, lol it runs in Windows on the Linux Subsystem (which the Docker install painlessly updates).  It's a bit odd compared to Linux installs as it can also run Windows containers (I have no idea about that) but I had to run it on my work notebook because I keep ESPHome devices at work.  It's a bit slow though, probably because Windows itself is using 1/4 of the CPU cycles.

andrewjswan, 2024-06-21:
> If you say so but this thread and that thread say otherwise. Literally takes seconds to update ESPHome, no futzing about with versions or Python environments. And it's speed is close enough to running on bare-metal.

In Python it's all the same, I haven't had any problems for over 4 years. Everything is simple and hassle-free. To be precise I have 3 ESPHome, one in HA OS, and two in Windows in Python, no problems...

> And yes, lol it runs in Windows on the Linux Subsystem (which the Docker install painlessly updates). It's a bit odd compared to Linux installs as it can also run Windows containers (I have no idea [...]

It's easy when it's easy. But I have Windows 11 and Core 2 Duo, and when installing Docker, it just says, "Sorry, but I won't install on this hardware, much less work with it." But Python works.

guillempages, 2024-09-01:
https://github.com/esphome/esphome/pull/7365 should fix this issue

lubeda, 2024-11-01:
this is not direct related to ehmtx, its more a platform thing.

closed 2024-11-01 by lubeda