lubeda/EspHoMaTriXv2 #220: [QUESTION] Is my Config wrong?

7R0J4Ner, 2024-03-19:
Hello everyone

I have finally managed to update my display (custom) to v2. Not easy for me as a layman, as a lot of the documentation seems to be outdated. But it works now. ( I think?)

If i send the following from Home Assistant:

```
service: esphome.matrix3_icon_screen
data:
  default_font: true
  icon_name: ten
  screen_time: 10
  lifetime: 1
  r: 200
  b: 200
  g: 200
  text: Test
``` 

Text and icon appear as desired for 10 seconds and then switch back to the clock.
After that, the text and icon appear 4 more times within the next 90 seconds or so.

What do I have to do so that text & icon are only displayed once for 10 seconds?

In addition, the example in the WIKI "[display timer](https://github.com/lubeda/EspHoMaTriXv2/wiki/display-timer)" does not seem to work.
If I set the timer to 60 seconds, only "59" is displayed and the time does not run down.

Can anybody help me?
Thank you so much

## Config
This is my Config:
[Config.txt](https://github.com/lubeda/EspHoMaTriXv2/files/14648015/Config.txt)

## Additional information
  - Hardware: ESP32
  - EspHoMaTriXv2 version: [2024.3.0]
  
## Logs
Log when Calling the icon_screen:
[logs_matrix3_logs.txt](https://github.com/lubeda/EspHoMaTriXv2/files/14647787/logs_matrix3_logs.txt)

andrewjswan, 2024-03-19:
If you need to show this screen once, you can use `alert_screen`, or set `lifetime: 0` and then call `force_screen`.
This is true for **2024.3.0** https://github.com/lubeda/EspHoMaTriXv2/tree/2024.1.0-prerelease

7R0J4Ner, 2024-03-20:
Brilliant, thank you very much!
`alert_screen` is nowhere found in the Docs, this works perfect.

If anyone is interested, with this version of the automation in Home Assistant, the timer from the example now works:
```
alias: "Matrix: Timer test"
description: Zeigt einen Timer auf der Matrix an
trigger:
  - platform: state
    entity_id:
      - timer.eier_timer
    to: active
condition:
  - condition: state
    entity_id: timer.eier_timer
    state: active
action:
  - repeat:
      sequence:
        - service: esphome.matrix3_alert_screen
          data:
            default_font: true
            icon_name: ten
            screen_time: 1
            r: 200
            b: 200
            g: 50
            text: >-
              {{ (state_attr('timer.eier_timer', 'finishes_at') | as_datetime -
              now()).total_seconds() | int(0) }}
        - delay:
            hours: 0
            minutes: 0
            seconds: 1
            milliseconds: 0
      while:
        - condition: state
          entity_id: timer.eier_timer
          state: active
mode: single
```

7R0J4Ner, 2024-03-20:
Brilliant, it works perfectly, thank you very much!

If anyone is interested, with this version of the automation in Home Assistant, the timer from the example now works:

```
alias: "Matrix: Timer test"
description: Zeigt einen Timer auf der Matrix an
trigger:
  - platform: state
    entity_id:
      - timer.eier_timer
    to: active
condition:
  - condition: state
    entity_id: timer.eier_timer
    state: active
action:
  - repeat:
      sequence:
        - service: esphome.matrix3_alert_screen
          data:
            default_font: true
            icon_name: ten
            screen_time: 1
            r: 200
            b: 200
            g: 50
            text: >-
              {{ (state_attr('timer.eier_timer', 'finishes_at') | as_datetime -
              now()).total_seconds() | int(0) }}
        - delay:
            hours: 0
            minutes: 0
            seconds: 1
            milliseconds: 0
      while:
        - condition: state
          entity_id: timer.eier_timer
          state: active
mode: single

```

closed 2024-03-20 by 7R0J4Ner