lubeda/EspHoMaTriXv2 #224: [QUESTION] How to set night_mode_screens with included packages

popy2k14, 2024-03-31:
# Question

I have a setup like this:

```
substitutions:
  sub_night_modes_screens: "[2,16]"

#include ulanzi_tc001 yaml
packages:
  device_base: !include common/devices/ulanzi_tc001v2.yaml  
```

than in ulanzi_tc001v2.yaml  :

```
ehmtxv2:
  ...
  night_mode_screens: $sub_night_modes_screens
  ...
```  
 
Error on build:
  
```
  Unknown value '[2,16]', valid options are '1', '2', '3', '4', '5', '9', '10', '11', '12', '13', '14', '15', '16', '17', '18', '19', '20', '21', '22', '23', '24', '25', '26'.
  night_mode_screens: [2,16]
```

It seems the conversion from the string substitution fails.
When using:

```
ehmtxv2:
  ...
  night_mode_screens: [2,16]
  ...
```  

it works.

Any ideas how i can substitute the night screens? 

## Additional information

- used Hardware:
  - ESP32

  - EspHoMaTriXv2 version: 2024.4.0

## Logs

see above

andrewjswan, 2024-03-31:
> sub_night_modes_screens: "[2,16]"

Maybe:
```
sub_night_modes_screens: [2,16]
```
?

popy2k14, 2024-03-31:
Alread tried and sadly not working.
I think it's because substitutions are strings.

popy2k14, 2024-04-02:
@andrewjswan @lubeda any other hint's how i can change the config (with an included package) so that i can give different night mode screens into the package?

popy2k14, 2024-04-02:
How? Sorry but i dont get it.
I have two device yamls (see above) which everyone includes the device_base where the ehmtx lives.

Also tried with vars, like described here: https://github.com/esphome/esphome/pull/3602
But the issue is it has to be a list and will always interpreted as one variable like "[2,16]".

Is any other method like map?

Soory for being a little bit off topic here, but maybe this will also help others setup the component in the future.

andrewjswan, 2024-04-02:
Sorry, but i dont know :(

popy2k14, 2024-04-02:
No problem :-) i am currently trying any way i find on the net :-)

popy2k14, 2024-04-02:
Found a solution with packages.
Before, nothing worked. Tried substituions, vars, anchors, ...

Just using a second package for different night modes work.
Packages will be combined when have the same keys, like ehtmx -> night_mode_screens.
So here is my solution:

device1 yaml:

```
substitutions:
  name: "pixel-clock-wz"
  night_mode_brightness: "50"
  aab_min: "40"
  night_mode_r: "205"
 
esphome:
  name: $name 
  friendly_name: "Pixel Clock WZ"

#include ulanzi_tc001 yaml
packages:
  night_modes: !include common/devices/ulanzi_tc001v2_night_modes_wz.yaml 
  device_base: !include common/devices/ulanzi_tc001v2.yaml
```

device2 yaml:

```
substitutions:
  name: "pixel-clock-sz"
  night_mode_brightness: "20"
  aab_min: "30"
  night_mode_r: "235"
 
esphome:
  name: $name 
  friendly_name: "Pixel Clock SZ"

#include ulanzi_tc001 yaml
packages:
  #device_base: !include common/devices/ulanzi_tc001.yaml
  night_modes: !include common/devices/ulanzi_tc001v2_night_modes_sz.yaml 
  device_base: !include common/devices/ulanzi_tc001v2.yaml
```

ulanzi_tc001v2.yaml has the main yaml part:

```
substitutions:
  # Pin definition from https://github.com/aptonline/PixelIt_Ulanzi 
  battery_pin: GPIO34 
  ldr_pin: GPIO35 
  matrix_pin: GPIO32 
  left_button_pin: GPIO26 
  mid_button_pin: GPIO27 
  right_button_pin: GPIO14 
  scl_pin: GPIO22 
  sda_pin: GPIO21
  
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: 2024.4.1
    refresh: 1min
    components: [ehmtxv2]    
  
esphome:
  comment: "Ulanzi TC001"
[... 519 lines cut ...]
text_sensor:
  - <<: !include text_sensor/version.config.yaml
  - <<: !include text_sensor/wifi_info.config.yaml

button:
  - <<: !include button/restart.config.yaml
```

And finally the device specific package with different night_mode_screens:

ulanzi_tc001v2_night_modes_wz.yaml:

```
ehmtxv2:
  night_mode_screens: [2,5,16]
```

ulanzi_tc001v2_night_modes_sz.yaml:

```
ehmtxv2:
  night_mode_screens: [2,16]
```

Hope this helps someone :-)

popy2k14, 2024-04-02:
closing for now

closed 2024-04-02 by popy2k14