lubeda/EspHoMaTriXv2 #225: [BUG] LOOP & on_empty_queue does'nt get's triggered in night mode when "clock" screen expires

popy2k14, 2024-03-31:
# Bug report

## Describe the bug

I am using:

`  night_mode_screens: [2,16]
`

and when ill add an clock screen with 1 minute lifetime:

```
  on_start_running:
    then:
      lambda: |-
            id(rgb8x32)->clock_screen(1,true,235,0,0);

  on_empty_queue:
    then:
      lambda: |-
            id(rgb8x32)->clock_screen(1,true,235,0,0);
```

after boot trigger night mode within 1 minute:

```
switch:
  - platform: template
    name: "Night mode"
    id: switch_night_mode
    icon: mdi:sleep
    restore_mode: RESTORE_DEFAULT_OFF
    lambda: |-
        return id(rgb8x32)->night_mode;
    turn_on_action:
      lambda: |-
        id(rgb8x32)->set_night_mode_on();
    turn_off_action:
      lambda: |-
        id(rgb8x32)->set_night_mode_off();
```

Wait a minute until the lifetime is over and **"on_empty_queue" is not triggered.**

As a workarounf i am using clock lifetime of 1440 minutes and setting it again in an intervall of 720 minutes.

But the bad thing, it loops really fast:

**
[13:30:12][D][EHMTXv2:1079]: remove expired queue element: slot 0: mode: 2 icon_name: text: 
[13:30:12][D][EHMTXv2:1013]: oldest queue element is first: 0/2
[13:30:12][D][EHMTXv2:095]: empty slot
[13:30:12][D][EHMTXv2:1261]: on_next_screen trigger
[13:30:12][D][EHMTXv2:1164]: no screen to draw!
[13:30:12][D][EHMTXv2:1013]: oldest queue element is first: 0/2
[13:30:12][D][EHMTXv2:095]: empty slot
[13:30:12][D][EHMTXv2:1261]: on_next_screen trigger
[13:30:12][D][EHMTXv2:1164]: no screen to draw!
[13:30:12][D][EHMTXv2:1013]: oldest queue element is first: 0/2
[13:30:12][D][EHMTXv2:095]: empty slot
[13:30:12][D][EHMTXv2:1261]: on_next_screen trigger
[13:30:12][D][EHMTXv2:1164]: no screen to draw!
[13:30:12][D][EHMTXv2:1013]: oldest queue element is first: 0/2
[13:30:12][D][EHMTXv2:095]: empty slot
[13:30:12][D][EHMTXv2:1261]: on_next_screen trigger
[13:30:12][D][EHMTXv2:1164]: no screen to draw!
[13:30:12][D][EHMTXv2:1013]: oldest queue element is first: 0/2
[13:30:12][D][EHMTXv2:095]: empty slot**

## Additional information

- used Hardware:
  - ESP32 or ESP8266
  - EspHoMaTriXv2 version: [e.g. 2023.5.0]

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Configuration

```text
(optional) The YAML you used in epshome without any password
```

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Services calls

```text
(optional) The YAML of your service calls
```

andrewjswan, 2024-03-31:
> Wait a minute until the lifetime is over and **"on_empty_queue" is not triggered.**

Night mode simply skips this screen and does not take it into account and does not count its lifetime, but since it is there, then the queue is not empty. According to the log, it looks like this screen is not allowed for night mode, so it just quickly spins into emptiness.

popy2k14, 2024-03-31:
Got you.
So a new trigger like "on_empty_queue_night_mode" would be a solution?

If i do not take this into account my screen will be empty in night mode.

With such a new trigger, i can react an fill thje queue again.

andrewjswan, 2024-03-31:
> So a new trigger like "on_empty_queue_night_mode" would be a solution?

It seems to me that it is superfluous, we need to find where to view the queue and assess whether it is over or not. I'll try, but I don't promise.

popy2k14, 2024-03-31:
thx for looking into it.
As state i am using a workaround now, so take your time :-)

andrewjswan, 2024-03-31:
It looks like I found the reason, but I'm not sure.

popy2k14, 2024-03-31:
The night mode loop and missing "on_empty_queue" trigger is sadly not working with your fix.

But please set the initial brightness not to 255 (= full blast)!
You have set both to 255, so it will dim down from full blast rather than dim up (which i think is better, especially when you have the clock in your sleeping room :-)

So please set it to

```
    uint8_t brightness_ = 0;
    uint8_t target_brightness_;
```

PS.: target_brightness_ will be initialized from the pathon CONF variable, so no need to set.
But when you want to be sure they are initialized also when there is an python CONF parsing change,
please set it to:

```
    uint8_t brightness_ = 0;
    uint8_t target_brightness_ = 0;
```

thx

andrewjswan, 2024-04-01:
> The night mode loop and missing "on_empty_queue" trigger is sadly not working with your fix.

Any log?

andrewjswan, 2024-04-01:
PR updated ...

popy2k14, 2024-04-02:
@andrewjswan @lubeda Tested your new PR (tag: 2024.4.1) and it's working as expected.
closing this now thx

closed 2024-04-02 by popy2k14