lubeda/EspHoMaTriXv2 #230: [QUESTION] How to change default clock and date time?

williknow, 2024-04-25:
Moin!
Great Work here. Thank you very much.
# Question

I changed default clock and date color in the lambda section of the yaml
id(rgb8x32)->set_clock_color(173,216,230);
But it is ignored.

## Additional information

- used Hardware:
  - ESP32  D1 mini

  - EspHoMaTriXv2 version: 2023.7.1

## Logs

I think this is the Problem. It is only shown on first boot after flashing:
[11:21:03][D][EHMTXv2:955]: free_screen: found by endtime 0
[11:21:03][D][EHMTXv2:906]: clock_screen_color lifetime: 20160 screen_time: 10 red: 240 green: 240 blue: 240
[11:21:04][D][EHMTXv2:030]: queue: clock for: 10 sec
[11:21:04][D][EHMTXv2:916]: date_screen lifetime: 20160 screen_time: 5 red: 240 green: 240 blue: 240

Full Log:

INFO ESPHome 2024.4.1
INFO Reading configuration /config/esphome/ehmtxv2.yaml...
WARNING 
WARNING This will only work with esphome >= 2023.7.0
WARNING Please check the documentation and wiki https://github.com/lubeda/EspHoMaTriXv2
WARNING 
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Preparing icons, this may take some seconds.
INFO EsphoMaTrix: wrote html-file with icon preview: /config/esphome/ehmtxv2.html
INFO List of icons for e.g. blueprint:

["error","scale","sleep8x32",]

INFO Compiling app...
Processing ehmtxv2 (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- NeoPixelBus @ 2.7.3
Compiling .pioenvs/ehmtxv2/src/main.cpp.o
Linking .pioenvs/ehmtxv2/firmware.elf
RAM:   [====      ]  44.5% (used 36428 bytes from 81920 bytes)
Flash: [=====     ]  52.6% (used 549799 bytes from 1044464 bytes)
Building .pioenvs/ehmtxv2/firmware.bin
esp8266_copy_factory_bin([".pioenvs/ehmtxv2/firmware.bin"], [".pioenvs/ehmtxv2/firmware.elf"])
========================= [SUCCESS] Took 26.17 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.79
INFO Uploading /data/build/ehmtxv2/.pioenvs/ehmtxv2/firmware.bin (553952 bytes)
INFO Compressed to 374101 bytes
Uploading: [============================================================] 100% Done...

INFO Upload took 7.91 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.79 using esphome API
INFO Successfully connected to ehmtxv2 @ 192.168.1.79 in 19.698s
INFO Successful handshake with ehmtxv2 @ 192.168.1.79 in 3.026s
[11:21:03][I][app:100]: ESPHome version 2024.4.1 compiled on Apr 25 2024, 11:20:10
[11:21:03][C][wifi:580]: WiFi:
[11:21:03][C][wifi:408]:   Local MAC: CC:50:E3:45:4A:BC
[11:21:03][C][wifi:413]:   SSID: 'Smartzeugs'[redacted]
[11:21:03][C][wifi:416]:   IP Address: 192.168.1.79
[11:21:03][C][wifi:419]:   BSSID: 04:18:D6:A2:FD:41[redacted]
[11:21:03][C][wifi:421]:   Hostname: 'ehmtxv2'
[11:21:03][C][wifi:423]:   Signal strength: -79 dB ▂▄▆█
[11:21:03][C][wifi:427]:   Channel: 1
[11:21:03][C][wifi:428]:   Subnet: 255.255.255.0
[11:21:03][C][wifi:429]:   Gateway: 192.168.1.1
[11:21:03][C][wifi:430]:   DNS1: 192.168.1.1
[11:21:03][C][wifi:431]:   DNS2: 0.0.0.0
[11:21:03][C][logger:166]: Logger:
[11:21:03][C][logger:167]:   Level: DEBUG
[11:21:03][C][logger:169]:   Log Baud Rate: 115200
[11:21:03][C][logger:170]:   Hardware UART: UART0
[11:21:03][C][light:103]: Light 'ehmtxv2 Light'
[11:21:03][C][light:105]:   Default Transition Length: 1.0s
[11:21:03][C][light:106]:   Gamma Correct: 2.00
[11:21:03][C][status:034]: Status Binary Sensor 'ehmtxv2 Status'
[11:21:03][C][status:034]:   Device Class: 'connectivity'
[11:21:03][C][homeassistant.time:010]: Home Assistant Time:
[11:21:03][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[11:21:03][C][mdns:115]: mDNS:
[11:21:03][C][mdns:116]:   Hostname: ehmtxv2
[11:21:03][C][ota:096]: Over-The-Air Updates:
[11:21:03][C][ota:097]:   Address: ehmtxv2.local:8266
[11:21:03][C][ota:100]:   Using Password.
[11:21:03][C][ota:103]:   OTA version: 2.
[11:21:03][C][api:139]: API Server:
[11:21:03][C][api:140]:   Address: ehmtxv2.local:6053
[11:21:03][C][api:142]:   Using noise encryption: YES
[11:21:03][D][EHMTXv2:419]: time sync => start running
[11:21:03][D][EHMTXv2:955]: free_screen: found by endtime 0
[11:21:03][D][EHMTXv2:906]: clock_screen_color lifetime: 20160 screen_time: 10 red: 240 green: 240 blue: 240
[11:21:04][D][EHMTXv2:030]: queue: clock for: 10 sec
[11:21:04][D][EHMTXv2:916]: date_screen lifetime: 20160 screen_time: 5 red: 240 green: 240 blue: 240
[11:21:04][D][EHMTXv2:955]: free_screen: found by endtime 1
[11:21:04][D][EHMTXv2:033]: queue: date for: 5 sec
[11:21:04][C][EHMTXv2:1057]: EspHoMatriXv2 version: 2023.7.1
[11:21:04][C][EHMTXv2:1058]: Icons: 3 of 90
[11:21:04][C][EHMTXv2:1059]: Clock interval: 90 s
[11:21:04][C][EHMTXv2:1060]: Date format: %d.%m.
[11:21:04][C][EHMTXv2:1061]: Time format: %H:%M
[11:21:04][C][EHMTXv2:1062]: Interval (ms) scroll: 120
[11:21:04][C][EHMTXv2:1065]: show day of week
[11:21:04][C][EHMTXv2:1069]: show date
[11:21:04][C][EHMTXv2:1075]: Fade in activated: 16 steps
[11:21:04][C][EHMTXv2:1079]: weekstart: monday
[11:21:05][D][EHMTXv2:503]: forced clock_interval
[11:21:05][D][api:102]: Accepted 192.168.1.221
[11:21:05][D][api:102]: Accepted 192.168.1.221
[11:21:05][W][component:237]: Component api took a long time for an operation (572 ms).
[11:21:05][W][component:238]: Components should block for at most 30 ms.
[11:21:07][D][api.connection:1210]: Home Assistant 2024.4.4 (192.168.1.221): Connected successfully
[11:21:07][D][api.connection:1210]: ESPHome Logs 2024.4.1 (192.168.1.221): Connected successfully
[11:21:07][D][time:048]: Synchronized time: 2024-04-25 11:21:07
[11:21:07][D][time:048]: Synchronized time: 2024-04-25 11:21:07
[11:21:14][D][EHMTXv2:033]: queue: date for: 5 sec
[11:21:20][D][EHMTXv2:030]: queue: clock for: 10 sec
[11:21:31][D][EHMTXv2:033]: queue: date for: 5 sec

williknow, 2024-04-25:
Sorry missed the latest Release 2024.3.0. 
Ill try that. Still interested in ESP8266 yaml for that.
Thank you very much

closed 2024-04-25 by williknow

andrewjswan, 2024-04-25:
> Sorry missed the latest Release 2024.3.0.

Latest - https://github.com/lubeda/EspHoMaTriXv2/releases/tag/2024.4.1

williknow, 2024-05-08:
Where do I find the latest release of the yaml file for an ESP8266 not Ulanzi? 
Thanks a lot.

trip5, 2024-05-08:
I think you may be a bit confused.  You can't just load a YAML and make it work for an ESP8266.  It's the EHMTXv2 libraries that will not work on an ESP8266 correctly.

You will need to reference older versions in your YAML something like:

```
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: 2023.3.5 # this where you specify an older version
      refresh: 60s 
      components: [ ehmtxv2 ]
```

The `ref:` points to the release number or the commit (ie 6ed829c).

You can dig through the [old releases](https://github.com/lubeda/EsphoMaTrix/releases) and try older ones.... I have a feeling that it was 2023 sometime but I can't be sure on when.  Click on the release date at the top of each entry to see the old frontpage and notes.  I have a feeling that it was 2023.04 that becomes problematic for the ESP8266.

The ESP8266 is just too limited in memory and speed to handle all the functions that were being added to it. Too much functions, features, and schnickschnack (did I spell that right?) has been added to the v2 to even load on an ESP8266. 
 I think I last tried compiling it on ESP8266 summer of last year and it barely worked.  It crashed sometimes just trying to switch screens and I could no longer flash new firmware via wi-fi.

It might just be easier to go over to the old [EHMTX (v1)](https://github.com/lubeda/EsphoMaTrix) and use that on an ESP8266.  The old version is still very functional.

But seriously, for something like this, you want an ESP32... if you're looking at upgrading an old Awtrix, I used an D1 ESP32 in mine.  The pins are completely compatible, though I did have to edit my YAML because the old Awtrix-style setup is very different than the Ulanzi setup... Try something like this: https://www.aliexpress.com/item/1005005972627549.html - For 5 bucks, it's worth it.