lubeda/EspHoMaTriXv2 #231: [BUG] on_empty_queue loop after ~25 days runtime

popy2k14, 2024-04-28:
# Bug report

## Describe the bug

The clock displayed nothing but was reachable with HA.
But it's really busy! Viewing Logs online was hard and HA freezes because of the amount of logs messages (because of the loop).
Display and "Night mode" was on.
It looped and always triggered "on_empty_queue"

## Additional information

- used Hardware:
  - ESP32 (ulanzi TC0001)
  - EspHoMaTriXv2 version: 2024.4.1

## To Reproduce

Steps to reproduce the behavior:
I dont know.
I just was running stable for ~25 days and one night this issue occured.

## Expected behavior

Trigger "on_empty_queue" -> my yaml triggers "set_night_mode_on()" (or off) which add's at least an clock_screen.
Do not loop.

## Configuration

```
substitutions:
  # Pin definition from https://github.com/aptonline/PixelIt_Ulanzi 
  battery_pin: GPIO34 
  ldr_pin: GPIO35 
  matrix_pin: GPIO32 
  left_button_pin: GPIO26 
  mid_button_pin: GPIO27 
  right_button_pin: GPIO14 
  scl_pin: GPIO22 
  sda_pin: GPIO21
  
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: 2024.4.1
    refresh: 1min
    components: [ehmtxv2]    
  
esphome:
  comment: "Ulanzi TC001"
  project:
    name: "Ulanzi.EHMTXv2"
    version: "2.0.0"
  on_boot:
    then:
      - ds1307.read_time:

esp32:
  board: esp32dev

#include device_base yaml
packages:
  device_base: !include /config/esphome/common/device_base.yaml

font:
  # Font included in this folder
  # - file: mateine.ttf
  #   size: 16
  #   id: default_font
  #   glyphs:  |
[... 532 lines cut ...]
[07:37:43][D][EHMTXv2:2096]: date_screen lifetime: 1440 screen_time: 5 red: 240 green: 240 blue: 240
[07:37:43][D][EHMTXv2:107]: queue: date for: 5.0 sec
[07:37:43][W][component:232]: Component display took a long time for an operation (62 ms).
[07:37:43][W][component:233]: Components should block for at most 30 ms.
[07:37:43][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Middle button'
[07:37:43][C][gpio.binary_sensor:016]:   Pin: GPIO27
[07:37:43][D][EHMTXv2:1358]: on_empty_queue trigger
[07:37:43][I][on_empty_queue:180]: set display on!
[07:37:43][D][EHMTXv2:158]: display on
[07:37:43][I][on_empty_queue:191]: set night mode on!
[07:37:43][D][EHMTXv2:180]: night mode on
[07:37:43][D][EHMTXv2:2082]: clock_screen_color lifetime: 1440 screen_time: 10 red: 235 green: 0 blue: 0
[07:37:43][D][EHMTXv2:104]: queue: clock for: 10.0 sec
[07:37:43][I][EHMTXv2:2335]: set_brightness 20 => 7.84 %
[07:37:43][D][EHMTXv2:2096]: date_screen lifetime: 1440 screen_time: 5 red: 240 green: 240 blue: 240
[07:37:43][D][EHMTXv2:107]: queue: date for: 5.0 sec
[07:37:43][W][component:232]: Component display took a long time for an operation (63 ms).
[07:37:43][W][component:233]: Components should block for at most 30 ms.
[07:37:43][C][light:103]: Light 'Light'
[07:37:43][C][light:105]:   Default Transition Length: 1.0s
[07:37:43][C][light:106]:   Gamma Correct: 2.00
[07:37:43][C][template.switch:068]: Template Switch 'Display'
[07:37:43][C][template.switch:070]:   Icon: 'mdi:power'
[07:37:43][D][EHMTXv2:104]: queue: clock for: 10.0 sec
```

popy2k14, 2024-04-28:
I have now made a workaround -> reboot when more than 20x on_empty_queue triggers in one minute.

```
  on_empty_queue:
    then:
      lambda: |-
        id(queue_empty_count) = id(queue_empty_count) + 1;
        ...

interval:
    #workaround that clock reboots, when it loops
  - interval: 1min
    then:
      lambda: |-
        //check on empty queue count
        ESP_LOGI("interval", "empty queue counter is: %i", id(queue_empty_count));
        if (id(queue_empty_count) > 20) {
            ESP_LOGI("interval", "empty queue is looping -> reboot");
            id(reboot).press();
        }
        id(queue_empty_count) = 0;

        ...

```

popy2k14, 2024-05-01:
> There is possibly a problem with the memory management that leads to your behavior. I will try to find a solution or implement your work araound.

Thx s lot for looking into this.
I have changed my workaround to > 10000, with 20 it has false triggers. I could give feedback in 1-2 months if it helps. 

If you need tests (ex. debug Version with memory log messages), I am here to help testing 😃

lubeda, 2024-05-01:
`on_empty_queue` is triggered each tick (~16ms) which is "stupid". To trigger it once (breaking change) or in a larger interval (not so breaking change) is more clever. I think I will go for once, which is more structured. Or are there aspects which are better with once a minute?

andrewjswan, 2024-05-01:
> Or are there aspects which are better with once a minute?

Isn’t it possible that we won’t have anything on the screen for almost a minute?

popy2k14, 2024-05-01:
I personally find it rather irritating that you, as a user have the possibility to have an empty/black screen and must serve the clock as content! It's a clock and should always show the clock (as it did in previous versions).

The workaround in yaml (see above or if it's in c code) is just a workaround.
As you can see in my yaml i am adding a clock_screen when `on_empty_queue` is triggered.
So i think the real issue is an underlaying issue in the queue (on adding or consuming screens).
I think adding something in c is not the real solution.

So i think there are multiple ways to solve this (sor):

- Search the underlaying issue with the queue
   As stated, i am here to let an debug version run and see if the issue pop's up again and this time have more precise logs.
   
- Add https://github.com/lubeda/EspHoMaTriXv2/issues/208 and maybe this issue is not present anymore.
  But that's also rather an workaround. But it's not so irritating as having the possibility to have an black screen!
  I know the feature was added because you as a user have more possibilitys in features,
  But i find it irritating for most of us users.
  Looked at the code and there is an defen alerady so clock_screens does'nt get remove.
  Maybe as an beginning fpr #208 just add this define as default for clock and date_screen and add and config option to set other screens also as static later.
  
  
- Add my yaml workaound in c code
   not prefered.
   
What do you think which route we should go?

lubeda, 2024-05-01:
> It's a clock and should always show the clock

If you see this as a clock, you are totally right, if you see this as a status-display the on_empty_queue way is more flexible.

> As you can see in my yaml i am adding a clock_screen when `on_empty_queue` is triggered.

That is the way to go if you want a clock.

popy2k14, 2024-12-14:
> `on_empty_queue` is triggered each tick (~16ms) which is "stupid". To trigger it once (breaking change) or in a larger interval (not so breaking change) is more clever. I think I will go for once, which is more structured. Or are there aspects which are better with once a minute?

Saw that my clock hat the same issue (looping on_empty_queue) but now slower.
This was changed in 2024.5.1 to 15 seconds.
So my workaround with > 10000 in under an minute does'nt work.
Changed it to >=3 and see how it goes.

As stated, the issue doesnt pop up frequntly but it happens.
When i look over the code it seems that the tick() function does'nt find the oldest queued element:

```
        if (this->screen_pointer == MAXQUEUE)
        {
          this->screen_pointer = find_oldest_queue_element();
        }

        if (this->screen_pointer != MAXQUEUE)
        {
           ...
        else
        {
          this->next_action_time = ts + 15000 ; // come back in 15 Seconds
          for (auto *t : on_empty_queue_triggers_)
          {
              ESP_LOGD(TAG, "on_empty_queue trigger");
              t->process();
            } 
        }
```

Maybe because of memory corruption of the queue.
You can see my yaml above, that on empty queue i am adding an clock_screen, so it should not loop!

A simple workaround is to reboot in such a case to get rid of the memory corruption.
I know it's an ugly workaround, but i dont have time to investigate further.
My assumption is the "queue" is the culprit here.