lubeda/EspHoMaTriXv2 #233: [QUESTION] [on_empty_queue] is an invalid option for [ehmtxv2]. Did you mean [allow_empty_screen]?

robudus, 2024-05-09:
# Question

Hi i would like to compile the latest version to my esp32 but get the Error:
I use EspHome in HomeAssistant 2024.4.2

see logs for the error.

I use this yaml
https://github.com/lubeda/EspHoMaTriXv2/blob/latest/install/ulanzi-easy.yaml

Can you please help me. 

Thanks

ROBudus

## Additional information

- used Hardware:
  - ESP32

  - EspHoMaTriXv2 version: 2024.5.0

## Logs

```text
  [on_empty_queue] is an invalid option for [ehmtxv2]. Did you mean [allow_empty_screen]?
  on_empty_queue: 
    then: 
      - lambda: |-
          id(rgb8x32).set_infotext_color(20,20,20,20,20,20,false,1);
          id(rgb8x32).icon_clock("calendar|day#0",1440,10,false);
```

robudus, 2024-05-09:
`substitutions:
  devicename: esphomatrixv2
  friendly_name: EspHoMaTriXv2
  board: esp32dev
  # Pin definition from https://github.com/aptonline/PixelIt_Ulanzi 
  battery_pin: GPIO34 
  ldr_pin: GPIO35 
  matrix_pin: GPIO32 
  left_button_pin: GPIO26 
  mid_button_pin: GPIO27 
  right_button_pin: GPIO14 
  scl_pin: GPIO22 
  sda_pin: GPIO21 

output:
  - platform: gpio
    pin:
      number: GPIO15
      ignore_strapping_warning: true
    id: buzzer_pin

switch:
  - platform: template
    name: "Auto-Adjust Brightness"
    id: switch_autobrightness
    icon: mdi:brightness-auto
    restore_mode: RESTORE_DEFAULT_ON
    lambda: |-
      if (id(aab_enable)) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      lambda: |-
        id(aab_enable) = true;
    turn_off_action:
      lambda: |-
        id(aab_enable) = false;

globals:
  # aab = auto-adjustable brightness
  - id: aab_enable
    type: "bool"
    restore_value: true
    initial_value: "true"
  - id: aab_add
    type: int
    initial_value: '10'
  - id: aab_max
    type: int
    initial_value: '220'
  - id: aab_min
    type: int
    initial_value: '20'

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: "2024.5.0"
    refresh: 600s 
    components: [ ehmtxv2 ]   

esphome:
  comment: "EHMTXv2 from LuBeDa"
  name: $devicename 
  project:
    name: "Ulanzi.EHMTXv2"
    version: "2024.5.0"
[... 205 lines cut ...]
    - lameid: 34801
      id: shopping
    - lameid: 37533
      id: thunder
    - lameid: 5313
      id: mic
    - lameid: 102
      id: birthday
    - lameid: 1782
      id: xmastree
    - id: lock
      lameid: 7775
  on_empty_queue:
    then:
      - lambda: |-
          id(rgb8x32).set_infotext_color(20,20,20,20,20,20,false,1);
          id(rgb8x32).icon_clock("calendar|day#0",1440,10,false);

animation:
  - file: "1pixel.gif"
    id: animation_1p

image:
  - file: "1pixel.gif"
    id: image_1p`

robudus, 2024-05-09:
``
substitutions:
  devicename: esphomatrixv2
  friendly_name: EspHoMaTriXv2
  board: esp32dev
  # Pin definition from https://github.com/aptonline/PixelIt_Ulanzi 
  battery_pin: GPIO34 
  ldr_pin: GPIO35 
  matrix_pin: GPIO32 
  left_button_pin: GPIO26 
  mid_button_pin: GPIO27 
  right_button_pin: GPIO14 
  scl_pin: GPIO22 
  sda_pin: GPIO21 

output:
  - platform: gpio
    pin:
      number: GPIO15
      ignore_strapping_warning: true
    id: buzzer_pin

switch:
  - platform: template
    name: "Auto-Adjust Brightness"
    id: switch_autobrightness
    icon: mdi:brightness-auto
    restore_mode: RESTORE_DEFAULT_ON
    lambda: |-
      if (id(aab_enable)) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      lambda: |-
        id(aab_enable) = true;
    turn_off_action:
      lambda: |-
        id(aab_enable) = false;

globals:
  # aab = auto-adjustable brightness
  - id: aab_enable
    type: "bool"
    restore_value: true
    initial_value: "true"
  - id: aab_add
    type: int
    initial_value: '10'
  - id: aab_max
    type: int
    initial_value: '220'
  - id: aab_min
    type: int
    initial_value: '20'

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: "2024.5.0"
    refresh: 600s 
    components: [ ehmtxv2 ]   

esphome:
  comment: "EHMTXv2 from LuBeDa"
  name: $devicename 
  project:
    name: "Ulanzi.EHMTXv2"
[... 207 lines cut ...]
      id: shopping
    - lameid: 37533
      id: thunder
    - lameid: 5313
      id: mic
    - lameid: 102
      id: birthday
    - lameid: 1782
      id: xmastree
    - id: lock
      lameid: 7775
  on_empty_queue:
    then:
      - lambda: |-
          id(rgb8x32).set_infotext_color(20,20,20,20,20,20,false,1);
          id(rgb8x32).icon_clock("calendar|day#0",1440,10,false);

animation:
  - file: "1pixel.gif"
    id: animation_1p

image:
  - file: "1pixel.gif"
    id: image_1p
``

lubeda, 2024-05-09:
Yes, it's a bug! It works with 2024.4.0.

I will investigate

N3rdix, 2024-05-16:
Just got the same error when copying the 2024.3.0 files to my local environment and validating the config. When restoring the files from the previous release the validation works as expected.

Using EspHome 2024.4.2. Do you mean the problem could be fixed by downgrading it to 2024.4.0, at least as a workaround until the real issue is solved?

lubeda, 2024-05-16:
I just found out:
There is no EspHoMaTriXv2 version tagged 2024.5.0! So, as fallback the default branch is used (see https://esphome.io/components/external_components#external-components), which is very old. Please use "latest" or an available version number as ref, e,g, latest or 2024.3.0,

So this is not a bug in EspHoMaTriX!

closed 2024-05-16 by lubeda

N3rdix, 2024-05-16:
Where exactly do you mean should this "latest" be adjusted?

In my case I just have a source of type local in the yaml

lubeda, 2024-05-17:
You have to specify a valid branch under the **ref** tag in your yaml.

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: latest
    refresh: 60s 
    components: [ ehmtxv2 ]   
```

N3rdix, 2024-05-19:
I got that when using a git resource, but how to handle the error when using a local source with the files copied to a local folder?

```
external_components:
  - source:
      path: /copied_components

```

andrewjswan, 2024-05-19:
> files copied to a local folder

Copy files from `latest` branch...

N3rdix, 2024-05-19:
Ok my fault, shame on me 😀 now I got the point, thanks