lubeda/EspHoMaTriXv2 #235: [QUESTION] How show on Display Text or Picture when OTA start?

andrewjswan, 2024-05-17:
# Question

How show on Display Text or Picture when OTA start?
I try:
```
ota:
  on_begin:
    then:
      - lambda: id(rtttl_out).turn_off();
      - lambda: |-
          id(rgb8x32)->alert_screen("wait", "OTA", 15, true, 240, 240, 240);
          id(rgb8x32)->draw();
```

But it’s clear that the rendering doesn’t keep up and the last information displayed is on the screen.

lubeda, 2024-11-01:
try this

```
on_start_running:
     then:
       lambda: |-
          id(rgb8x32)->alert_screen("wait", "OTA", 15, true, 240, 240, 240);
```

closed 2024-11-01 by lubeda

trip5, 2024-11-02:
Hey.  Your original code is probably fine.  I tried this:
````
ota:
  - platform: esphome
    password: !secret ota_password
    # a note here that for some reason these clocks seem unable to do a "pretty" OTA update
    # but leaving the code here in hopes that it does work in the future
    on_begin:
      then:
        - lambda: 'id(rgb8x32)->alert_screen("wait", "OTA...", 15, true, 240, 240, 240);'
        - logger.log: "OTA Update started."
    on_progress:
      then:
        - lambda: |-
            id(rgb8x32)->alert_screen("wait", std::string("OTA ") + std::to_string(x) + "%", 15, true, 240, 240, 240);
        - logger.log:
            format: "OTA progress %0.1f%%"
            args: ["x"]
    on_end:
      then:
        - lambda: 'id(rgb8x32)->alert_screen("wait", "OTA Done", 15, true, 240, 240, 240);'
        - logger.log: "OTA end"
    on_error:
      then:
        - lambda: 'id(rgb8x32)->alert_screen("wait", "Error!", 15, true, 240, 240, 240);'
        - logger.log:
            format: "OTA update error %d"
            args: ["x"]
````
And I've tried similar routines on my MAX7219 Clocks and different 7-segment display clocks.

There's something about the driver that means the display driver is completely stopped while OTA is running.

Strangely, a driver I wrote for a VFD display can show all OTA messages.  I've been trying to figure out what's different in the hope that I can make it work, but for now... It just can't function.  It's on the drivers that ESPHome is built on, I think.

And it doesn't seem to have anything to do where or not the display's interface is I2C or SPI...  The VFD is SPI and so is the neopixel display sued in the Ulanzi...  I'd say leave something like this in your YAML in the hope that one day it works (it doesn't hurt to leave it in anyways).  If it does, it's because me or someone like me fixes the underlying ESPHome Driver.

andrewjswan, 2024-11-02:
> try this

Not work :(

This not work to:
```
ota:
  on_begin:
    then:
      - lambda: |-
          id(rgb8x32)->alert_screen("wait", "OTA", 15, true, 240, 240, 240);
          id(rgb8x32)->tick();
          id(rgb8x32)->draw();
```

trip5, 2024-11-03:
Yup, as I said, blame to this falls on the driver itself.

It's weird to me that a driver I wrote myself (when I barely knew what I was doing) actually works for the OTA display function but none of the official drivers do.  I think it has something to do with the `void loop()` in the driver.  As I said, leave the code in there.  It doesn't do any harm. I'll take a look at it in the next week and see if there's something I can do about it.

Hopefully, I can make some PRs to some official ESPHome drivers and get a few displays fixed for this.

trip5, 2024-11-05:
Make no mistake.  I wasn't pointing to the `ehmtxv2:` component.  I didn't rename mine out of Lubeda's usual.  With my `id(ehmtx_display)` it's pointing to the `display:` component.  There are no functions in ehmtxv2 with the names of `clear()` and `loop()` - they're part of the display component.  Those lines will cause an error.

But by calling `id(rgb8x32)->update()` in a `on_progress:` block, the clock will try to continue it's usual updates - just slower.  It might be pretty weird unless there's a screen in place like the `alert_screen`?

So your idea  might work (just to show there's an OTA going on... but it won't display the progress of the OTA.  Maybe that's where there's some confusion here.  I was trying to show the percentage of the OTA update.  It's been successful on my other clocks through various methods...

andrewjswan, 2024-11-05:
@trip5 I don't want to show progress, I want to remove the frozen image during OTA, i.e. display the inscription that OTA is currently running, and that's it...

andrewjswan, 2024-11-05:
> Make no mistake. I wasn't pointing to the `ehmtxv2:` component. I didn't rename mine out of Lubeda's usual. With my `id(ehmtx_display)` it's pointing to the `display:` component.

Like this:
```Yaml
ota:
  on_begin:
    then:
      - lambda: |-
          id(ehmtx_display)->clear();
          id(rgb8x32)->alert_screen("wait", "OTA", 15, true, 240, 240, 240);
          id(ehmtx_display)->update();
          id(ehmtx_display)->loop();
```

andrewjswan, 2024-11-05:
@trip5 I tried different options, the screen is always frozen, which was before OTA, and during the OTA process nothing changes.
```Yaml
ota:
  on_begin:
    then:
      - lambda: id(rtttl_out).turn_off();
      - lambda: |-
          id(ehmtx_display)->clear();
          id(ehmtx_display)->print(0, 0, id(default_font), "OTA");
          id(ehmtx_display)->update();
          id(ehmtx_display)->loop();
  on_progress:
    then:
      - lambda: |-
          id(ehmtx_display)->clear();
          id(ehmtx_display)->print(0, 0, id(default_font), "OTA");
          id(ehmtx_display)->update();
          id(ehmtx_display)->loop();
```