lubeda/EspHoMaTriXv2 #240: [BUG] ESPhome 2024.6.1 Breaking Change: Update YAML File for OTA Fix

idodov, 2024-06-20:
In ESPhome 2024.6.1, there is a breaking change that requires updating the YAML file. To fix this, you need to add `platform: esphome` under the OTA section. This adjustment should address the issue.

```yaml
ota:
  password: !secret ota_password
  platform: esphome
```

trip5, 2024-06-21:
Not sure 'bug' is the right tag for this.  Maybe a 'heads-up' since this is an ESPHome breaking change and not related to EspHoMaTriXv2

https://community.home-assistant.io/t/esphome-ota-configuration/730032

Sigh, ESPHome is great but looks like they introduced TWO breaking changes in 2024.6 including dallas sensors too (shouldn't affect this project but... good to be aware).

andrewjswan, 2024-06-21:
Fixed

closed 2024-12-11 by lubeda