lubeda/EspHoMaTriXv2 #243: [QUESTION] 8x8 Matrix Support

ataa, 2024-07-13:
# Question

Hello,

Is it possible to use ESP32-S3 Modules with 8x8 matrix and display an icon like weather icon first, then scroll the text?

I am using ESP32-S3-Matrix (8x8 addressable RGB + Gyro & Acc. Sensor) which can be bought for about $7.

[image]

andrewjswan, 2024-07-15:
The current firmware should work fine with this type of matrix, with all the limitations that 8x8 imposes on the algorithm designed for 8x32. 
You can also rewrite the firmware purely for 8x8, and remove the extra ones (screens that only support 8x32).
But it is better to add 3 8x8 matrices and get 8x32.

lubeda, 2024-11-01:
8x8 is to small to display readable text.

closed 2024-11-01 by lubeda