lubeda/EspHoMaTriXv2 #244: text box helper home assistant and local temperature sensor

bepppppo, 2024-07-19:
# Question

I have read and tried different solutions to get my display 8x32 to show a local temperature sensor DHT22 attached to the esp32, not only when there are changes to the sensor but at a specific time, for example every 2 minutes.
My target is to get the display to show clock and every 2 minutes room temperature via DHT22 and ouside temperature from home assistant weather service.
Also when I need I wanted to be able to Type in a text box from home assistant (via helpers)  and the display will show that message for around 2 minutes.
I have accomplish this with regular esphome code but I wanted to try out your firmware as it is very nice.

So just to make it simple I would like to have these 2 functions:

1- input box from home assistant input box helper without using automations

2- show local DHT22 temperature sensor every 2 minutes

I appreciate your help on this

## Additional information

- used Hardware:
  - ESP32 

  - EspHoMaTriXv2 version: [e.g. 2023.5.0]
  - Latest version

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

this is my current working code for esphome without EspHoMaTriXv2 component:

esphome:
  name: admin-front-clock
  friendly_name: "Admin Front Clock"
  on_boot:
    priority: 800
    then:
      - script.execute: show_pages
esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino
wifi:
  networks:
  - ssid: !secret wifi_ssid
    eap:
      username: edu002\8383-guest
      password: !secret wifi_password
  ap:
    ssid: "Clock-Display Fallback Hotspot"
    password: "12345678"   
api:
  encryption:
    key: 
ota:
 - platform: esphome
   password: 

time:
  - platform: sntp
    id: sntp_time
    timezone: 

# Enable logging
logger:
  level: WARN
web_server:
  port: 80
  include_internal: true
button:
  - platform: restart
    name: "Restart Clock"

sensor:
  - platform: homeassistant
    entity_id: weather.forecast_home
    attribute: temperature
    id: temp_outside

  - platform: dht
    pin: 18
    model: AM2302
    temperature:
      name: "Temperature Inside"
      id: temp_inside
      unit_of_measurement: "°C"
      filters:
        - offset: -3.1
    humidity:
      name: "Humidity Inside"
      id: hum_inside
      filters:
        - offset: 1.0
    update_interval: 60s

text_sensor:

  - platform: homeassistant
    name: "HA admin front Txt"
    id: ha_admin_front_txt
    entity_id: input_text.admin_front_text
    internal: true

  - platform: homeassistant
    name: "HA Txt"
    id: hatxt
    entity_id: input_text.matrix_text
    internal: true

light:
  - platform: neopixelbus
    type: GRB
    variant: WS2812
    pin: GPIO16
    num_leds: 256
    name: "led_matrix"
    id: led_matrix_32x8
    color_correct: [10%, 10%, 60%]

script:
  - id: show_pages
    mode: single
    then:
      - while:
          condition:
            - script.is_running: show_pages
          then:
            - display.page.show: page1
            - component.update: led_matrix_32x8_display
            - delay: 120s
            - display.page.show: page2
            - component.update: led_matrix_32x8_display
            - delay: 5s
            - display.page.show: page3
            - component.update: led_matrix_32x8_display
            - delay: 5s

globals:
   - id: coords
     type: int
     restore_value: no
     initial_value: '0'
   - id: length
     type: int
     restore_value: no
     initial_value: '0'
interval:
  - interval: 100ms
    then:
        lambda: |-
            if (id(coords) < -(id(length))) {
             id(coords) = 8;
                }
            else  {
               id(coords) -= 2;
            }

display:
  - platform: addressable_light
    id: led_matrix_32x8_display
    addressable_light_id: led_matrix_32x8
    width: 32
    height: 8
    pixel_mapper: |-
      if (x % 2 == 0) {
        return (x * 8) + y;
      }
      return (x * 8) + (7 - y);
    rotation: 0°
    pages:
      - id: page1
        lambda: |-
          if ((id(ha_admin_front_txt).state) != "") {
          std::string printout=(id(pixelmix_8), TextAlign::TOP_LEFT, "%s", id(ha_admin_front_txt).state.c_str());
          int clength = printout.length();
          id(length)=clength*8;
          it.print(id(coords), 0, id(pixelmix_8), printout.c_str());;
          }
          else if ((id(hatxt).state) != "") {
          std::string printout=(id(pixelmix_8), TextAlign::TOP_LEFT, "%s", id(hatxt).state.c_str());
          int clength = printout.length();
          id(length)=clength*8;
          it.print(id(coords), 0, id(pixelmix_8), printout.c_str());;
          }
          else
          {
          it.strftime(4, -1, id(pixelmix_8), "%H:%M", id(sntp_time).now());
          }
      - id: page2
        lambda: |-
          it.printf(0, -1, id(pixelmix_8), "%.1f°I" , id(temp_inside).state);
      - id: page3
        lambda: |-
          it.printf(0, -1, id(pixelmix_8), "%.1f°O" , id(temp_outside).state);

font:
  - file: "pixelmix.ttf"
    id: pixelmix_8
    size: 8

andrewjswan, 2024-07-19:
In EspHoMaTriXv2 there is a concept of a screen queue. Each screen has a display time and a lifetime from the moment it is placed in the queue. Therefore, there is no simple solution to the problem of displaying a screen every 2 minutes.
If you have only 2 screens, a clock and a temperature, then make the display time of the clock 1 minute 50 seconds for the clock screen, and for the temperature screen the display time is 10 seconds, conditionally this will give what you want.

Regarding the text output, in HA you create an input_text, based on it you create a template sensor, and add it to ESPHome, in ESPHome you create an automation to change this sensor that will display it. But it is easier to do all this on the HA side.
You can create a text entity in ESPHome, and create automation and output for changing the value.

bepppppo, 2024-07-20:
thank you for your reply andrew, do you have an example code I can refer to create the HA automation for the text box? also the text entity in esphome? 
Also thank you for the hint regarding the clock display time, I would have liked to display at least 3 screens: clock, local sensor and HA temperature. Anyway I appreciate your help

trip5, 2024-07-20:
I feel like you might be a bit new to Home Assistant... I'll try to add to your knowledge but you will absolutely need to experiment and try and fail and try again until you get the result you want.  And that's OK.  It took me a weeks to wrap my head around how to do all of this but I learned a lot and applied that knowledge to other things I wanted to do...

Adding the device to Home Assistant adds a service.  These are visible in Developer Tools and you can test out what you would like to accomplish there.  Once you figure it how you want it there, you can move that over to an Automation or Script.

I personally use an automation that is triggered every 5 minutes that starts a script that first deletes the screens from the queues and then screens to the queue.  There are some good examples in the wiki right here already you can take a look at.

https://github.com/lubeda/EspHoMaTriXv2/blob/main/wiki/home.md

Please note the main branch differs very much from what some people are actually using.  I'm personally still stuck on a 2023 branch because I haven't had the time to make sure I know what I'm doing with the latest - upgrading is not always needed if you're happy with what works!

https://github.com/lubeda/EspHoMaTriXv2/tree/2024.6.pre/wiki

But there's also some extra info in the wiki of the latest branch too.  I have to use the error-checking code that is shown in this part because my living-room and weather-sensors sometimes don't work and the script will crash my clock goes blank if I don't:

https://github.com/lubeda/EspHoMaTriXv2/blob/2024.6.pre/wiki/weather-display.md

andrewjswan, 2024-07-20:
> also the text entity in esphome?

https://esphome.io/components/text/index.html

> thank you for your reply andrew, do you have an example code I can refer to create the HA automation for the text box?

This is a banal automation, a trigger for a change in state, an action for output, to any screen that is suitable.

bepppppo, 2024-07-20:
thank you both, I will try and experiment a bit.