lubeda/EspHoMaTriXv2 #245: [BUG] empty queue loop

popy2k14, 2024-07-21:
# Bug report

## Describe the bug

I woke up in the middle of the night and the clock was empty (black).
It seems

## Additional information

- used Hardware:
  - ESP32 ulazi tc0001
  - EspHoMaTriXv2 version: 2024.5.1
  - I was running: 2024.4.1 longer, without issues, maybe i try to go back!?

## To Reproduce

Use the config provided and activate night mode.
Sadly, this happens really rare :-(
As you can see, on_empty_queue() was called every 15 seconds, which called set_night_mode_on() which finally set an clock_screen with 1440 minutes.

Why this happens every 15 seonds!?

## Expected behavior

No unexpected on_empty_queue loop.

## Configuration

```text
substitutions:
  # Pin definition from https://github.com/aptonline/PixelIt_Ulanzi 
  battery_pin: GPIO34 
  ldr_pin: GPIO35 
  matrix_pin: GPIO32 
  left_button_pin: GPIO26 
  mid_button_pin: GPIO27 
  right_button_pin: GPIO14 
  scl_pin: GPIO22 
  sda_pin: GPIO21
  
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: 2024.5.1  #lief lange mit 2024.4.1
    refresh: 1min
    components: [ehmtxv2]    
  
esphome:
  comment: "Ulanzi TC001"
  project:
    name: "Ulanzi.EHMTXv2"
    version: "2.0.0"
  on_boot:
    then:
      - ds1307.read_time:

esp32:
  board: esp32dev

#include device_base yaml
packages:
  device_base: !include /config/esphome/common/device_base.yaml

font:
  # Font included in this folder
  # - file: mateine.ttf
  #   size: 16
  #   id: default_font
  #   glyphs:  |
[... 734 lines cut ...]
[05:24:46][D][EHMTXv2:1356]: on_empty_queue trigger
[05:24:46][I][on_empty_queue:184]: set display on!
[05:24:46][D][EHMTXv2:156]: display on
[05:24:46][I][on_empty_queue:195]: set night mode on!
[05:24:46][D][EHMTXv2:178]: night mode on
[05:24:46][D][EHMTXv2:2082]: clock_screen_color lifetime: 1440 screen_time: 10 red: 235 green: 0 blue: 0
[05:24:46][D][EHMTXv2:104]: queue: clock for: 10.0 sec
[05:24:46][I][EHMTXv2:2335]: set_brightness 20 => 7.84 %
[05:24:46][D][EHMTXv2:2096]: date_screen lifetime: 1440 screen_time: 5 red: 240 green: 240 blue: 240
[05:24:46][D][EHMTXv2:107]: queue: date for: 5.0 sec
[05:24:54][D][sensor:094]: 'Illuminance': Sending state 0.00000 lx with 0 decimals of accuracy
[05:24:55][D][number:012]: 'Brightness': Sending state 20.000000
[05:25:01][D][EHMTXv2:1356]: on_empty_queue trigger
[05:25:01][I][on_empty_queue:184]: set display on!
[05:25:01][D][EHMTXv2:156]: display on
[05:25:01][I][on_empty_queue:195]: set night mode on!
[05:25:01][D][EHMTXv2:178]: night mode on
[05:25:01][D][EHMTXv2:2082]: clock_screen_color lifetime: 1440 screen_time: 10 red: 235 green: 0 blue: 0
[05:25:01][D][EHMTXv2:104]: queue: clock for: 10.0 sec
[05:25:01][I][EHMTXv2:2335]: set_brightness 20 => 7.84 %
[05:25:01][D][EHMTXv2:2096]: date_screen lifetime: 1440 screen_time: 5 red: 240 green: 240 blue: 240
[05:25:01][D][EHMTXv2:107]: queue: date for: 5.0 sec
[05:25:01][W][component:237]: Component display took a long time for an operation (57 ms).
[05:25:01][W][component:238]: Components should block for at most 30 ms.
```

trip5, 2024-07-21:
I'm just going to interject.  Sorry if I can't analyze your problem more deeply (and I'm certainly on an older version of EHMTXv2 than you)...

Is it possible your Home Assistant automation triggered the blank screen instead?

I encountered hours of black screens simply because my weather service was down and the HA Yaml had no error-checking so the script never got past the step it was crashing at.

See the end of this wiki page for an example: https://github.com/lubeda/EspHoMaTriXv2/blob/2024.6.pre/wiki/weather-display.md

popy2k14, 2024-07-22:
@trip5 thx for trying to help, appreciate that.
No i think not that my HA instance is ever sending clear/blank screen.
Checked my scripts for that.
Also in this case there is no issue with the weather sensor.
My clock was in "night mode" and it's configured to not show weather in night mode.
It was not home assistant which was crashing, it was maybe an bug in the clock firmware.
It should just show the clock which does my "on_empty_queue" yaml.

lubeda, 2025-04-26:
fixed with https://github.com/lubeda/EspHoMaTriXv2/pull/289

closed 2025-04-26 by lubeda

popy2k14, 2025-04-27:
thx for fixing this issue!
Will test it as soon esphome 2025.5.X drops stable.