lubeda/EspHoMaTriXv2 #247: [FEATURE REQUEST] Limit battery charge Ulanzi TC001

iz3man, 2024-09-30:
# Feature Request

### Is your feature request related to a problem?

Not problem related

### Describe the solution / feature you'd like

Battery always gets charged to 100% with is very unhealthy

### Describe alternatives you've considered

Connect to a smart charger and turn it on/off accordingly.

### Additional context

I'd like to see an internal charge control mechanism, but am pretty sure it's not possible. Just wanted to ask for confirmation.

trip5, 2024-09-30:
You can see pictures inside in this thread about the voltage regulator failing: https://github.com/Blueforcer/awtrix3/issues/67 - I also had this issue.  I can confirm this is not software-controlled.  It is hardware controlled.

Actually I doubt it's being charged to 100%.  Most chargers have a cut-off.  That said, I'm not sure what mA is the cutoff on this board.  I'd like to think the DC from the USB goes to the ESP simultaneously as the battery charger.  I believe this can be demonstrated with an Lolin ESP32 Lite which can continue running as long as one or the other is plugged in - with no interruption.  You can confirm this yourself if you desire to take apart your clock.  I didn't think to do this test when I had my faulty clock apart and now it can't be done since that clock was eventually sacrificed to the gods of bad soldering.

Also, the 18650s here probably have a shelf-life of about 10-15 years, whether in use or not.  Li-on batteries even put in storage and unused for a few years eventually lose their ability to charge, and since the clock housing isn't restricted to this type of dual 18650 pack, it's extremely easy to replace if your batteries fail... opening the clock is a pain but can be done fairly easily compared to a Macbook or something.

Battery danger with a device like this is pretty minimal.  Phones and notebooks are more abused: charging and using your device at the same time is a good way to wreck your battery just from heat alone.

I'd love to give citations to you but as a 15 year vaper, fan of Mooch, and aficionado of batteries and battery tests, I can assure you this is not something to worry about.  If you're still worried about it, take it apart, and add a switch to the battery cable with a relay wired directly to one of the ESP pins.  It's possible but in terms of time and money, replacing the battery in 2034 is more effective.

trip5, 2024-09-30:
If you're really up for a change... read through that thread and see if they've improved the bypass method.  With the first method, there's a weird plug that when plugging it in again, you have to press the Left and Right buttons together to get the display to show anything.  I couldn't find a software fix (ie emulating the button hold).

I ended up tossing the whole board (saving the regulator) and replacing it with an ESP32-Dev Board which was mostly reliable.  One pixel started to flicker and I opened it up (while tired) to do some work on it and ended up ruining 6 pixels at once.

In retrospect, now that I've studied RGB Leds more, I realize now the board probably wasn't supplying enough power and I should've just added a regulator or capacitor or something... oops.

iz3man, 2024-10-01:
Thanks @trip5. That's important news. If it doesn't start automatically w/o battery, then this is no valid solution. Adding additional hardware is not my plan, so I may leave all as it is :)

iz3man, 2024-10-01:
Clarified all hardware possibilities.

closed 2024-10-01 by iz3man