lubeda/EspHoMaTriXv2 #250: icon_screen is 'unknown action', according to Home Assistant?

clfberlin, 2024-10-21:
Hi there. I am a bit confused, because I keep getting the warning in my devices overview that my automizations use icon_screen, which is an 'unknown action', eventhough it works.
Thanks and cheers

[image]

andrewjswan, 2024-10-21:
What is the version of EspHoMaTriX?

clfberlin, 2024-10-21:
It is 2.0.0
But I just realized that this problem has a different cause: The Ulanzis had been renamed (during the phase with the MAC addressing issues). And indeed there were still references to Ulanzis that are now of course unkown to the system. The error didn't refer to icon_screen itself, but to the device it was (falsely) addressing. Sorry!

closed 2024-10-21 by clfberlin

riker09, 2024-10-24:
@clfberlin Can you please elaborate on "during the phase with the MAC addressing issues"? Because my Ulanzi is constantly being re-discovered by HA for the last few days. If you could provide a link to the subject it would be greatly appreciated.

clfberlin, 2024-10-24:
Hi @riker09 - I am referring to this issue: https://github.com/lubeda/EspHoMaTriXv2/issues/248
With ESPHome 2024.10.0 Ulanzis became unknown or were all addressed as the same device. This was due to changes in dealing with MAC addresses.
If this is your issue too, please follow the setup changes up until Oct. 19th (lights).
I tried going back to previous settings for Ulanzis after updateing ESPHome to 2024.10.1, but that didn't work for me. I am still using the workaround decribed in issue 248. And that works fine.

riker09, 2024-10-24:
yeah, I read that thread right after this issue. sorry for the spam