lubeda/EspHoMaTriXv2 #251: [FEATURE REQUEST] Simple ID screen

MiguelAngelLV, 2024-10-21:
# Feature Request

### Is your feature request related to a problem?

Currently a screen is defined by combining the icon (which is composed of name, category and ID) and screen mode, a simple unique ID could be used, which would greatly simplify both deleting and updating screens.

### Describe the solution / feature you'd like

When creating screens, they will be assigned an ID, for example, "clock", "date", or "notification1".

If another screen is sent with the same ID, even if it is in a different mode, it will replace the existing one.

When deleting, only the ID must be sent, without needing to know what type of screen it was.

Since the ID is not related to the icon, as many IDs as necessary can be created dynamically, which can be very useful for displaying notifications of any type of event in HA.

andrewjswan, 2024-10-21:
https://github.com/lubeda/EspHoMaTriXv2/tree/2024.6.pre?tab=readme-ov-file#screen_id

MiguelAngelLV, 2024-10-21:
> https://github.com/lubeda/EspHoMaTriXv2/tree/2024.6.pre?tab=readme-ov-file#screen_id

I know it, but it is only valid for reuse icons. If you want change a icon you first remove the old screen (sending the mode too) and send the new.

I propose a "unique id" across modes and icons.

andrewjswan, 2024-10-21:
> If you want change a icon you first remove the old screen (sending the mode too) and send the new.

As far as I remember, this is not true.

`icon_name: "icon1|some_id"` -> `icon_name: "icon2|some_id"` - Change icon1 to icon2 in screen with id = `some_id`

andrewjswan, 2024-10-21:
> I propose a "unique id" across modes and icons.

To be able to change the screen type by a given ID?

MiguelAngelLV, 2024-10-21:
> `icon_name: "icon1|some_id"` -> `icon_name: "icon2|some_id"` - Change icon1 to icon2 in screen with id = `some_id`
 Yeah, sorry, I mean different mode.
 
 For example send a simple Only Text (without icon) and I want replace it by a text progress.

andrewjswan, 2024-10-21:
> For example send a simple Only Text (without icon) and I want replace it by a text progress.

Got it, change the screen type. It's not difficult, but then we need to add screen_id to all screens, and then just delete it if there is one and create a new one.

andrewjswan, 2024-10-21:
That's how it is now, but the text doesn't have an icon or identifier.

MiguelAngelLV, 2024-10-21:
Yeah, I mean, we could add a ID field, but, if you don't fill the ID field, it automatically set it with icon.

lubeda, 2024-10-30:
The service calls in esphome are not as flexible as you think. E.G. there are no default values or optional values.

So you have to specify all parameters per service call.

lubeda, 2024-10-30:
I see no need to implement the "id"-thing, since i use  this tool for two years with the "icon"-way of selecting screen. But feel free to implement this feature.