lubeda/EspHoMaTriXv2 #252: [FEATURE REQUEST] Individual weekday color

MiguelAngelLV, 2024-10-29:
# Feature Request

### Is your feature request related to a problem?

### Describe the solution / feature you'd like

Set every weekday of a different color allow mark holidays or specials days

### Describe alternatives you've considered

### Additional context

MiguelAngelLV, 2024-10-30:
No, not a calendar, only a function as:

```c
set_weekday_color(day, r, g, b)
```

For example

```c
set_weekday_color(0, 0, 255)  //Default
set_weekday_color(5, 255, 0, 0)  //Saturday red
set_weekday_color(6, 255, 0, 0)  //Sunday red
``` 

So I can set the  weekend in differents color.

fthiery, 2024-12-28:
On my end, I would have loved to be able to set each pixel of each day to a specific color to display weather predictions, like: white first pixel if there is snow planned in the morning, blue second pixel of there is rain planned.

andrewjswan, 2024-12-29:
@fthiery try esphome.pixel_clock_**icon_prognosis_screen_rgb** https://github.com/lubeda/EspHoMaTriXv2/issues/149