lubeda/EspHoMaTriXv2 #254: [BUG] ESPHome 2024.11.0 - missing 2 glyphs error

UlrichThiess, 2024-11-21:
```
INFO ESPHome 2024.11.0
INFO Reading configuration /config/esphome/buro-ug-led-matrix.yaml...
INFO Updating https://github.com/lubeda/EspHoMaTriXv2@main
WARNING 
WARNING This will only work with esphome >= 2023.7.0
WARNING Please check the documentation and wiki https://github.com/lubeda/EspHoMaTriXv2
WARNING 
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

font: [source /config/esphome/buro-ug-led-matrix.yaml:81]
  
  Font EHMTXv2.ttf is missing 2 glyphs:
      
   (b'\\n')
      § (b'\\xa7').
  - file: EHMTXv2.ttf
    size: 16
    id: default_font
    glyphs: |-
      !?"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnÖÄÜöäüopqrstuvwxyz@<>ß§€/
```

trip5, 2024-11-21:
In the font you are using, `§` isn't available.  You can't use glyphs if they aren't present in the font file.  Delete it or... try using one of [my fonts](https://github.com/trip5/Matrix-Fonts)?  Character 167 `§` is present in all of mine.

Also, are you sure you should be using `size: 16`? You should probably be using 6 or 8.

As for `\\n` - this represents new-line.  It (a) doesn't need to be added and (b) shouldn't be possible to add... Maybe move the `/` after `:` and see if that at least gets rid of that error?  Maybe it's being interpreted as something else in ESPHome.

UlrichThiess, 2024-11-21:
I don't changed the code so i wonder why this happend with ESPHome 2024.11.0 ?¿?

Thank you, i will follow your hints.

UlrichThiess, 2024-11-21:
1. missing ```-``` after  ```glyphs: >-```
2. removed §
```
font:
  - file: EHMTXv2.ttf
    size: 16
    id: default_font
    glyphs: >-
      !?"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnÖÄÜöäüopqrstuvwxyz@<>ß€/
```

now it works and i will try your fonts - thanks again

closed 2024-11-21 by UlrichThiess

andrewjswan, 2024-11-21:
Maybe:
**ignore_missing_glyphs** (Optional, boolean): By default, warnings are generated for any glyph that is included in the defined glyphsets but not present in the configured font. Use this setting to suppress those warnings. Please note that the absence of any manually defined glyphs (specified via the glyphs option) will always be treated as an error and will not be influenced by this setting.

UlrichThiess, 2024-11-25:
Yes, the - at the end means not to add an CR/LF to the following line. So >- seems to be correct: The next line is used wihtout trailing spaces in it's length without adding CR/LF. Next (second) line was added with a space. I use this without problems in all of my template code.

riker09, 2024-12-19:
Sorry for commenting on closed issue, but this website has been very helpful to me when it comes to understand newlines in YAML: https://yaml-multiline.info/