lubeda/EspHoMaTriXv2 #255: [BUG] Text behind icon in action 'esphome.ulanzi_icon_screen'

hiscorebob, 2024-11-25:
# Bug report

## Describe the bug

I had to update my Ulanzi TC001 to ESPHome2024.11.1 because I had a missing weather state icon.
Now, when Home Assistant calls the service 'esphome.ulanzi_icon_screen' , the icon and text are displayed, but the text scrolls visibly behind the icon, which was not the case before.

## Additional information

- used Hardware:
  - ESP32 / Ulanzi TC001
  - EspHoMaTriXv2 version: main

## To Reproduce

Steps to reproduce the behavior:
Update Ulanzi to Esphome 2024.11.1
Use Home Assistant Action/Service Call 'esphome.ulanzi_icon_screen' with an icon and text

## Expected behavior

Scrolling Text should not be visible behind Icon

## Configuration

```
substitutions:
  devicename: ulanzi
  friendly_name: LED Matrix
  board: esp32dev
  # Pin definition from https://github.com/aptonline/PixelIt_Ulanzi 
  battery_pin: GPIO34 
  ldr_pin: GPIO35 
  matrix_pin: GPIO32 
  left_button_pin: GPIO26 
  mid_button_pin: GPIO27 
  right_button_pin: GPIO14 
  scl_pin: GPIO22 
  sda_pin: GPIO21
  buzzer_pin: GPIO15 

rtttl:
  output: rtttl_out

switch:
  - platform: template
    name: "Auto-Adjust Brightness"
    id: switch_autobrightness
    icon: mdi:brightness-auto
    restore_mode: RESTORE_DEFAULT_ON
    lambda: |-
      if (id(aab_enable)) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      lambda: |-
        id(aab_enable) = true;
    turn_off_action:
      lambda: |-
        id(aab_enable) = false;

globals:
  # aab = auto-adjustable brightness
  - id: aab_enable
    type: "bool"
    restore_value: true
    initial_value: "true"
  - id: aab_add
[... 300 lines cut ...]
[image]
[image]

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Services calls

```text
data_template:
  default_font: true
  icon_name: |-
      {{ states.weather.town.state }}

  text: "{{ states.weather.town.attributes.temperature }}°C"
  lifetime: 60
  screen_time: 15
  r: 255
  g: 255
  b: 255
action: esphome.ulanzi_icon_screen
```

gzxto1, 2024-11-25:
Similar issues here with 2024.11.1  I first fixed the issue with the Glyphs, but then I also had the issue with the icon_screen. No Icons anymore, just flickering... I did a roll-back to 2024.10.3

korttoma, 2024-11-25:
Icons are messed up for me also after updating to ESPHome  2024.11.1.
Icons are not displayed correctly and the scrolling text enters the icon area.

lubeda, 2024-11-25:
Hi, i think this is because of: https://github.com/esphome/esphome/pull/7631

lubeda, 2024-11-25:
@ hiscorebob 
Do you use the home-assistant-addon from esphome?
Can you send me your rain.gif?

I created a transparent gif which works in my setup! (esphome add-on and transparent gif as file)

hiscorebob, 2024-11-25:
> @ hiscorebob Do you use the home-assistant-addon from esphome? Can you send me your rain.gif?

Yes I use the official ESPHome Add-On in Home Assistant.
I have attached the rain.gif to this message.
[image]

gzxto1, 2024-11-26:
I'm also using the official esphome add-on and took all of my gifs from la metric.

kauthmbt, 2024-11-26:
Having the same issue with icons messed up but running ESPHome in a LXC container under Proxmox.

Example lameids from LaMetric:
- 22160 (animated)
- 2283 (animated)
- 12212 (static)

andrewjswan, 2024-11-26:
Could it possibly be related to this: https://github.com/esphome/esphome/pull/7566

lubeda, 2024-11-28:
I'm still searching for the cause and the cure. I found a major bug which wasted memory but no cure to this problem

https://github.com/esphome/issues/issues/6498

lubeda, 2024-11-28:
please try the latest branch, this is at least a workaround. @clydebarrow may find  a better solution.

hiscorebob, 2024-11-28:
Note: I have already updated to ESPHome 2024.11.2, but this has no effect.

Then I tried with the latest branch, and can confirm that this fixes the messed up icons.  Static and animated icons are now displayed correctly

But I did notice 2 new issues:
Compile error when leaving the following section under "icons":
```
   - id: sleep8x32
      #file: pixel-icons-8x8/sleep.gif
      url: https://user-images.githubusercontent.com/16407309/224850723-634c9b2d-55d9-44f2-9f93-765c0485b090.gif
```
(tried both url and file options)
the compile error log :
```
INFO Preparing icons, this may take some seconds.
WARNING  icon snowy has transparency!
WARNING  icon fog has transparency!
WARNING  icon sleep8x32 has transparency!
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 503, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 212, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 224, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 684, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 204, in wrapped
    await coro(conf)
  File "/data/external_components/0d442204/components/ehmtxv2/__init__.py", line 481, in to_code
    data[pos] = rgb >> 8
    ~~~~^^^^^
IndexError: list assignment index out of range
```
Disabling this icon, and the code compiles.

But the biggest issue, the Ulanzi does not display date or time, it only displays the notifications sent to it.
When the notifications expire, the screen turns black.

imCharlieB, 2024-12-02:
Im having this issue after a update a few days ago as well... But my icons dont even show they are like pixelated and text is running behind them.

gzxto1, 2024-12-07:
I just tried 2024.11.3, still the same issues. I don't have running text for my indoor/outdoor temperature display since there are only 4 digits needed, still the icons on the left are completely screwed. I restored back to 24.10.3 via WEB OTA upload.

clydebarrow, 2024-12-07:
Please test with this PR: https://github.com/esphome/esphome/pull/7908

It should solve the problem, but won't get merged until there is some feedback from testing the PR because it is a breaking change.

andrewjswan, 2024-12-07:
@clydebarrow Maybe it is possible to solve this problem at once? 
https://github.com/esphome/issues/issues/6446

imCharlieB, 2024-12-07:
> Please test with this PR: [esphome/esphome#7908](https://github.com/esphome/esphome/pull/7908)
> 
> It should solve the problem, but won't get merged until there is some feedback from testing the PR because it is a breaking change.

i get this error in compile INFO ESPHome 2024.11.2
INFO Reading configuration /config/esphome/mymatrix.yaml...
WARNING 
WARNING This will only work with esphome >= 2023.7.0
WARNING Please check the documentation and wiki https://github.com/lubeda/EspHoMaTriXv2
WARNING 
INFO Detected timezone 'America/New_York'
INFO Generating C++ source...
WARNING Image file /config/esphome/1pixel.gif has no animation frames
INFO Preparing icons, this may take some seconds.
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 503, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 212, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 224, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 684, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 204, in wrapped
    await coro(conf)
  File "/data/external_components/e7db8eef/components/ehmtxv2/__init__.py", line 377, in to_code
    cg.new_Pvariable(
  File "/esphome/esphome/cpp_generator.py", line 578, in new_Pvariable
    rhs = id_.type.new(*args)
          ^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/cpp_generator.py", line 762, in __call__
    call = CallExpression(self.base, *args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/cpp_generator.py", line 131, in __init__
    self.args = ExpressionList(*args)
                ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/cpp_generator.py", line 98, in __init__
    self.args = [safe_exp(arg) for arg in args]
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/cpp_generator.py", line 98, in <listcomp>
    self.args = [safe_exp(arg) for arg in args]
                 ^^^^^^^^^^^^^
  File "/esphome/esphome/cpp_generator.py", line 381, in safe_exp
    raise ValueError("Object is not an expression", obj)
ValueError: ('Object is not an expression', <class 'esphome.components.image.ImageRGB565'>)

hiscorebob, 2024-12-09:
> I just tried 2024.11.3, still the same issues. I don't have running text for my indoor/outdoor temperature display since there are only 4 digits needed, still the icons on the left are completely screwed. I restored back to 24.10.3 via WEB OTA upload.

Make sure to use "latest" tag in esphome code, like this:
```
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: latest
    refresh: 60s 
    components: [ ehmtxv2 ]   
```

and "clean Build files" before trying to compile again

hiscorebob, 2024-12-09:
> * The Clock is not shown by default anymore as before. I didn't have any special clock configuration. I guess now I need to trigger it via service action.

Same here, I managed to workaround this by adding an automation in Home Assistant which calls the actions "esphome.ulanzi_clock_screen" and "esphome.ulanzi_date_screen" every 24h with a lifetime of 1440.

gzxto1, 2024-12-10:
I change color and brightness every sunrise/sunset anyway, I've just embedded the clock_screen in this script.  Not showing the clock/date by default anymore seems the new strategy.. while Ulanzi is in general a clock, this integration is also used for generic matrix LED displays where a clock may not be needed. Would be good if the documentation could be updated reflecting this change. But anyway, all issues solved for me.  Happy runing all on ESPhome core component rather then to install AWTRIX from HACS and running yet another additional service just for this....

andker87, 2024-12-12:
> Same here, I managed to workaround this by adding an automation in Home Assistant which calls the actions "esphome.ulanzi_clock_screen" and "esphome.ulanzi_date_screen" every 24h with a lifetime of [...]

wouldn't it be possible, in the esphome configuration, to set an action like date_screen? so everything happens at boot time

```
esphome:
  comment: “Ulanzi Clock”
  on_boot:
    - HERE
```

lubeda, 2024-12-29:
It's better to add the clock in the on_empty_queue

closed 2024-12-29 by lubeda