lubeda/EspHoMaTriXv2 #261: [BUG] Can't compile in ESPHome 2024.12.0b1

skgsergio, 2024-12-11:
# Bug report

## Describe the bug

ESPHome 2024.12.0b1 fails to compile EspHoMaTriXv2 due to the introduction of this change: https://github.com/esphome/esphome/pull/7891

Log:
```
ERROR Unable to load component ehmtxv2:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 177, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/bee26781/components/ehmtxv2/__init__.py", line 331, in <module>
    CONFIG_SCHEMA = cv.All(font.validate_pillow_installed, EHMTX_SCHEMA)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'esphome.components.font' has no attribute 'validate_pillow_installed'
```

## Additional information

- used Hardware:
  - Ulanzi TC001 (ESP32)
  - EspHoMaTriXv2 version: 2024.12.0b1

## To Reproduce

Steps to reproduce the behavior:

- Install ESPHome 2024.12.0b1
- Try to compile project
- Fails with error

closed 2024-12-11 by lubeda

skgsergio, 2024-12-11:
Thanks, just updated it and working perfectly :)

MaxAu72, 2024-12-18:
I tried with: 
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: 2024.12.1
    refresh: 60s 
    components: [ ehmtxv2 ]

but I receive error:
INFO ESPHome 2024.12.0
INFO Reading configuration /config/esphome/esphome-ulanzi-clock.yaml...
WARNING 
WARNING Please check the documentation and wiki https://github.com/lubeda/EspHoMaTriXv2
WARNING 
INFO Detected timezone 'Europe/Rome'
Failed config

ehmtxv2: [source /config/esphome/esphome-ulanzi-clock.yaml:212]
  id: rgb8x32
  icons2html: True
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  time_format: %H:%M
  date_format: %d.%m.
  show_seconds: False
  
  [show_date] is an invalid option for [ehmtxv2]. Did you mean [show_dow]?
  show_date: True
  show_dow: False
  
  [allow_empty_screen] is an invalid option for [ehmtxv2]. Did you mean [on_next_screen], [on_expired_screen], [on_empty_queue]?
  allow_empty_screen: False
  default_font_id: default_font
  default_font_yoffset: 8

N3rdix, 2024-12-22:
> I tried with: external_components:
>
> * source:
>   type: git
>   url: https://github.com/lubeda/EspHoMaTriXv2
>   ref: 2024.12.1
>   refresh: 60s
>   components: [ ehmtxv2 ]
>
> but I receive error: INFO ESPHome [...]

Same error for me, both show_date and allow_empty_screen are unknown when using this ref 🤔

skgsergio, 2024-12-22:
You need to check changes, you might be updating from a very old version.

N3rdix, 2024-12-22:
> You need to check changes, you might be updating from a very old version.

thanks for the reply, but a very old version of what? The YAML? 
My understanding of using a git reference was that it would download that source automatically.
I always keep ESPHome up to date, so am I missing something?

MaxAu72, 2024-12-22:
I resolved, with:
ehmtxv2:
  id: rgb8x32
  icons2html: true
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  time_format: "%H:%M"
  date_format: "%d.%m."
  show_seconds: true
  #show_date: true
  #show_dow: false
  #allow_empty_screen: false
for to come back the Hour and Date I created HA automation:
  - action: esphome.esphome_ulanzi_clock_clock_screen
    data:
      default_font: true
      lifetime: 1440
      screen_time: 10
      r: 192
      g: 192
      b: 192
  - action: esphome.esphome_ulanzi_clock_date_screen
    data:
      default_font: false
      lifetime: 1440
      screen_time: 5
      r: 192
      g: 192
      b: 192