lubeda/EspHoMaTriXv2 #265: [BUG] Black screen after splash screen

mlukasek, 2024-12-21:
# Bug report

## Describe the bug

After updat to ESPHome 2024.12.0 I used ref: 2024.2.1 and it helped, but after the next update of ESPHome to 2024.12.2, I have black screen. Reset shows splash screen with color Ehmtx and heart and then screen goes black. The device is alive, as I can see sensor values in HA, can do OTA update, but screen shows nothing.

I noticed, that Auto adjust brightness cannot be turned off, but I do not know if it behaved this way before.

## Additional information

- used Hardware:
  - Ulanzi Clock
  - EspHoMaTriXv2 version: [2024.2.1]

## Configuration

```text
substitutions:
  devicename: ulanzi-clock-01
  friendly_name: LED Matrix
  board: esp32dev
  # Pin definition from https://github.com/aptonline/PixelIt_Ulanzi 
  battery_pin: GPIO34 
  ldr_pin: GPIO35 
  matrix_pin: GPIO32 
  left_button_pin: GPIO26 
  mid_button_pin: GPIO27 
  right_button_pin: GPIO14 
  scl_pin: GPIO22 
  sda_pin: GPIO21 

switch:
  - platform: template
    name: "Auto-Adjust Brightness"
    id: switch_autobrightness
    icon: mdi:brightness-auto
    restore_mode: RESTORE_DEFAULT_ON
    lambda: |-
      if (id(aab_enable)) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      lambda: |-
        id(aab_enable) = true;
    turn_off_action:
      lambda: |-
        id(aab_enable) = false;

globals:
  # aab = auto-adjustable brightness
  - id: aab_enable
    type: "bool"
    restore_value: true
    initial_value: "true"
  - id: aab_add
    type: int
    initial_value: '220'
  - id: aab_max
    type: int
    initial_value: '220'
  - id: aab_min
    type: int
    initial_value: '220'

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: 2024.12.1
    refresh: 60s 
    components: [ ehmtxv2 ]   

esphome:
  comment: "EHMTXv2 from LuBeDa"
  name: $devicename 
  project:
    name: "Ulanzi.EHMTXv2"
    version: "2.0.0"
  on_boot:
    then:
      - ds1307.read_time:

esp32:
  board: esp32dev

font:
  # Font included in this folder
  # - file: mateine.ttf
  #   size: 16
  #   id: default_font
  #   glyphs:  |
  #     !?"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnÖÄÜöäüopqrstuvwxyz@<>ß§€/
  # Matrix Clock Fonts
  # Grab these from https://github.com/trip5/Matrix-Fonts
  # See https://github.com/lubeda/EspHoMaTriXv2#font
  # Store in a "fonts" subfolder of your ESPHome config folder or change paths below
  - file: fonts/MatrixLight8.bdf
    id: default_font
    glyphs:  |
      ! "#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz°
  - file: fonts/MatrixChunky6.bdf
    id: special_font
    glyphs:  |
      ! "#$%&'()*+,-./0123456789:APMTapmt
  
binary_sensor:
  - platform: status
    name: "$devicename Status"
  - platform: gpio
    pin:
      number: $left_button_pin
      inverted: true
    name: "Left button"
  - platform: gpio
    pin: 
      inverted: true
      number: $mid_button_pin
      mode: INPUT_PULLUP
    name: "Middle button"
  - platform: gpio
    pin: 
      number: $right_button_pin
      inverted: true
    name: "Right button"

logger:
  level: WARN

output:
  - platform: gpio
    pin:
      number: GPIO15
      ignore_strapping_warning: true
    id: buzzer_pin

api:

sensor:
  - platform: sht3xd
    temperature:
      name: "$devicename Temperature"
    humidity:
      name: "$devicename Relative Humidity"
    update_interval: 60s
  - platform: adc
    pin: $battery_pin
    name: "$devicename Battery"
    id: battery_voltage
    update_interval: 10s
    device_class: battery
    accuracy_decimals: 0
    attenuation: auto
    filters:
      - sliding_window_moving_average:
          window_size: 15
          send_every: 15
          send_first_at: 1
      - multiply: 1.6
      - lambda: |-
          auto r = ((x - 3) / 0.69 * 100.00);
          if (r >= 100) return 100;
          if (r > 0) return r;
          if (r <= 0) return 1;
          return 0;
    unit_of_measurement: '%'
  - platform: adc
    id: light_sensor
    name: "$devicename Illuminance"
    pin: $ldr_pin
    update_interval: 10s
    attenuation: auto
    unit_of_measurement: lx
    device_class: illuminance
    accuracy_decimals: 0
    filters:
      - lambda: |-
          return (x / 10000.0) * 2000000.0 - 15 ;
    on_value:
      then:
        - lambda: |-
            if ( id(aab_enable) ) {
              int n = x / 4 + id(aab_add); // new_value
              if (n > id(aab_max)) n = id(aab_max);
              if (n < id(aab_min)) n = id(aab_min);
              int c = id(rgb8x32)->get_brightness(); // current value
              c = c>0?c:1 ; 
              int d = (n - c) * 100 / c; // diff in %
              if ( abs(d) > 2 ) id(rgb8x32)->set_brightness(n);
            }
      
ota:
  - platform: esphome
    password: "removed"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:

i2c:
  sda: $sda_pin
  scl: $scl_pin
  scan: true
  id: i2cbus

light:
  - platform: neopixelbus
    id: ehmtx_light
    type: GRB
    internal: true
    variant: WS2812
    pin: $matrix_pin
    num_leds: 256
    color_correct: [30%, 30%, 30%]
    gamma_correct: 2.0
    name: "$devicename Light"
    restore_mode: ALWAYS_OFF
    
time:
  - platform: homeassistant
    on_time_sync:
      then:
        ds1307.write_time:
  - platform: ds1307
    update_interval: never
    id: ehmtx_time

display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (y % 2 == 0) {
        return (y * 32) + x;
      }
      return (y * 32) + (31 - x);
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();
      
ehmtxv2:
  id: rgb8x32
  icons2html: true
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  time_format: "%H:%M"
  date_format: "%d.%m."
  show_seconds: false
  # Uncomment below if using the mateine font
  # default_font_id: default_font
  # special_font_id: default_font
  # Comment out below if using mateine font
  default_font_id: default_font
  default_font_yoffset: 8
  special_font_id: special_font
  special_font_yoffset: 6
  # until here
  boot_logo: "[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63519,63519,63519,63519,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63519,0,0,0,0,2016,0,0,0,0,0,0,0,0,0,0,31,0,0,0,0,0,0,0,0,0,63488,0,63488,0,0,0,63519,0,0,0,0,2016,2016,0,0,0,65514,0,65514,0,0,0,31,0,0,0,64512,0,0,64512,0,63488,63488,0,63488,63488,0,0,63519,63519,63519,0,0,2016,0,2016,0,65514,0,65514,0,65514,0,31,31,31,0,0,0,64512,64512,0,0,63488,63488,63488,63488,63488,0,0,63519,0,0,0,0,2016,0,2016,0,65514,0,65514,0,65514,0,0,31,0,0,0,0,64512,64512,0,0,0,63488,63488,63488,0,0,0,63519,63519,63519,63519,0,2016,0,2016,0,65514,0,65514,0,65514,0,0,0,31,31,0,64512,0,0,64512,0,0,0,63488,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]"
  icons: 
    - id: error
      lameid: 40530
    - id: home_assistant
      lameid: 47693
    - id: temperature
      lameid: 2056
    - id: lightbulb
      lameid: 1762
    - id: music
      lameid: 45625
    - id: phone
      lameid: 1232
    - id: car
      lameid: 2819
    - id: sleep8x32
      url: https://user-images.githubusercontent.com/16407309/224850723-634c9b2d-55d9-44f2-9f93-765c0485b090.gif 
    - id: weather_clear_night
      lameid: 53383
    - id: weather_cloudy
      frame_duration: 192
      lameid: 25991
    - id: weather_exceptional
      lameid: 16754
    - id: weather_fog
      lameid: 52167
    - id: weather_hail
      lameid: 53288
    - id: weather_lightning
      lameid: 23713
    - id: weather_lightning_rainy
      lameid: 49299
    - id: weather_partlycloudy
      frame_duration: 210
      lameid: 22160
    - id: weather_pouring
      lameid: 49300
    - id: weather_rainy
      lameid: 26565
    - id: weather_snowy
      lameid: 2289
    - id: weather_snowy_rainy
      lameid: 49301
    - id: weather_sunny
      lameid: 11201
    - id: weather_windy
      lameid: 15618
    - id: weather_windy_variant
      lameid: 15618
    - id: weather_cloudy_night
      lameid: 12195
  on_next_screen:
    - homeassistant.event:
        event: esphome.new_screen
        data_template:
          iconname: !lambda "return icon.c_str();"
          text: !lambda "return text.c_str();"

image:
   - file: 1pixel.gif
     id: breaking20237

animation:
  - file: 1pixel.gif
    id: breaking202371

```

## Logs

```text
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/ulanzi-clock-01.yaml...
INFO Updating https://github.com/lubeda/EspHoMaTriXv2@2024.12.1
WARNING 
WARNING Please check the documentation and wiki https://github.com/lubeda/EspHoMaTriXv2
WARNING 
INFO Detected timezone 'Europe/Prague'
INFO Detected timezone 'Europe/Prague'
INFO Starting log output from 192.168.1.158 using esphome API
```

andrewjswan, 2024-12-21:
https://github.com/lubeda/EspHoMaTriXv2/blob/e11dd39dd1ba407a5845a7ad9ff19658974e8c20/install/ulanzi-easy.yaml#L293-L297

mlukasek, 2024-12-21:
Thank you [andrewjswan](https://github.com/andrewjswan) this works, I can see calendar, I can see my own screens, but why the 2 previous default screens are missing?

andrewjswan, 2024-12-21:
There are no default screens in this firmware, you determine which screens will be visible, you can do this in ESPHome or in HA.

mlukasek, 2024-12-21:
Thanks, that explains the black screen without adding any additional.

lubeda, 2024-12-25:
Solution is here https://github.com/lubeda/EspHoMaTriXv2/issues/265#issuecomment-2558154499

closed 2024-12-25 by lubeda