lubeda/EspHoMaTriXv2 #266: [BUG] missing SConscript file error on 2024.12.1

fthiery, 2024-12-28:
# Bug report

Using 2024.12.1 with recent config file.

## Describe the bug

Compile fails.

## Additional information

- used Hardware:
  - ESP32 (ulanzi)
  - EspHoMaTriXv2 version: 2024.12.1

## To Reproduce

Run install

## Configuration

The yml file located in /install

## Logs

```text
INFO Compiling app...
Processing ulanzi (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing SPI
INFO Installing SPI
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

*** missing SConscript file '/data/cache/platformio/packages/framework-arduinoespressif32/tools/platformio-build.py'
File "/data/cache/platformio/platforms/espressif32/builder/frameworks/arduino.py", line 41, in <module>
```

andrewjswan, 2024-12-29:
It seems to me that this has nothing to do with `EspHoMaTriX` at all, it is a problem of `ESPHome` itself.

lubeda, 2024-12-29:
I also think this is not an esphome problem. Try "Clean Build Files" in the three dot menu of your esphome project an then  compile as usuall.

closed 2024-12-29 by lubeda

fthiery, 2024-12-29:
Already tried to clean, still fails. Sorry for the noise I'll report to esphome

fthiery, 2025-01-08:
For the record, reinstalling the ESPHome Builder HA add-on fixed it