lubeda/EspHoMaTriXv2 #267: [BUG] Compilation for ESP8266 fails when linking elf (multiple definition of `info')

BbMaj7, 2025-01-05:
# Compilation for ESP8266 fails when linking elf (multiple definition of `info')

## Description

Formerly this code was compiling without problems. I wanted to change font type and size and recompile. Fails with error:

```text
Linking .pioenvs/ehmtx8266/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lib/NONOSDK22x_190703/libmain.a(app_main.o):(.bss+0x4): multiple definition of `info'; .pioenvs/ehmtx8266/src/main.cpp.o:(.bss.info+0x0): first defined here
```

## Additional information

- used Hardware:
  - ESP8266
  - EspHoMaTriXv2 version: 2024.12.1

ESPHome version: 2024.12.2

## To Reproduce

1. Use configuration shown below
2. Compile

## Configuration

```python
substitutions:
  name: esphome-web-f11478
  devicename: ehmtx8266
  friendly_name: LEDMatrix2
  ledpin: GPIO00
  board: d1_mini
  loglevel: DEBUG

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: 2024.12.1
    refresh: 60s 
    components: [ ehmtxv2 ]   

esphome:
  name: $devicename
  comment: 8x32 RGB Display

# To have a "next url" for improv serial
web_server:
  port: 80

esp8266:
  board: $board

font:
  - file: _fonts/mateine.ttf
    id: default_font
    size: 16
    glyphs:  |-
       !"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@üöäÜÖÄß
    #glyphs:  |-
    #  !?"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@<>/

# Enable logging
logger:
  level: $loglevel

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

# Allow provisioning Wi-Fi via serial
improv_serial:

light:
  - platform: neopixelbus
    id: ehmtx_light
    type: GRB
    variant: WS2812
    pin: $ledpin
    num_leds: 256
    color_correct: [30%, 30%, 30%]
    name: "$devicename Light"
    restore_mode: ALWAYS_OFF
    on_turn_on:
      lambda: |-
         id(ehmtx_display)->set_enabled(false);
    on_turn_off:
       lambda: |-
         id(ehmtx_display)->set_enabled(true);

time:
  - platform: homeassistant
    id: ehmtx_time

display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (x % 2 == 0) {
        return (x * 8) + y;
      }
      return (x * 8) + (7 - y);
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();

sensor:
  - platform: uptime
    name: Uptime Sensor

animation:
  - id: animation_black
    file: "black.gif"

image:
  - id: image_black
    file: "black.gif"

ehmtxv2:
  id: rgb8x32
  show_seconds: true
  rainbow_interval: 32
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  icons2html: true
  default_font_id: default_font
  special_font_id: default_font
  special_font_yoffset: 8
  default_font_yoffset: 8

  icons: 
    - id: info
      lameid: 620
    - id: attention
      lameid: 555
    - id: error
      lameid: 26465
    - id: arrow_up
      lameid: 120
    - id: arrow_down
      lameid: 124
    - id: temperature_cold
      lameid: 8043
    - id: fullscreen
      file: sample8x32.gif
    - id: samplegif
      file: sample8x8.gif
```

## Logs

```text
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/esphome-web-f11478.yaml...
WARNING 
WARNING Please check the documentation and wiki https://github.com/lubeda/EspHoMaTriXv2
WARNING 
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Preparing icons, this may take some seconds.
INFO animation info with 1 frame(s)
INFO animation attention with 1 frame(s)
INFO animation error with 1 frame(s)
INFO animation arrow_up with 1 frame(s)
INFO animation arrow_down with 1 frame(s)
INFO animation temperature_cold with 1 frame(s)
INFO animation fullscreen with 2 frame(s)
INFO animation samplegif with 2 frame(s)
INFO EsphoMaTrix: wrote html-file with icon preview: /config/esphome/esphome-web-f11478.html
INFO List of icons for e.g. blueprint:

["info","attention","error","arrow_up","arrow_down","temperature_cold","fullscreen","samplegif",]

INFO Compiling app...
Processing ehmtx8266 (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
|-- NeoPixelBus @ 2.7.3
Linking .pioenvs/ehmtx8266/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lib/NONOSDK22x_190703/libmain.a(app_main.o):(.bss+0x4): multiple definition of `info'; .pioenvs/ehmtx8266/src/main.cpp.o:(.bss.info+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pioenvs/ehmtx8266/firmware.elf] Error 1
========================= [FAILED] Took 19.56 seconds =========================
```

lubeda, 2025-01-05:
Hi, try to name your first icon not "info", e.g. "error" should work

BbMaj7, 2025-01-06:
That does solve it. Thanks. Seems obvious now but I was unable to read that from the stacktrace.

closed 2025-01-06 by BbMaj7