lubeda/EspHoMaTriXv2 #271: [BUG] 'class esphome::EHMTX' has no member named 'fire_screen'

spunkt123, 2025-01-08:
# Bug report

## Describe the bug

When compiling the version 2024.12.1 i get this error: 

/config/esphome-holocube.yaml: In lambda function:
/config/esphome-holocube.yaml:277:16: error: 'class esphome::EHMTX' has no member named 'fire_screen'; did you mean 'force_screen'?
         id(rgb8x32)->fire_screen(360,30);
                ^~~~~~~~~~~
                force_screen

i use "id(rgb8x32)->fire_screen(360,30);" in an automation.
In older versions this worked.

## Additional information

- used Hardware:
  - ESP32 / Ulanzi TC001
  - EspHoMaTriXv2 version: 2024.12.1

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

i use "id(rgb8x32)->fire_screen(360,30);" in an automation.
In older versions this worked.

## Configuration

```text
(optional) The YAML you used in epshome without any password
```

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Services calls

```text
(optional) The YAML of your service calls
```

andrewjswan, 2025-01-08:
I didn't find any mention of this feature in the author's repository.
https://github.com/search?q=owner%3Alubeda%20fire_screen&type=code

spunkt123, 2025-01-08:
i found it in EHMTX.cpp: https://github.com/lubeda/EspHoMaTriXv2/blob/2024.12.1/components/ehmtxv2/EHMTX.cpp#L2125

I suspect that "def USE_Fireplugin" is needed somewhere. However, I have very little idea how it works. :(

andrewjswan, 2025-01-09:
> But what does this mean?

If ESP32 or ESP32S3 and ESP IDF framework, then remove this function from the config.

lubeda, 2025-01-10:
Thats it! If you need more colourful animations take a look at https://blueforcer.github.io/awtrix3/#/README