lubeda/EspHoMaTriXv2 #273: [QUESTION] Timer not counting down?

allenm518, 2025-01-20:
# Question

I am working on getting a timer for my washer and dryer to show up on my custom home matrix screen. I have a timer in home assistant that calculates remaining time from the dryer sensor. I'm using this timer in the example from the wiki. The timer shows up on the matrix screen appropriately but does not update or countdown. It remains with the same time as what initially populates and does not change at all. It does not countdown while on screen and does not change between screen cycles.

My automation yaml:

```text
alias: Matrix dryer 2 test
description: ""
triggers:
  - trigger: state
    entity_id:
      - timer.dryer_timer
    to: active
conditions: []
actions:
  - data:
      default_font: true
      icon_name: timer
      screen_time: 10
      lifetime: 60
      r: 200
      b: 200
      g: 50
      text: |-
        {{ (state_attr('timer.dryer_timer', 'finishes_at') | as_datetime -
            now()).total_seconds() | int(0) | timestamp_custom('%H:%M:%S', false)}}
    action: esphome.matrix_icon_screen
  - data:
      icon_name: timer
    action: esphome.matrix_force_screen
  - delay:
      hours: 0
      minutes: 0
      seconds: 1
      milliseconds: 0
  - data:
      skip_condition: false
    target:
      entity_id: automation.matrix_dryer_2_test
    action: automation.trigger
mode: restart
```

## Additional information

- used Hardware:
  - ESP32
  - ESPHome  2024.12.2 
  - EspHoMaTriXv2 version: 2024.12.1
  - HA 2025.1.0

## Additional context

I have also tried replacing the text that gets sent to the matrix with a direct measure of the dryer.sensor end time and it ends up in the same results: initial time measure that does not change or count down.

```text
{{ ((state_attr('sensor.dryer', 'end_time') | as_datetime) -
    now()).total_seconds() | timestamp_custom('%H:%M:%S', false) | default(
    'test', true) }}

```

## Matrix Config yaml
```text
substitutions:
  devicename: matrix
  friendly_name: LED Matrix
  board: esp32dev
  matrix_pin: GPIO32
  scl_pin: GPIO22 
  sda_pin: GPIO21 

switch:
  - platform: template
    name: "Auto-Adjust Brightness"
    id: switch_autobrightness
    icon: mdi:brightness-auto
    restore_mode: RESTORE_DEFAULT_ON
    lambda: |-
      if (id(aab_enable)) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      lambda: |-
        id(aab_enable) = true;
    turn_off_action:
      lambda: |-
        id(aab_enable) = false;

image:
   - file: 1pixel.gif
     id: breaking20237

animation:
  - file: 1pixel.gif
    id: breaking202371
    
globals:
  # aab = auto-adjustable brightness
  - id: aab_enable
    type: "bool"
    restore_value: true
    initial_value: "true"
  - id: aab_add
    type: int
    initial_value: '10'
  - id: aab_max
    type: int
    initial_value: '220'
  - id: aab_min
    type: int
    initial_value: '20'

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: 2024.12.1
    refresh: 60s 
    components: [ ehmtxv2 ]  

esphome:
  comment: "EHMTXv2 from LuBeDa"
  name: $devicename 
  project:
    name: "Ulanzi.EHMTXv2"
    version: "2.0.0"

esp32:
  board: esp32dev

font:
  # Matrix Clock Fonts
  # Grab these from https://github.com/trip5/Matrix-Fonts
  # See https://github.com/lubeda/EspHoMaTriXv2#font
  # Store in a "fonts" subfolder of your ESPHome config folder or change paths below
  - file: fonts/MatrixLight8.bdf
    id: default_font
    glyphs:  |
      ! "#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz°
  - file: fonts/MatrixChunky6.bdf
    id: special_font
    glyphs:  |
      ! "#$%&'()*+,-./0123456789:APMTapmt
  
binary_sensor:
  - platform: status
    name: "$devicename Status"

logger:
  level: WARN

output:

api:

ota:
  - platform: esphome

wifi:
  ssid: Ziggy
  password: swwts_1972!

web_server:

i2c:
  sda: $sda_pin
  scl: $scl_pin
  scan: true
  id: i2cbus

light:
  - platform: neopixelbus
    id: ehmtx_light
    type: GRB
    internal: true
    variant: WS2812
    pin: $matrix_pin
    num_leds: 256
    color_correct: [30%, 30%, 30%]
    gamma_correct: 2.0
    name: "$devicename Light"
    restore_mode: ALWAYS_OFF
    
time:
  - platform: homeassistant
    id: ehmtx_time

display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (x % 2 == 0) {
        return (x * 8) + y;
      }
      return (x * 8) + (7 - y);
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();
      
ehmtxv2:
  id: rgb8x32
  icons2html: true
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  time_format: "%H:%M"
  date_format: "%b %d"
  show_seconds: true
  week_start_monday: false # false equals sunday
  default_font_id: default_font
  #i changed the font to the small size 6 
  default_font_yoffset: 8
  special_font_id: special_font
  special_font_yoffset: 7
  boot_logo: "[0,0,0,0,0,0,0,0,0,0,0,0,16777215,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,16777215,16777215,0,0,0,16777215,16777215,0,0,0,0,16777215,16777215,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,16777215,16777215,0,16777215,0,16777215,16777215,16777215,0,0,0,0,16777215,16777215,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,16777215,0,0,16777215,0,16777215,0,16777215,16777215,0,0,16777215,16777215,0,16777215,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,16777215,16777215,0,0,16777215,16777215,0,0,0,16777215,0,0,16777215,16777215,0,16777215,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,16777215,16777215,0,0,0,0,0,0,0,0,16777215,16777215,0,16777215,0,16777215,16777215,16777215,16777215,16777215,16777215,16777215,16777215,16777215,16777215,0,0,0,0,0,0,16777215,16777215,0,0,0,0,0,0,0,0,0,0,16777215,16777215,0,0,0,0,16777215,0,0,0,0,0,0,0,0,0,0,0,0,0,16777215,0,0,0,0,0,0,0,0,0,0,0,0,16777215,0,0,0,0,16777215,0,0,0,0,0,0,0,0,0,0,0]"
  icons:
    - id: error
      lameid: 40530
    - id: home_assistant
      lameid: 47693
    - id: temperature
      lameid: 2056
    - id: lightbulb
      lameid: 1762
    - id: washer
      lameid: 56004
    - id: dryer
      lameid: 56907
  on_empty_queue: 
     then: 
       - lambda: |- 
           id(rgb8x32).set_infotext_color(20,20,20,20,20,20,false,1); 
           id(rgb8x32)->expand_icon_to_9(3);
           id(rgb8x32).icon_clock("calendar|day#2",1440,10,false); 
           id(rgb8x32).set_today_color(0,128,255);

```

andrewjswan, 2025-01-20:
Your trigger works once, displays text with the timer state, and never updates this state again. You need to change the automation in HA so that it updates the screen when the timer changes.

allenm518, 2025-01-20:
Ok thank you I'll look into that. I'm learning HA at the same time. I'll follow up with results