lubeda/EspHoMaTriXv2 #274: [QUESTION] Compiling error after latest update

Worschtbrot, 2025-01-20:
# Question

after the last esphome update, i get the below error message when trying to update my ESP. 
Is there any solution around?

## Additional information

- used Hardware:
  - ESP8266

  - EspHoMaTriXv2 version: latest

## Logs
ERROR Unable to load component ehmtxv2:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 177, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/0d442204/components/ehmtxv2/__init__.py", line 325, in <module>
    CONFIG_SCHEMA = cv.All(font.validate_pillow_installed, EHMTX_SCHEMA)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'esphome.components.font' has no attribute 'validate_pillow_installed'
Failed config

ehmtxv2: [source /config/esphome/edesp2866.yaml:150]
  
  Component not found: ehmtxv2.

## Additional context

Add any other context about the problem here.

andrewjswan, 2025-01-20:
https://github.com/lubeda/EspHoMaTriXv2/issues/262

clfberlin, 2025-01-23:
Thank you. Just to copy the solution here:
```
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: 2024.12.1
    refresh: 60s 
    components: [ ehmtxv2 ]   
```
The change is ref: 2024.12.1 instead of ref: latest.
The official latest release is 2024.12.0

Worschtbrot, 2025-01-29:
sorry, still having trouble to get it running again... In the logs I can see, that after:

EHMTXv2:852]: Setting up services   it throws an exception:

[09:47:51]User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

[09:47:51]Unhandled C++ exception: OOM
[09:47:51]
[09:47:51]>>>stack>>>
[09:47:51]
[09:47:51]ctx: cont
[09:47:51]sp: 3ffffca0 end: 3fffffd0 offset: 0010
[09:47:51]3ffffcb0:  00000001 00000020 3fff8e04 40244cd8  
[09:47:51]3ffffcc0:  3fffbdb4 3fff5c0c 3ffffcd0 4021be2f <
[09:47:51]3ffffcd0:  3ffffcd8 00000009 6e6f6369 6d616e5f  
[09:47:51]3ffffce0:  6f5f0065 00006666 3ffffcf0 00000004 <
[09:47:51]3ffffcf0:  74786574 745f0000 00656d69 40100cba  
[09:47:51]3ffffd00:  3ffffd08 00000008 676f7270 73736572  
[09:47:51]3ffffd10:  746e6f00 3ffe8700 3ffffd20 00000008 <
[09:47:51]3ffffd20:  6566696c 656d6974 746e6f00 40100d00  
[09:47:51]3ffffd30:  3ffffd38 0000000b 65726373 745f6e65  
[09:47:51]3ffffd40:  00656d69 00000000 3ffffd50 0000000c <
[09:47:51]3ffffd50:  61666564 5f746c75 746e6f66 00000400  
[09:47:51]3ffffd60:  3ffffd68 00000001 00000072 5f746c75  
[09:47:51]3ffffd70:  746e6f66 00000000 3ffffd80 00000001 <
[09:47:51]3ffffd80:  666f0067 74657366 3fff9d00 40101514  
[09:47:51]3ffffd90:  3ffffd98 00000001 3fff0062 40100cba  
[09:47:51]3ffffda0:  3fffff20 3ffe8700 3ffffdb0 00000001 <
[09:47:51]3ffffdb0:  00000062 3ffe8700 3fff99f8 40100cba  
[09:47:51]3ffffdc0:  3fffbdb4 00000000 00000000 7263735f  
[09:47:51]3ffffdd0:  006e6565 00726f00 3fff8e0c 00000014  
[09:47:51]3ffffde0:  00000014 63735f72 6e656572 40100c00  
[09:47:51]3ffffdf0:  40217de8 00000000 40215ad4 00000000  
[09:47:51]3ffffe00:  40215764 00000000 40217fc4 00000000  
[09:47:51]3ffffe10:  40217324 00000000 40216f8c 00000000  
[09:47:51]3ffffe20:  402173e4 00000000 40217248 00000000  
[09:47:51]3ffffe30:  40217114 00000000 40219270 00000000  
[09:47:51]3ffffe40:  4021946c 00000000 4021968c 00000000  
[09:47:51]3ffffe50:  40219884 00000000 40219a90 00000000  
[09:47:51]3ffffe60:  40219fcc 00000000 40215cf0 00000000  
[09:47:51]3ffffe70:  40216670 00000000 40215ab4 00000000  
[09:47:51]3ffffe80:  40215a94 00000000 40215f24 00000000  
[09:47:51]3ffffe90:  40215ed4 00000000 402157b8 00000000  
[09:47:51]3ffffea0:  402157f0 00000000 40215a08 00000000  
[09:47:51]3ffffeb0:  4021587c 00000000 402158b8 00000000  
[09:47:51]3ffffec0:  40215a24 00000000 40215a5c 00000000  
[09:47:51]3ffffed0:  40215b84 00000000 40215bd0 00000000  
[09:47:51]3ffffee0:  40215934 00000000 40215914 00000000  
[09:47:51]3ffffef0:  402159e8 00000000 40215af0 00000000  
[09:47:51]3fffff00:  40215b0c 00000000 40215c84 00000000  
[09:47:51]3fffff10:  40215fc4 00000000 40215f74 00000000  
[09:47:51]3fffff20:  40216ccc 00000000 3fffff40 00000040  
[09:47:51]3fffff30:  3fff520c 3ffe8e53 3fff520c 4021386b  
[09:47:51]3fffff40:  3ffe9404 3fffbdd4 000000f0 3fff2368  
[09:47:51]3fffff50:  4a200200 00000000 00000000 00000044  
[09:47:51]3fffff60:  3fff5c0c 00000010 3fff5c0c 4023263a  
[09:47:51]3fffff70:  3fff520c 0000000f 3fff23c4 40231ef5  
[09:47:51]3fffff80:  0000000a 0000000a 3fff82ec feefeffe  
[09:47:51]3fffff90:  00000102 00000024 00000009 00000000  
[09:47:51]3fffffa0:  feefeffe feefeffe feefeffe 3fff2878  
[09:47:51]3fffffb0:  3fffdad0 00000000 3fff284c 402450d0  
[09:47:51]3fffffc0:  feefeffe feefeffe 3fffdab0 40100535  
[09:47:51]<<<stack<<<
[09:47:51]
[09:47:51]last failed alloc call: 4021BE2F(260)
[09:47:51]
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

And then reboots....

Any idea how to fix? It was working like a charm in November :-(

closed 2025-02-24 by lubeda