lubeda/EspHoMaTriXv2 #275: [BUG] Can't compile in ESPHome 2025.2.0b1

skgsergio, 2025-02-12:
# Bug report

## Describe the bug

ESPHome 2025.2.0b1 fails to compile EspHoMaTriXv2 I due to the introduction of this change: https://github.com/esphome/esphome/pull/7908

I guess the solution is adapting this changes: https://github.com/esphome/esphome/pull/7908/files#diff-4f1cb0c17ef5d328ea36838b7aadaeb4295d06083e8987379b6ce92620dd3204R83-R100

Here: https://github.com/lubeda/EspHoMaTriXv2/blob/2024.12.1/components/ehmtxv2/__init__.py#L503-L519

Log:

```python
INFO Preparing icons, this may take some seconds.
INFO animation ha with 7 frame(s)
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1049, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1036, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 503, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 212, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 224, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 674, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 204, in wrapped
    await coro(conf)
  File "/data/external_components/3d742d23/components/ehmtxv2/__init__.py", line 509, in to_code
    cg.new_Pvariable(
  File "/esphome/esphome/cpp_generator.py", line 578, in new_Pvariable
    rhs = id_.type.new(*args)
          ^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/cpp_generator.py", line 762, in __call__
    call = CallExpression(self.base, *args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/cpp_generator.py", line 131, in __init__
    self.args = ExpressionList(*args)
                ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/cpp_generator.py", line 98, in __init__
    self.args = [safe_exp(arg) for arg in args]
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/cpp_generator.py", line 98, in <listcomp>
    self.args = [safe_exp(arg) for arg in args]
                 ^^^^^^^^^^^^^
  File "/esphome/esphome/cpp_generator.py", line 381, in safe_exp
    raise ValueError("Object is not an expression", obj)
ValueError: ('Object is not an expression', <class 'esphome.components.image.ImageRGB565'>)
```

## Additional information

- used Hardware:
  - Ulanzi TC001 (ESP32)
  - EspHoMaTriXv2 version: 2025.2.0b1

## To Reproduce

Steps to reproduce the behavior:

- Install `ESPHome Device Builder (beta)` add-on version `2024.12.0b1`
- Try to compile project
- Fails with error

lubeda, 2025-02-12:
Again an esphome update crashed my component. This is a more complex problem. The internal handling of animations in esphome change. I will investigate...

lubeda, 2025-02-12:
@skgsergio pointed to the cause, of the trouble. https://github.com/esphome/esphome/pull/7908/files#diff-4f1cb0c17ef5d328ea36838b7aadaeb4295d06083e8987379b6ce92620dd3204R83-R100

lubeda, 2025-02-12:
https://github.com/clydebarrow/esphome/issues/123

lubeda, 2025-02-12:
https://github.com/esphome/esphome-docs/pull/4663

andrewjswan, 2025-02-13:
It looks like the file loading procedure has been unified into a single function:

https://github.com/clydebarrow/esphome/blob/649ca5381752e3fe2d88eee5ca601c1c6e297884/esphome/components/image/__init__.py#L512-L581

jesserockz, 2025-02-13:
https://github.com/lubeda/EspHoMaTriXv2/blob/4610a81eab88e5ce5548647b7f48db40ec3f1210/components/ehmtxv2/__init__.py#L515

This should be a minor change from `espImage.IMAGE_TYPE["RGB565"],` to `espImage.get_image_type_enum("RGB565")` but I cannot verify that is all that is needed.

Unfortunately for ESPHome to grow as a project, breaking changes must be made which can affect `external_components` like this one. It is always a good idea to run a daily CI on any repo that has external components to be notified ASAP when a change is merged into ESPHome dev that will "break" external components

skgsergio, 2025-02-14:
> This should be a minor change from espImage.IMAGE_TYPE["RGB565"], to espImage.get_image_type_enum("RGB565") but I cannot verify that is all that is needed.

I'll try to validate it if I have some time.

> Unfortunately for ESPHome to grow as a project, breaking changes must be made which can affect external_components like this one.

Hey man, don't worry happy to report failures and not pushing to be resolved ASAP. Also the good thing is you can have devices compiled with old versions of ESPHome running in newer versions so is not that it impacts much.

> It is always a good idea to run a daily CI on any repo that has external components to be notified ASAP when a change is merged into ESPHome dev that will "break" external components.

I'm the CI, also reported last version breaking change 😹

skgsergio, 2025-02-14:
@lubeda as @jesserockz hinted that part works with that change.

However more changes are required, will try to check how to fix it:
```cpp
src/esphome/components/ehmtxv2/EHMTX_icons.cpp: In constructor 'esphome::EHMTX_Icon::EHMTX_Icon(const uint8_t*, int, int, uint32_t, esphome::image::ImageType, std::__cxx11::string, bool, uint16_t)':
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:7:73: error: no matching function for call to 'esphome::animation::Animation::Animation(const uint8_t*&, int&, int&, uint32_t&, esphome::image::ImageType&)'
       : Animation(data_start, width, height, animation_frame_count, type)
                                                                         ^
In file included from src/esphome.h:5,
                 from src/esphome/components/ehmtxv2/EHMTX_icons.cpp:1:
src/esphome/components/animation/animation.h:11:3: note: candidate: 'esphome::animation::Animation::Animation(const uint8_t*, int, int, uint32_t, esphome::image::ImageType, esphome::image::Transparency)'
   Animation(const uint8_t *data_start, int width, int height, uint32_t animation_frame_count, image::ImageType type,
   ^~~~~~~~~
src/esphome/components/animation/animation.h:11:3: note:   candidate expects 6 arguments, 5 provided
src/esphome/components/animation/animation.h:9:7: note: candidate: 'constexpr esphome::animation::Animation::Animation(const esphome::animation::Animation&)'
 class Animation : public image::Image {
       ^~~~~~~~~
src/esphome/components/animation/animation.h:9:7: note:   candidate expects 1 argument, 5 provided
src/esphome/components/animation/animation.h:9:7: note: candidate: 'constexpr esphome::animation::Animation::Animation(esphome::animation::Animation&&)'
src/esphome/components/animation/animation.h:9:7: note:   candidate expects 1 argument, 5 provided
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:13:11: error: 'class esphome::EHMTX_Icon' has no member named 'transparent_'; did you mean 'transparency_'?
     this->transparent_=false;
           ^~~~~~~~~~~~
           transparency_
```

skgsergio, 2025-02-14:
Fixed everything: https://github.com/lubeda/EspHoMaTriXv2/pull/277

lubeda, 2025-02-14:
@jesserockz 
> Unfortunately for ESPHome to grow as a project, breaking changes must be made which can affect `external_components` like this one. It is always a good idea to run a daily CI on any repo that has [...]

Well, i love esphome! It is the coolest way to create "embed solutions" with wlan. I grew up in a world of atmel microcontrollers (oh man, am i old ;-)) So i really love it.

But its hard (impossible) for me to maintain ehmtx at a professional level. But @skgsergio found a fix for this bug and i will try to figure out how to make daily builds with the beta esphome, so i can act earlier.

Keep esphome growing and  i will try to keep ehmtx alive.

Thanx

Ludger

skgsergio, 2025-02-15:
Let's close this, it is solved in [`2025.2.0`](https://github.com/lubeda/EspHoMaTriXv2/tree/2025.2.0) branch 🚀

closed 2025-02-15 by skgsergio