lubeda/EspHoMaTriXv2 #283: [BUG] Font not rendering on 2025.3.0 beta

skgsergio, 2025-03-13:
# Bug report

## Describe the bug

Project builds and its uploaded to the device, screens work normally but font doesn't render 🙀

PS: Is that time of the month 😹

## Additional information

- used Hardware:
  - Ulanzi TC001 (ESP32)
  - EspHoMaTriXv2 version: 2025.3.0b2

## To Reproduce

Steps to reproduce the behavior:

- Install `ESPHome Device Builder (beta)` add-on version `2025.3.0b2`
- Compile and upload the project
- See how the clock is fontless

## Screenshots

(please, ignore all the cat hair)

[image]

[image]

skgsergio, 2025-03-13:
Starting to look as an upstream issue.

It fails using MatrixChunky fonts that is what I was using. Calcium and emthv2 fonts work.

```
font:
  # - id: font_matrixchunky6
  #   file: resources/fonts/MatrixChunky6.ttf
  #   size: 6
  #   glyphs: |-
  #     |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ

  # - id: font_matrixchunky8
  #   file: resources/fonts/MatrixChunky8.ttf
  #   size: 8
  #   glyphs: |-
  #     |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ

  - id: font_calcium
    file: resources/fonts/Calcium.ttf
    size: 16
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ

  - id: font_ehmtxv2
    file: resources/fonts/EHMTXv2.ttf
    size: 16
    glyphs: |-
      |!?"%()+*=,-_.:°€$@<>\/µ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÚÑáéíóúñ
```

skgsergio, 2025-03-13:
Definitely an upstream issue, look at the width and height in the GlyphData.

- 2025.3.0 beta
```
DEBUG Running to_code in esphome.components.font (num 16)
DEBUG Adding: // font:
DEBUG Adding: //   id: font_matrixchunky6
//   file:
//     path: /config/esphome/resources/fonts/MatrixChunky6.ttf
//     type: local
//   size: 6
//   glyphs:
//   - '|¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ'
//   glyphsets: []
//   ignore_missing_glyphs: false
//   bpp: 1
//   extras: []
//   raw_data_id: uint8_t_id
//   raw_glyph_id: font_glyphdata_id
DEBUG Adding: static const uint8_t uint8_t_id[] PROGMEM = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};
DEBUG Registered variable uint8_t_id of type uint8_t
DEBUG Adding: static const font::GlyphData font_glyphdata_id[] = {font::GlyphData{
  .a_char = (const uint8_t *)" ",
  .data = uint8_t_id + 0,
  .advance = 5,
  .offset_x = 0,
  .offset_y = 5,
  .width = 1,
  .height = 1,
}, font::GlyphData{
  .a_char = (const uint8_t *)"!",
  .data = uint8_t_id + 1,
  .advance = 3,
  .offset_x = 0,
  .offset_y = 5,
  .width = 1,
  .height = 1,
}, [...], font::GlyphData{
  .a_char = (const uint8_t *)"\342\202\254",
  .data = uint8_t_id + 101,
  .advance = 9,
  .offset_x = 0,
  .offset_y = 5,
  .width = 1,
  .height = 1,
}};
DEBUG Registered variable font_glyphdata_id of type font::GlyphData
DEBUG Adding global: font::Font *font_matrixchunky6;
DEBUG Adding: font_matrixchunky6 = new font::Font(font_glyphdata_id, 102, 6, 6, 1);
DEBUG Registered variable font_matrixchunky6 of type font::Font
DEBUG  -> finished
```

- 2025.2.2
```
DEBUG Running to_code in esphome.components.font (num 16)
DEBUG Adding: // font:
DEBUG Adding: //   id: font_matrixchunky6
//   file:
//     path: /config/resources/fonts/MatrixChunky6.ttf
//     type: local
//   size: 6
//   glyphs:
//   - '|¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ'
//   glyphsets: []
//   ignore_missing_glyphs: false
//   bpp: 1
//   extras: []
//   raw_data_id: uint8_t_id
//   raw_glyph_id: font_glyphdata_id
DEBUG Adding: static const uint8_t uint8_t_id[] PROGMEM = {0x00, 0x00, 0xA8, 0x80, 0xAA, 0x00, 0x00, 0x6C, 0xE6, 0xC0, 0xA2, 0x48, 0xA0, 0x52, 0x44, 0x00, 0x89, 0x28, 0x00, 0xA4, 0xA0, 0x00, 0x4E, 0x40, 0x00, 0xA0, 0xE0, 0x00, 0x80, 0x22, 0x48, 0x80, 0xEA, 0xAA, 0xE0, 0x4C, 0x44, 0xE0, 0xE2, 0xE8, 0xE0, 0xE2, 0xE2, 0xE0, 0xAA, 0xE2, 0x20, 0xE8, 0xE2, 0xE0, 0xE8, 0xEA, 0xE0, 0xE2, 0x22, 0x20, 0xEA, 0xEA, 0xE0, 0xEA, 0xE2, 0xE0, 0x88, 0x00, 0x24, 0x84, 0x20, 0xE0, 0xE0, 0x00, 0x84, 0x24, 0x80, 0xE2, 0x40, 0x40, 0x76, 0xAD, 0x0E, 0x00, 0xEA, 0xEA, 0xA0, 0xEA, 0xCA, 0xE0, 0xEA, 0x8A, 0xE0, 0xCA, 0xAA, 0xE0, 0xE8, 0xE8, 0xE0, 0xE8, 0xE8, 0x80, 0xE4, 0x2D, 0x2F, 0x00, 0x94, 0xBD, 0x29, 0x00, 0xE4, 0x44, 0xE0, 0x22, 0x2A, 0xE0, 0xAA, 0xCA, 0xA0, 0x88, 0x88, 0xE0, 0x8B, 0x6A, 0xA2, 0x88, 0x00, 0x96, 0xAD, 0x29, 0x00, 0xF4, 0xA5, 0x2F, 0x00, 0xEA, 0xE8, 0x80, 0xF4, 0xA5, 0x4D, 0x00, 0xEA, 0xCA, 0xA0, 0x68, 0xE2, 0xC0, 0xE4, 0x44, 0x40, 0xAA, 0xAA, 0xE0, 0xAA, 0xAE, 0x40, 0xAA, 0xAA, 0xBE, 0x50, 0x00, 0xAA, 0x4A, 0xA0, 0xAA, 0xE4, 0x40, 0xE2, 0x48, 0xE0, 0x88, 0x42, 0x20, 0xE0, 0xE6, 0xAE, 0x00, 0x88, 0xEA, 0xE0, 0xE8, 0x8E, 0x00, 0x22, 0xEA, 0xE0, 0xEA, 0xCE, 0x00, 0x64, 0xE4, 0x40, 0xEA, 0xE2, 0x60, 0x88, 0xEA, 0xA0, 0x40, 0xC4, 0xE0, 0x20, 0x22, 0xAE, 0x8A, 0xAC, 0xA0, 0xC4, 0x44, 0xE0, 0xFA, 0xAA, 0xAA, 0x00, 0xEA, 0xAA, 0x00, 0xEA, 0xAE, 0x00, 0xEA, 0xAE, 0x80, 0xEA, 0xAE, 0x20, 0xE8, 0x88, 0x00, 0xE8, 0x6E, 0x00, 0x4E, 0x44, 0x60, 0xAA, 0xAE, 0x00, 0xAA, 0xE4, 0x00, 0xAA, 0xAF, 0x94, 0x00, 0xA6, 0xCA, 0x00, 0xAA, 0xE2, 0x60, 0xE6, 0xCE, 0x00, 0xAA, 0x80, 0x8A, 0x80, 0xEA, 0xE0, 0x00, 0xC4, 0x60, 0x00, 0xE6, 0xE0, 0x00, 0xAA, 0xE8, 0x40, 0x42, 0xE0, 0x24, 0xEA, 0xEA, 0x2E, 0x8E, 0x8E, 0x24, 0xE4, 0x4E, 0xF0, 0x25, 0xAB, 0x48, 0x24, 0x0E, 0xAE, 0x24, 0xE6, 0xAE, 0x24, 0xEA, 0xCE, 0x24, 0x0C, 0x4E, 0xE0, 0xEA, 0xAA, 0x24, 0x0E, 0xAE, 0x24, 0x0A, 0xAE, 0x64, 0xE4, 0x60};
DEBUG Registered variable uint8_t_id of type uint8_t
DEBUG Adding: static const font::GlyphData font_glyphdata_id[] = {font::GlyphData{
  .a_char = (const uint8_t *)" ",
  .data = uint8_t_id + 0,
  .offset_x = 0,
  .offset_y = 2,
  .width = 4,
  .height = 4,
}, font::GlyphData{
  .a_char = (const uint8_t *)"!",
  .data = uint8_t_id + 2,
  .offset_x = 0,
  .offset_y = 0,
  .width = 2,
  .height = 6,
}, [...], font::GlyphData{
  .a_char = (const uint8_t *)"\342\202\254",
  .data = uint8_t_id + 303,
  .offset_x = 0,
  .offset_y = 0,
  .width = 4,
  .height = 6,
}};
DEBUG Registered variable font_glyphdata_id of type font::GlyphData
DEBUG Adding global: font::Font *font_matrixchunky6;
DEBUG Adding: font_matrixchunky6 = new font::Font(font_glyphdata_id, 102, 6, 6, 1);
DEBUG Registered variable font_matrixchunky6 of type font::Font
DEBUG  -> finished
```

skgsergio, 2025-03-13:
Opened an upstream issue: https://github.com/esphome/issues/issues/6863

closed 2025-03-13 by skgsergio

skgsergio, 2025-03-13:
I'm away from home right now, but tried anyway:

```
Failed config

font: [source /config/esphome/pixel-clock.yaml:92]
  
  Codepoints in bitmap fonts must be in the range 0-255.
  - id: font_matrixchunky6
    file: resources/fonts/MatrixChunky6.bdf
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ
  - id: font_matrixchunky8
    file: resources/fonts/MatrixChunky8.bdf
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ
```

skgsergio, 2025-03-13:
Crosspostig it here:

This works:
```yaml
font:
  - id: font_matrixchunky6
    file: https://raw.githubusercontent.com/trip5/Matrix-Fonts/refs/heads/main/6-series/MatrixChunky6.bdf
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ

  - id: font_matrixchunky8
    file: https://raw.githubusercontent.com/trip5/Matrix-Fonts/refs/heads/main/8-series/MatrixChunky8.bdf
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ
```

[image]

This fails on compile with `Codepoints in bitmap fonts must be in the range 0-255.` error and is just a `curl -LO` of the URLs above:
```yaml
font:
  - id: font_matrixchunky6
    file: resources/fonts/MatrixChunky6.bdf
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ

  - id: font_matrixchunky8
    file: resources/fonts/MatrixChunky8.bdf
    glyphs: |-
      |¡!¿?"%()+*=,-_.:°€$@<>\/µ²³ 0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyzÁÉÍÓÑáéíóúñ

```

trip5, 2025-03-13:
OK.  The character that is likely causing the issue is `€` - outside the usual 8 bit range, which is where Pillow failed on BDF files.

Try printing that character to the clock on the "successful" pull-from-github run.  I'd be willing to bet it will not display correctly... it's just skipping over it when encoding, which is an interesting way to handle an error.  Lol.

trip5, 2025-03-13:
Let's continue the rest over at the ESPHome bug...

lubeda, 2025-03-14:
My solution is to use @trip5 's bdf fonts

```
font:
  - file: "https://raw.githubusercontent.com/trip5/Matrix-Fonts/refs/heads/main/8-series/MatrixLight8.bdf"
    id: default_font
    glyphs:  |-
      ! "#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]_abcdefghijklmnopqrstuvwxyzöäüÖÄÜ°{|}€~

  - file: "https://github.com/trip5/Matrix-Fonts/raw/refs/heads/main/6-series/MatrixChunky6X.bdf"
    id: special_font
    glyphs:  |-
      ! "#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyzöäüÖÄÜ°{|}¨¸€~   
```

with 

```
ehmtxv2:
  id: rgb8x32
  ....
  default_font_id: default_font
  default_font_yoffset: 9
  special_font_id: special_font
```

trip5, 2025-03-14:
Hey Lubeda.  Looks like the dev version is still stuck on checking Unicode characters but only when using local files for font.  Fetching from github has no issue at all.  Might be totally fixes with a PR on the ESPHome issue.  I think we got it before it makes it into an official release.  Kudos to @skgsergio for their diligence!

And really, ` default_font_yoffset: 9`?

lubeda, 2025-03-14:
```
font:
  - file: "https://raw.githubusercontent.com/trip5/Matrix-Fonts/refs/heads/main/6-series/MatrixChunky6.bdf"
    id: special_font
    glyphs:  |-
      ! "#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]_abcdefghijklmnopqrstuvwxyzöäüÖÄÜ°{|}€~

  - file: "https://raw.githubusercontent.com/trip5/Matrix-Fonts/refs/heads/main/8-series/MatrixChunky8.bdf"
    id: default_font
    glyphs:  |-
      ! "#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyzöäüÖÄÜ°{|}¨¸€~
```
and 

```
ehmtxv2:
  ...
  default_font_id: default_font
  default_font_yoffset: 8
  special_font_id: special_font
  special_font_yoffset: 6
``` 

and:

 on_empty_queue:
   then:
       - homeassistant.event:
           event: esphome.ehmtx_empty_queue
       - lambda: |-  
           id(rgb8x32).set_infotext_color(20,20,20,20,20,20,false,0);

skgsergio, 2025-03-14:
It should be fixed on next beta, the PR they've merged works for me with the ttf and local bdf. Will report otherwise.