lubeda/EspHoMaTriXv2 #285: [BUG] Spaces between words disappear with rainbow_shimmer option

wkneppers, 2025-03-22:
# Bug report

## Describe the bug
When using the option to rainbow_shimmer texts, spaces between words disappeared since the update to ESPHome 2025.3.0. 

## Additional information

- used Hardware:
  - ESP32/Ulanzi Pixelclock
   - EspHoMaTriXv2 version: 2025.2.1

## To Reproduce

Steps to reproduce the behavior:
Send out manual action with 'rainbow_icon_text_screen' action. Use more than 1 word, be sure the 'rainbow_shimmer' function is enabled.

## Expected behavior

Normally text is shown as filled in, but now all white space between the words is gone.

## Configuration

```text
ehmtxv2:
  id: rgb8x32
  icons2html: true
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  time_format: "%H:%M"
  date_format: "%d.%m"
  show_seconds: false
  show_dow: false
  default_font_id: px6_font
  default_font_yoffset: 8
  default_font_xoffset: 1
  special_font_id: ehmtx_font
  special_font_yoffset: 8
  special_font_xoffset: 1
  rainbow_shimmer: true
```

## Screenshots

-

## Logs

```text
-
```

## Services calls

```text
action: esphome.ulanzi_pixelclock_child_rainbow_icon_screen
metadata: {}
data:
  default_font: true
  icon_name: "{{ states('weather.forecast_thuis') }}"
  text: |-
    Het is buiten {{
      (states('sensor.climacontrol_slaapkamer_child_buiten_temperatuur') | float | round(1)) 
      if (states('sensor.climacontrol_slaapkamer_child_buiten_temperatuur') != 'unknown')
      else (state_attr('weather.openweathermap', 'temperature') | float | round(1))
    }}°C en {{ states('sensor.weather_condition_nl') }}
  lifetime: 1
  screen_time: 20
alias: Huidige weer
```

## Local fix
I disabled the rainbow_shimmer function by changing 'true' to 'false' and the problem was gone. So, it's not a big issue, but it isn't working like expected.

andrewjswan, 2025-03-23:
At low brightness, the symbols themselves may disappear, these are limitations of the matrix itself, at normal brightness nothing disappears, everything is in place and there are no problems.

wkneppers, 2025-06-19:
@andrewjswan It seems that's a completely different issue you're describing. The white space between characters disappears.

andrewjswan, 2025-06-19:
> The white space between characters disappears.

What font is used? BDF? Try TTF. I output information using rainbow shimmer every day and do not see such problems.