lubeda/EspHoMaTriXv2 #291: [BUG]

nouser2013, 2025-06-01:
# Bug report

## Describe the bug

`del_screen` does not (yet) work with `screen_id`'d icon names.

## Additional information

- used Hardware:
  - ESP32 (TC001 PixelClock)
  - EspHoMaTriXv2 version: `2025.5.2`

## To Reproduce

Steps to reproduce the behavior:
- Define an icon `plant` with e.g. a Lametric icon
- schedule an `icon_screen_progress` through home-assistant  
```yaml
- action: esphome.pixelclock_1_icon_screen_progress
  data:
    default_font: false
    icon_name: plant|01
    progress: 20
    lifetime: 10
    screen_time: 2
    r: 255
    g: 0
    b: 0
    text: "testtext"
```
- Call the `del_screen` service
```yaml
- action: esphome.pixelclock_1_del_screen
  data:
    icon_name: plant|01
    mode: 19
```

## Expected behavior

The screen should be removed from the queue.

## Logs

As seen from the logs, `del_screen` recognizes the correct mode, and jumps into the `if` statement, but then nothing.

```text
[14:01:31][D][EHMTXv2:1113]: oldest queue element is: 6/2
[14:01:31][D][EHMTXv2:119]: queue: icon screen: "temperature_outside" text: +23,3 for: 2.0 sec
[14:01:31][D][EHMTXv2:1350]: on_next_screen trigger
[14:01:31][D][EHMTXv2:1706]: del_screen: icon plant|02 in position: 02 mode 19
[14:01:33][D][EHMTXv2:1113]: oldest queue element is: 0/2
[14:01:33][D][EHMTXv2:122]: queue: icon progress: "02" text: #FFFFFF02#00FFFF:#FF000062% for: 2.0 sec
```

I suspect that the `screen_id` code that is already available in other functions is missing here. Namely, perhaps the comparison in line [1732](https://github.com/lubeda/EspHoMaTriXv2/blob/31cd208dd0b981763b782ce17cf48c5e9459fde9/components/ehmtxv2/EHMTX.cpp#L1732) fails and returns something else than `0`.

nouser2013, 2025-06-02:
daYUM - that surprisingly works, but I'd still consider this bug to be open:
* In theory, I could have several screen in mode `19` - `MODE_ICON_PROGRESS` with different icons showing, but having the same ID.
* Also, screens without ID can be deleted with the icon name alone 😅

But thanks, it seems to be one acceptable workaround for now (at least for me), besides the one to remove all `19` screens with:
```yaml
- action: esphome.pixelclock_1_del_screen
  data:
    icon_name: "*"
    mode: 19
```

nouser2013, 2025-06-02:
> This shouldn't happen, the icon changes according to the identifier.

I don't understand? I would picture a use case like this for 5 icon progress screens being queued (and deleted) independently:
* `icon_name: light|01`: brightness of desk light
* `icon_name: light|02`: brightness of ceiling light
* `icon_name: plant|01`: My first plant sensor's soil moisture
* `icon_name: plant|02`: My second plant sensor's soil moisture
* `icon_name: plant|03`: My third plant sensor's soil moisture

> This option should also work, I need to check it out, if not I'll fix it.

Agreed, it would be nice for consistency. And thank you.

andrewjswan, 2025-06-02:
> * `icon_name: light|01`: brightness of desk light

Show screen with icon `light` and ID `01`

> * `icon_name: light|02`: brightness of ceiling light

Show screen with icon `light` and ID `02`

> * `icon_name: plant|01`: My first plant sensor's soil moisture

Replace the icon on the screen (if it is still alive) with the ID `01` with the icon `plant`

> * `icon_name: plant|02`: My second plant sensor's soil moisture

Replace the icon on the screen (if it is still alive) with the ID `02` with the icon `plant`

> * `icon_name: plant|03`: My third plant sensor's soil moisture

Show screen with icon `plant` and ID `03`

nouser2013, 2025-06-02:
Ohhhhhh, that was not clear to me. I believed, also from the various `strcmp()`'s, that a screen's unique ID is the `icon_name` + `|` + `screen_id`. Good that you addressed this.

Might I then suggest to put above example and your explanation into the docs [here](https://github.com/lubeda/EspHoMaTriXv2/tree/2025.5.2?tab=readme-ov-file#screen_id)? Perhaps others will have the same question.

I now understand this sentence and your explanation better:

> If the screen identifier is specified, then all work on identification of the screen is carried out by it, as well as in functions like del_screen, force_screen it will be necessary to specify it and not the name of the icon.

This should definitely be more clarified:
* because the `del_screen()` (for example) expects an argument `icon_name` which implies that I should use the same format as I have when creating the screen with one of the icon-enabled screens.
* because I would not get from the sentence cited that the screen_id must be unique across a screen mode

andrewjswan, 2025-06-02:
Look like fixed ... https://github.com/lubeda/EspHoMaTriXv2/pull/292

nouser2013, 2025-06-02:
Great, thanks so much.

Hoping to have understood Github automatisms properly, I'd leave this issue open: merging the PR should close it then automatically.

andrewjswan, 2025-07-01:
> I use wildcard to delete all my different weather screen e.g. weather_sunny, weather_rain. It's not working anymore.

Thanks, as workaround use 
```yaml
data:
  icon_name: all|weather_*
  mode: 5
action: esphome.ulanzi_2_del_screen
```
or

https://github.com/lubeda/EspHoMaTriXv2/pull/305

andrewjswan, 2025-10-26:
Fixed ...