lubeda/EspHoMaTriXv2 #294: [BUG] ICON_SCREEN bottom line of text not shown

nouser2013, 2025-06-13:
# Bug report

## Describe the bug

If I have a text with a font which needs to write pixels to the last line of the PixelClock, this line is not shown. See attached picture, the German word `Mässig` has its `g` (lower case) truncated at the bottom.

## Additional information

- used Hardware:
  - ESP32
  - EspHoMaTriXv2 version: `2025.5.2`

## To Reproduce

Steps to reproduce the behavior:
- Use the indicated version of EspHoMaTrixV2
- See configuration below.
- Schedule an ICON_SCREEN with Text stretched across all 8 pixels, either:
  - with a lower case character, or
  - with an uppercase character when the font takes all 8 pixels for those

## Expected behavior

Text should be shown over the full display height.

## Configuration

```yaml
# [...]
font:
  - file: fonts/MatrixLight8.bdf
    id: default_font
    glyphs:  |-
      ! "#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZÄÖÜ[\]^_`abcdefghijklmnopqrstuvwxyzäöüß°
  - file: fonts/MatrixChunky6.bdf
    id: special_font
    glyphs: |-
      ! "#$%&'()*+,-./0123456789:APMTapmt
display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (y % 2 == 0) {
        return (y * 32) + x;
      }
      return (y * 32) + (31 - x);
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();
ehmtxv2:
  id: rgb8x32
  icons2html: true
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  show_seconds: false
  show_dow: true
  blend_steps: 0
  week_start_monday: true
  scroll_small_text: false
  vertical_scroll: false
  default_font_id: default_font
  default_font_yoffset: 8
  special_font_id: special_font
  special_font_yoffset: 6
  multicolor_text: true
  scroll_count: 1 
  scroll_interval: 32
  icons: 
    - id: error
      lameid: 26465
  # [...]
# [...]
```

## Screenshots

[image]

nouser2013, 2025-06-13:
@andrewjswan thanks, I just tried and rebooted the clock and rescheduled the screen, but same error. And yes, the clock's bottom row is working, I do have `ICON_SCREEN_PROGRESS` screens that work flawlessly 😅

andrewjswan, 2025-06-13:
> And yes, the clock's bottom row is working, I do have `ICON_SCREEN_PROGRESS` screens that work flawlessly

These two screens display text in the same way.

https://github.com/lubeda/EspHoMaTriXv2/blob/31cd208dd0b981763b782ce17cf48c5e9459fde9/components/ehmtxv2/EHMTX_queue.cpp#L884-L902

andrewjswan, 2025-06-13:
> In an icon_progress screen I would expect the text to be clipped due to the progress bar.

So far I don’t see where the bottom line in ICON_SCREEN can be cut off, there’s nothing here.

https://github.com/lubeda/EspHoMaTriXv2/blob/31cd208dd0b981763b782ce17cf48c5e9459fde9/components/ehmtxv2/EHMTX_queue.cpp#L991-L1007

This is strange, I need to read it...

https://github.com/lubeda/EspHoMaTriXv2/blob/31cd208dd0b981763b782ce17cf48c5e9459fde9/components/ehmtxv2/EHMTX_queue.cpp#L890

https://github.com/lubeda/EspHoMaTriXv2/blob/31cd208dd0b981763b782ce17cf48c5e9459fde9/components/ehmtxv2/EHMTX_queue.cpp#L903

Maybe you have the `ICON INDICATOR` turned on, large size?

andrewjswan, 2025-06-13:
I don't understand why this is here... 

```cpp
this->config_->display->start_clipping(0,0,0,0);
```

And why is there no `end_clipping`.

And I don't understand why `Screen Clipping` was added at all.

> [!NOTE]
> Screen clipping can be useful when you just want to show a part of an image or make sure that what you draw on the screen does not go outside a specific region on the screen.
>
> With `start_clipping(left, top, right, bottom);` start you the clipping process and when you are done drawing in that region you can stop the clipping process with `end_clipping();` . You can nest as many start_clipping(); as you want as long you end them as many times as well.

andrewjswan, 2025-06-13:
I would remove clipping altogether and make a PR

andrewjswan, 2025-06-13:
Great, that means it's fixed. :)

nouser2013, 2025-06-14:
Yes, seems that way 😀. Thank you very much for all your support, I really appreciate it.

andrewjswan, 2025-06-16:
@lubeda How are you?