lubeda/EspHoMaTriXv2 #310: [BUG] shifted text with 8x24px display

nerdyscout, 2025-08-25:
# Bug report

## Describe the bug

Hi,
am am using a esp32 wroom module and a 8x24px display, my complete config can be found [here](https://github.com/nerdyscout/esphome/blob/config/clock/config/clock.yaml)

## Additional information

- used Hardware:
  - ESP32
  - EspHoMaTriXv2 version: 2025.5.0-dev 

## To Reproduce

I can not really figure out why... but clock and date are always shifted to the right, as well the boot animation is off center. could this be something hardcoded in the EspHoMaTriXv2 component?
But the one screen with icon and temperature seems good to me. As I spent already several hours trying to fix this I hope for some hints what to do.
thanks alot. 

## Screenshots

[image]
[image]
[image]
[image]

andrewjswan, 2025-08-27:
It's not all that simple. Starting from `pixel_mapper`, and ending with calculations when processing rgb565 arrays.

```
    pixel_mapper: |-
      if (y % 2 == 0) {
        return (y * 32) + x;
      }
      return (y * 32) + (31 - x);
```

nerdyscout, 2025-08-27:
it is a dirty fix, but I managed to make it work at my [fork](https://github.com/nerdyscout/EspHoMaTriXv2/tree/width24)
maybe there are some side effects as I changed much more then I understood... but so far it is running nice!

closed 2025-08-30 by lubeda