lubeda/EspHoMaTriXv2 #312: [BUG]image distortions on platform esp-idf only

rrozema, 2025-09-14:
When the platform is set to esp-idf sometimes the display is temporarily distorted: horizontal lines that seem to be in an incorrect place, most of the times in the bottom of the display, but sometimes also other locations cabn show incorrect pixels or lines. It happens most when the image changes often, like while a scrolling text or an animated icon is shown. Most of the times the display is correct though. No errors are logged anywhere, only the image is sometimes incorrect.
When the same project is compiled for the arduino platform, the distortion disappears.

After a lot of searching I found a workaround in https://github.com/esphome/esphome/issues/10335. Still thought I'd share it here for others that experience the same issue. The workaround is to add
```wifi:
  power_save_mode: none
```

# Bug report

## Describe the bug

The screen shows temporary incorrect lines. I expect the screen to display as a correct text plus icon.

## Additional information

- used Hardware:
  - ulanzi TC001
  - EspHoMaTriXv2 version: [e.g. 2023.5.0]

## To Reproduce

Steps to reproduce the behavior:
Use the esp32 and light definitions as needed per recent esphome versions: framework set to esp-idf and thus light platform esp32_rmt_led_strip must be used.

## Expected behavior
Constantly correct image shown in the ulanzi's display.

## Configuration

mine is based on the ulanzi-easy.yaml example. Here are the relevant parts that trigger the issue:
```text
esp32:
  board: esp32dev
  framework:
    type: esp-idf

light:
  - platform: esp32_rmt_led_strip
    id: ehmtx_light
    rgb_order: GRB
    chipset: WS2812
```

## Screenshots

If applicable, add screenshots to help explain your problem.

https://github.com/user-attachments/assets/66b0e23d-6842-49d1-bf33-624a90895800

## Logs

Not Applicable

## Services calls

Not Applicable

andrewjswan, 2025-09-15:
> Other than the image distortions I had no issues with rmt, and those image issues are fixed now too using the work around.

This option on the same ESP32-C3 leads to an increase in temperature and sometimes to an unstable connection with the AP.

rrozema, 2025-10-17:
You are right, it does lead to the ulanzi getting warm too, so if anyone can create a proper fix that would be very welcome. But for now it is a work around for me.